linuxwacom/input-wacom #273: udev not creating proper event files?

mhamid3d, 2021-09-24:
Wacom Model: PTH-651K (Intuos Pro M)
Distro: Centos 7

It works completely fine in wired mode via USB, however when I use wireless mode via the USB bluetooth key, it does not work and I get the following errors in my `xorg` log:

```
[ 98871.893] (II) config/udev: Adding input device Wacom Intuos Pro M (WL) Finger (/dev/input/mouse3)
[ 98871.893] (II) No input driver specified, ignoring this device.
[ 98871.893] (II) This device may have been added with another device file.
[ 98871.894] (II) config/udev: Adding input device Wacom Intuos Pro M (WL) Pen (/dev/input/mouse2)
[ 98871.894] (II) No input driver specified, ignoring this device.
[ 98871.894] (II) This device may have been added with another device file.
[ 98871.903] (II) config/udev: Adding input device Wacom Intuos Pro M (WL) Finger (/dev/input/event20)
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "evdev touchpad catchall"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "evdev tablet catchall"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "touchpad catchall"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "Default clickpad buttons"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "Wacom USB tablet class"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "Wacom USB touchpad class"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "Wacom tablet class"
[ 98871.903] (**) Wacom Intuos Pro M (WL) Finger: Applying InputClass "Wacom touchpad class"
[ 98871.903] (II) LoadModule: "wacom"
[ 98871.904] (II) Loading /usr/lib64/xorg/modules/input/wacom_drv.so
[ 98871.904] (II) Module wacom: vendor="X.Org Foundation"
[ 98871.904] 	compiled for 1.20.4, module version = 0.40.0
[ 98871.904] 	Module class: X.Org XInput Driver
[ 98871.904] 	ABI class: X.Org XInput driver, version 24.1
[ 98871.904] (II) Build version: xf86-input-wacom-xf86-input-wacom-0.40.0
[ 98871.904] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
	Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
	Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
	Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
	CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
	CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
	Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
	PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
	DTU2231, DTU1631, DTH1152, DTH1152, DTK2451, DTH2452, DTH2452,
	Wacom One Pen Display 13, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
	Intuos2 12x12, Intuos2 12x18, Intuos2 6x8 , Volito, PenStation,
	Volito2 4x5, Volito2 2x3, PenPartner2, Bamboo, Bamboo1, Bamboo1 4x6,
	Bamboo1 5x8, Intuos3 4x5, Intuos3 6x8, Intuos3 9x12, Intuos3 12x12,
	Intuos3 12x19, Intuos3 6x11, Intuos3 4x6, Intuos4 4x6, Intuos4 6x9,
	Intuos4 8x13, Intuos4 12x19, Intuos4 WL USB Endpoint,
	Intuos4 WL Bluetooth Endpoint, Intuos5 touch S, Intuos5 touch M,
	Intuos5 touch L, Intuos5 S, Intuos5 M, Intuos Pro S, Intuos Pro M,
	Intuos Pro L, Cintiq 21UX, Cintiq 20WSX, Cintiq 12WX, Cintiq 21UX2,
	Cintiq 24HD, Cintiq 22HD, Cintiq 24HD touch (EMR digitizer),
	Cintiq 13HD, DTK2241, DTH2242, Cintiq 22HDT, Cintiq 27QHDT,
	Cintiq 27QHDT, Mobilestudio Pro 13, MobileStudio Pro 16,
	MobileStudio Pro 13, MobileStudio Pro 16, Cintiq 13 FHD Pro,
	Cintiq 16 UHD Pro, Cintiq Pro 24, Cintiq Pro 32, Cintiq 13 FHD Pro,
	Cintiq 16 UHD Pro, Cintiq Pro 24, Cintiq Pro 32, Cintiq Pro 24,
	Mobilestudio Pro 13, MobileStudio Pro 16, MobileStudio Pro 13,
	MobileStudio Pro 16, MobileStudio Pro 16, MobileStudio Pro 16,
	TabletPC 0x90, TabletPC 0x93, TabletPC 0x97, TabletPC 0x9A,
	CapPlus  0x9F, TabletPC 0xE2, TabletPC 0xE3, TabletPC 0xE5,
	TabletPC 0xE6, TabletPC 0xEC, TabletPC 0xED, TabletPC 0xEF,
	TabletPC 0x100, TabletPC 0x101, TabletPC 0x10D, TabletPC 0x116,
	TabletPC 0x12C, TabletPC 0x4001, TabletPC 0x4004, TabletPC 0x5000,
	TabletPC 0x5002, usb:172f:0024, usb:172f:0025, usb:172f:0026,
	usb:172f:0027, usb:172f:0028, usb:172f:0030, usb:172f:0031,
	usb:172f:0032, usb:172f:0033, usb:172f:0034, usb:172f:0035,
	usb:172f:0036, usb:172f:0037, usb:172f:0038, usb:172f:0039,
	usb:172f:0051, usb:172f:0052, usb:172f:0053, usb:172f:0054,
	usb:172f:0055, usb:172f:0056, usb:172f:0057, usb:172f:0058,
	usb:172f:0500, usb:172f:0501, usb:172f:0502, usb:172f:0503,
	usb:1b96:0001, usb:17ef:6004
[ 98871.906] (II) Using input driver 'wacom' for 'Wacom Intuos Pro M (WL) Finger'
[ 98871.906] (**) Wacom Intuos Pro M (WL) Finger: always reports core events
[ 98871.906] (**) Option "Device" "/dev/input/event20"
[ 98871.906] (EE) Wacom Intuos Pro M (WL) Finger: stat failed (No such file or directory). cannot check for duplicates.
[ 98871.906] (EE) xf86OpenSerial: Cannot open device /dev/input/event20
	No such file or directory.
[ 98871.906] (EE) Wacom Intuos Pro M (WL) Finger: Error opening /dev/input/event20 (No such file or directory)
[ 98871.906] (EE) PreInit returned 8 for "Wacom Intuos Pro M (WL) Finger"
[ 98871.906] (II) UnloadModule: "wacom"
[ 98871.909] AUDIT: Fri Sep 24 00:15:05 2021: 17886: client 16 connected from local host ( uid=1000 gid=1000 pid=18594 )
  Auth name: MIT-MAGIC-COOKIE-1 ID: 505
[ 98871.909] AUDIT: Fri Sep 24 00:15:05 2021: 17886: client 16 disconnected
[ 98871.916] (II) config/udev: Adding input device Wacom Intuos Pro M (WL) Pen (/dev/input/event19)
[ 98871.916] (**) Wacom Intuos Pro M (WL) Pen: Applying InputClass "evdev tablet catchall"
[ 98871.916] (**) Wacom Intuos Pro M (WL) Pen: Applying InputClass "Wacom USB tablet class"
[ 98871.916] (**) Wacom Intuos Pro M (WL) Pen: Applying InputClass "Wacom tablet class"
[ 98871.916] (II) Using input driver 'wacom' for 'Wacom Intuos Pro M (WL) Pen'
[ 98871.916] (**) Wacom Intuos Pro M (WL) Pen: always reports core events
[ 98871.916] (**) Option "Device" "/dev/input/event19"
[ 98871.927] (EE) xf86OpenSerial: Cannot open device /dev/input/event19
	No such device.
[ 98871.927] (EE) Wacom Intuos Pro M (WL) Pen: Error opening /dev/input/event19 (No such device)
[ 98871.927] (EE) PreInit returned 8 for "Wacom Intuos Pro M (WL) Pen"
[ 98871.927] (II) UnloadModule: "wacom"
```

jigpu, 2021-09-28:
I can replicate this issue as well on an updated CentOS 7.9 build and the (slightly older) PTH-650. It appears that the kernel device (e.g. `/dev/input/event20` in your log) is removed almost immediately after it is created. The process is so fast that the X driver can't even finish initializing. I don't see anything suspicious in the kernel logs when this occurs:

**dmesg**
~~~
[ 1291.709817] usb 1-3: new full-speed USB device number 3 using xhci_hcd
[ 1291.899153] usb 1-3: New USB device found, idVendor=056a, idProduct=0084, bcdDevice= 1.11
[ 1291.899160] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1291.899165] usb 1-3: Product: Wacom Wireless Receiver
[ 1291.899168] usb 1-3: Manufacturer: Wacom Co.,Ltd.
[ 1291.974270] usbcore: registered new interface driver wacom

=== SNIP ===

[ 1356.167571] wacom 1-3:1.0: wireless tablet connected with PID 27
[ 1356.167771] input: Wacom Intuos5 touch M (WL) Pen as /devices/pci0000:00/0000:00:02.1/0000:02:00.0/usb1/1-3/1-3:1.1/input/input7
[ 1356.171208] input: Wacom Intuos5 touch M (WL) Finger as /devices/pci0000:00/0000:00:02.1/0000:02:00.0/usb1/1-3/1-3:1.2/input/input8
~~~

Looking through sysfs, I see that the driver is still loaded and connected to the wireless dongle:
~~~
$ grep "" /sys/module/wacom/drivers/usb\:wacom/*/uevent
/sys/module/wacom/drivers/usb:wacom/1-3:1.0/uevent:DEVTYPE=usb_interface
/sys/module/wacom/drivers/usb:wacom/1-3:1.0/uevent:DRIVER=wacom
/sys/module/wacom/drivers/usb:wacom/1-3:1.0/uevent:PRODUCT=56a/84/111
/sys/module/wacom/drivers/usb:wacom/1-3:1.0/uevent:TYPE=0/0/0
/sys/module/wacom/drivers/usb:wacom/1-3:1.0/uevent:INTERFACE=3/0/0
/sys/module/wacom/drivers/usb:wacom/1-3:1.0/uevent:MODALIAS=usb:v056Ap0084d0111dc00dsc00dp00ic03isc00ip00in00
/sys/module/wacom/drivers/usb:wacom/1-3:1.1/uevent:DEVTYPE=usb_interface
/sys/module/wacom/drivers/usb:wacom/1-3:1.1/uevent:DRIVER=wacom
/sys/module/wacom/drivers/usb:wacom/1-3:1.1/uevent:PRODUCT=56a/84/111
/sys/module/wacom/drivers/usb:wacom/1-3:1.1/uevent:TYPE=0/0/0
/sys/module/wacom/drivers/usb:wacom/1-3:1.1/uevent:INTERFACE=3/0/0
/sys/module/wacom/drivers/usb:wacom/1-3:1.1/uevent:MODALIAS=usb:v056Ap0084d0111dc00dsc00dp00ic03isc00ip00in01
/sys/module/wacom/drivers/usb:wacom/1-3:1.2/uevent:DEVTYPE=usb_interface
/sys/module/wacom/drivers/usb:wacom/1-3:1.2/uevent:DRIVER=wacom
/sys/module/wacom/drivers/usb:wacom/1-3:1.2/uevent:PRODUCT=56a/84/111
/sys/module/wacom/drivers/usb:wacom/1-3:1.2/uevent:TYPE=0/0/0
/sys/module/wacom/drivers/usb:wacom/1-3:1.2/uevent:INTERFACE=3/0/0
/sys/module/wacom/drivers/usb:wacom/1-3:1.2/uevent:MODALIAS=usb:v056Ap0084d0111dc00dsc00dp00ic03isc00ip00in02
~~~

Looking at the CentOS kernel sources, the only thing that takes place after input devices are registered is initialization of the battery. It doesn't look like it prints out any error messages if it fails, but it *will* unregister the input devices. This might explain how the devices are created and then almost immediately silently disappear.

Debugging is a little complicated because of the split driver architecture used on RHEL/CentOS 7, but hopefully we can confirm whether or not the battery code is failing and then figure out how to fix or work around the problem.

jigpu, 2021-09-28:
It looks like the source of the problem is a conflict between our use of the battery APIs and Red Hat's decision to integrate some of the new battery API from Linux 4.1.

Specifically, the `power_supply_register` function (which we use to create the battery object) would previously return either an error code or the value `0` if there was no problem. Linux 4.1 changed its behavior to instead return either an error code _or a pointer to the created object_. This trips up our driver, which assumes that any non-zero value is an error. The "solution" is a patch similar to the following. After compiling a custom CentOS 7.9 kernel (based on the instructions at https://wiki.centos.org/HowTos/Custom_Kernel) with this change my PTH-650 began working again.

~~~patch
--- a/drivers/input/tablet/wacom_sys.c	2021-09-28 17:45:09.245562324 -0400
+++ b/drivers/input/tablet/wacom_sys.c	2021-09-28 17:46:46.039243709 -0400
@@ -1241,8 +1241,9 @@
 		}
 
 		error = wacom_initialize_battery(wacom);
-		if (error)
+		if (IS_ERR_OR_NULL(error)) {
 			goto fail;
+		}
 	}
 
 	return;
~~~

At the moment your options are to either apply this change and then compile the kernel from scratch (not easy or recommended), or wait for a patch like this to be accepted by the CentOS maintainers (could take a while). Making this change in input-wacom is unfortunately not so straightforward (due to that previously-mentioned split driver architecture), but we can see if there's a way to work around it...

mhamid3d, 2021-09-29:
Thank you @jigpu. Out of curiosity is the working test kernel you built able to be used by other CentOs users? I would really love to give the library a try since you have it built already and I don't want to take the risk of building my own.

jigpu, 2021-10-06:
I've filed the issue on Red Hat's tracker along with the patch. You can follow its progress at https://bugzilla.redhat.com/show_bug.cgi?id=2011442

I don't think I can provide a copy of the kernel packages. They're pretty big and may have issues when transplanted to another system (in theory it should work, but there's a lot that could go wrong). Finding a way to get input-wacom to build a driver is the better path forward.

mhamid3d, 2021-10-13:
@jigpu I can confirm this fix works flawlessly! Thank you so much.