linuxwacom/input-wacom #274: Thinkpad X380 Yoga Laggy Input

Shinmera, 2021-10-02:
Since a couple of months I'm getting super laggy pen input on my Thinkpad X380 Yoga (tablet reported as `Wacom Pen and multitouch sensor Pen stylus` in xinput). Using `libinput debug-tablet` I can see the values lagging way behind where the pen is. The cursor is also way behind, so it's not a UI toolkit issue. 

Sometimes instead of just lagging really badly the events drop out entirely and the cursor just jumps. Other times it works fine for a couple of seconds before lagging again. This makes it pretty unusable for drawing.

For what it's worth, using an external tablet like the Huion Kamvas works just fine without lagging, it's only the integrated wacom display that's exhibiting an issue. It used to work fine for years. It's possible that the hardware might just be biting the dust now, but since it sometimes works fine I really don't know.

Some system info:
```

Linux linuyy 5.14.8-arch1-1 #1 SMP PREEMPT Sun, 26 Sep 2021 19:36:15 +0000 x86_64 GNU/Linux
libwacom 1.12-1
xf86-input-wacom 0.40.0-1
```

If there's anything else I can provide or some other diagnostic I can run, please let me know!

jigpu, 2021-10-02:
I'm not aware of any other recent lag reports, so I'm unsure if this will turn out to be a driver issue (rather than, say, a hardware or firmware issue). On the topic of firmware, are you aware of any updates that might have been installed from Windows (if you have it installed)? Are there any updates available from Lenovo or support threads mentioning a similar issue?

Events dropping sounds like it could be a result of the pen not sending it's ID information in a timely manner. If this is occurring you would occasionally see error messages in dmesg saying that the kfifo is full/overflowing. It shouldn't result in lag per-se but might explain some of what's going on. You might try ensuring the pen is fully charged or has a fresh battery. Firmware issues can also cause unreliable ID transmission. We can provide an input-wacom patch to disable the ID check to see if it helps your situation at all as well.

Would you mind running and attaching the output of our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh)? This will give us a bit more info on the hardware, software, and configuration that may impact the driver.

Shinmera, 2021-10-03:
I don't have windows available on this laptop, but checking the manufacturer's site there don't seem to be any relevant updates. Can't find any relevant stuff on google either.

Not seeing any output in dmesg when I use the pen, even when it lags a lot. No idea how to check that the pen is charged, but I keep it inside the laptop's pen compartment at all times, which should charge it.

I've attached the sysinfo output: [sysinfo.3JvEJgYdXq.tar.gz](https://github.com/linuxwacom/input-wacom/files/7273108/sysinfo.3JvEJgYdXq.tar.gz)

Shinmera, 2021-10-05:
I do have tlp and powertop, though those have been set up for years without any changes from my side. The tunable for the pen and multitouch sensor is not active in powertop (meaning it should not be getting power suspended).

The lag also still occurs even when plugged in, I'm not noticing any difference. I'll see if I can test it with a live distro sometime soon, good idea!

Shinmera, 2022-01-17:
After someone suggested that the active pen battery might have been trashed, I ordered a new pen. This, in combination with deactivating tlp, seems to have fixed the issue, so I don't think it's related to the driver.

Sorry for the trouble!

closed 2022-01-17 by Shinmera