linuxwacom/input-wacom #276: Kernel panic on Cintiq Pro 24

vladlukianov, 2021-10-13:
Hello,

When we connect a touchscreen (usb) and hand moving on the screen in this moment, we get kernel panic.

OS: Oracle linux 8.4
Kernel version: 5.4.17-2102.202.5.el8uek.x86_64
wacom.ko: last current version (48)
Wacom model: Wacom cintiq pro 24

vmcore-dmesg.txt:
```
[  326.584139] input: Wacom Cintiq Pro 24 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:056A:0351.0018/input/input94
[  326.584234] input: Wacom Cintiq Pro 24 Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:056A:0351.0018/input/input96
[  326.584392] wacom 0003:056A:0351.0018: hidraw7: USB HID v1.10 Mouse [Wacom Co.,Ltd. Wacom Cintiq Pro 24] on usb-0000:00:14.0-1.6/input0
[  326.657350] BUG: kernel NULL pointer dereference, address: 0000000000000030
[  326.657352] #PF: supervisor read access in kernel mode
[  326.657352] #PF: error_code(0x0000) - not-present page
[  326.657353] PGD 0 P4D 0.
[  326.657354] Oops: 0000 [#1] SMP PTI
[  326.657356] CPU: 5 PID: 0 Comm: swapper/5 Kdump: loaded Tainted: G           OE     5.4.17-2102.202.5.el8uek.x86_64 #2
[  326.657357] Hardware name: Default string Default string/COMe-bCL6, BIOS BCL6R113 09/24/2020
[  326.657360] RIP: 0010:wacom_wac_report+0x841/0x8d0 [wacom]
[  326.657361] Code: e9 f2 fa ff ff 80 78 29 00 0f 85 b3 fd ff ff 80 78 01 00 0f 85 a9 fd ff ff e9 4b f9 ff ff 49 8b 44 c4 30 49 63 95 90 03 00 00 <48> 8b 40 30 8b 04 90 85 c0 0f 85 1e f9 ff ff e9 20 f9 ff ff c7 83
[  326.657362] RSP: 0018:ffff943280234b48 EFLAGS: 00010002
[  326.657363] RAX: 0000000000000000 RBX: 0000000000000004 RCX: 0000000000000001
[  326.657363] RDX: 0000000000000000 RSI: 000000000103ff00 RDI: 00000000000d0056
[  326.657364] RBP: ffff943280234ba8 R08: 0000000000000000 R09: ffff892bd755bf70
[  326.657364] R10: 0000000000000001 R11: 0000000000000081 R12: ffff892bce9d3000
[  326.657365] R13: ffff892bcca4e818 R14: ffff892bd755bf00 R15: 0000000000000001
[  326.657366] FS:  0000000000000000(0000) GS:ffff892bddb40000(0000) knlGS:0000000000000000
[  326.657366] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  326.657367] CR2: 0000000000000030 CR3: 0000000445e0a004 CR4: 00000000003606e0
[  326.657367] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  326.657368] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  326.657368] Call Trace:
[  326.657370]  <IRQ>
[  326.657374]  hid_report_raw_event+0x20c/0x450
[  326.657376]  hid_input_report+0x10b/0x160
[  326.657378]  hid_irq_in+0x1a2/0x1c8
[  326.657380]  __usb_hcd_giveback_urb+0x7a/0x11f
[  326.657382]  usb_hcd_giveback_urb+0xf3/0x109
[  326.657384]  xhci_giveback_urb_in_irq.isra.46+0x82/0xe9
[  326.657385]  xhci_td_cleanup+0x112/0x142
[  326.657386]  finish_td+0xbb/0x15e
[  326.657388]  handle_tx_event+0x4a9/0x1217
[  326.657390]  xhci_irq+0x1b1/0x388
[  326.657392]  xhci_msi_irq+0x11/0x13
[  326.657394]  __handle_irq_event_percpu+0x44/0x187
[  326.657395]  handle_irq_event_percpu+0x32/0x76
[  326.657397]  handle_irq_event+0x3b/0x5a
[  326.657398]  handle_edge_irq+0x87/0x18d
[  326.657400]  do_IRQ+0x54/0xe0
[  326.657401]  common_interrupt+0xf/0x1d2
[  326.657402]  </IRQ>
[  326.657403] RIP: 0010:cpuidle_enter_state+0xbd/0x44b
[  326.657404] Code: ff e8 57 4c 9a ff 80 7d c7 00 74 17 9c 58 0f 1f 44 00 00 f6 c4 02 0f 85 63 03 00 00 31 ff e8 2a 4a a0 ff fb 66 0f 1f 44 00 00 <45> 85 ed 0f 88 8d 02 00 00 49 63 cd 48 8b 75 d0 48 2b 75 c8 48 8d
[  326.657405] RSP: 0018:ffff9432800ffe48 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffde
[  326.657406] RAX: ffff892bddb6b1c0 RBX: ffffffff889439e0 RCX: 000000000000001f
[  326.657406] RDX: 0000004c0e4b6797 RSI: 000000004382b7f9 RDI: 0000000000000000
[  326.657407] RBP: ffff9432800ffe88 R08: 0000000000000002 R09: 000000000002a940
[  326.657407] R10: 000000991c4cfe51 R11: ffff892bddb69ec0 R12: ffffb4327fb40000
[  326.657408] R13: 0000000000000004 R14: ffffffff88943b78 R15: ffffffff88943b60
[  326.657411]  ? cpuidle_enter_state+0x99/0x44b
[  326.657412]  cpuidle_enter+0x2e/0x3d
[  326.657414]  call_cpuidle+0x23/0x3a
[  326.657416]  do_idle+0x21c/0x251
[  326.657418]  cpu_startup_entry+0x1d/0x1f
[  326.657420]  start_secondary+0x166/0x1b6
[  326.657421]  secondary_startup_64+0xb6/0xb6
[  326.657423] Modules linked in: wacom(OE) snd_seq_dummy snd_hrtimer fuse squashfs loop crypto_simd glue_helper dm_crypt binfmt_misc vfat fat intel_rapl_msr intel_rapl_common snd_hda_codec_hdmi iTCO_wdt iTCO_vendor_support ppdev snd_hda_codec_realtek x86_pkg_temp_thermal intel_pow
[  326.657449]  cxgb4i cxgb4 cxgb3i cxgb3 mdio libcxgbi libcxgb qla4xxx iscsi_boot_sysfs iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi [last unloaded: wacom]
[  326.657455] CR2: 0000000000000030
```

What can we do to solve this problem?
Thanks.

Pinglinux, 2021-11-30:
Can you build the kernel driver by following the steps at: https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source, if you haven't done so?

vladlukianov, 2021-12-01:
> Can you build the kernel driver by following the steps at: https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source, if you haven't done so?

Assembled the driver, the problem remained.
version:        v2.00-0.48.0.10.g59402f2

Pinglinux, 2021-12-03:
We tested it on a kernel 4.19 system. We could not reproduce the issue. Can you try the following steps to see if the issue occurs:

1. Start the system with the tablet connected.
     1.1 Touch the screen while the system is loading (this is not a recommended action since the system isn't ready!). Do you see the issue?
     1.2 Touch the screen after the system is fully started. What happens?
     1.3 Do you need to login to the system? If yes, please try it both before and after you login to see if there is any difference.
2. Start the system without the tablet connected.
     2.1 Touch the screen while plugging the tablet (again, not recommended). What happens?
     2.2 Touch the screen after the tablet is fully plugged in and stable. What happens?
     2.3 Does log in or not make a difference?

Also, do you have another kernel version to try it on?

vladlukianov, 2021-12-03:
[vmcore-dmesg.txt](https://github.com/linuxwacom/input-wacom/files/7649521/vmcore-dmesg.txt)

I have tested it on a kernel 4.18 sytem and wacom.ko version v2.00-0.48.0.10.g59402f2. Posible problem with USB 3.0. While touching and moving (!) finger the screen I connect/disconnect cable many times in USB 2.0, but when i tried to do this manipulations with USB 3.0 it crashed  Kernel  Panic. The dmesg demonstraits it. The touchscreen working correctly in any other cases.  You are not recommend touching the screen since the system isn't ready, but it's important  case for our customer because there is KVM switch in our system and when somebody switch it, a connect / disconnect occurs.

Pinglinux, 2021-12-04:
It is an electronic device. For all devices/tools, there are proper ways to use them. The devices react to what we tell them to do. If we give them confusing instruction, they get "confused" too. In theory, when data/device wasn't ready, the driver should not process any events. However, there could be corner cases or under race conditions, data from unexpected user activities trip in. Can you explain the use case a bit more, like why did your customers need to touch the screen while the KVM switch was in action? You probably need to make sure they understand the situation and avoid touching the device when switching is on process.

If USB 2 cable worked, can they stay with USB 2 cable? Maybe their system can not catch up with the superspeed USB 3 cable. Or maybe it's just that particular USB 3 cable is causing issue. Did you try another USB 3 cable?

Your Call Trace showed hid/usb common code...

jigpu, 2021-12-09:
Can you please attach the v2.00-0.48.0.10.g59402f2 wacom.ko driver that you compiled which triggered the previous crash on kernel 4.18? You should be able to find it in at `input-wacom/4.5/wac.ko`. I tried to find the line number of the crash from a locally-compiled copy but the location it pointed to was nonsense (or that another thread caused the variable to become null).

vladlukianov, 2021-12-10:
[wacom.zip](https://github.com/linuxwacom/input-wacom/files/7691573/wacom.zip)
1) There was a misunderstanding. We use the only USB cable, officiallly packaged with WACOM itself. 
The problem occurs both when we plug it in USB 3.0 socket or USB 2.0 socket. It doesn't depend on cable itself.

2)I accept all the things, written by Pinglinux, and the stuff about race conditions, but this shouldn't be an excuse for that particular case. We cannot force user to avoid touching WACOM surface, while it is in process of initialization. Any user input during initialization process should be simply ignored by the WACOM. At least not cause the kernel panic effect.

3) Anyway, how can we fix the problem, taking into account that we cannot guarantee that user won't touch WACOM surface before it is fully initialized?

4) **Pinglinux wrote:** _"Your Call Trace showed hid/usb common code..."_ Maybe we can provide any additional information, that would be usefull for you, and help to reproduce the effect?

jigpu, 2021-12-10:
Thanks for the copy of the driver. Unfortunately(?) I'm still seeing the same general area which makes little sense for a crash. The RIP in the crash is set to `wacom_wac_report+0x329`, which eu-addr2line and gdb both indicate traces back to line 2697 of wacom_wac.c:

https://github.com/linuxwacom/input-wacom/blob/59402f2ae1c87cf5b97f4b1d535ea545be03c3e0/4.5/wacom_wac.c#L2694-L2703

Squinting at the disassembly seems to indicate that the %rax register is storing the (null) `field` pointer:

~~~
2694                if (hid_data->cc_report != 0 &&
   0x00000000000044a6 <+774>:  45 8b 9f 88 03 00 00  mov    0x388(%r15),%r11d
   0x00000000000044ad <+781>:  45 85 db              test   %r11d,%r11d
   0x00000000000044b0 <+784>:  74 24                 je     0x44d6 <wacom_wac_report+822>

2695                   hid_data->cc_index >= 0) {
   0x00000000000044b2 <+786>:  49 63 87 8c 03 00 00  movslq 0x38c(%r15),%rax

2694                if (hid_data->cc_report != 0 &&
   0x00000000000044b9 <+793>:  85 c0                 test   %eax,%eax
   0x00000000000044bb <+795>:  78 19                 js     0x44d6 <wacom_wac_report+822>

2696                        struct hid_field *field = report->field[hid_data->cc_index];
2697                        int value = field->value[hid_data->cc_value_index];
   0x00000000000044bd <+797>:  48 8b 44 c3 30        mov    0x30(%rbx,%rax,8),%rax
   0x00000000000044c2 <+802>:  49 63 97 90 03 00 00  movslq 0x390(%r15),%rdx
   0x00000000000044c9 <+809>:  48 8b 40 30           mov    0x30(%rax),%rax                      ; <<< CRASH >>>
   0x00000000000044cd <+813>:  8b 04 90              mov    (%rax,%rdx,4),%eax

2698                        if (value)
   0x00000000000044d0 <+816>:  85 c0                 test   %eax,%eax
   0x00000000000044d2 <+818>:  75 09                 jne    0x44dd <wacom_wac_report+829>
   0x00000000000044d4 <+820>:  eb 0e                 jmp    0x44e4 <wacom_wac_report+836>

2699                                hid_data->num_expected = value;
2700                }
2701                else {
2702                        hid_data->num_expected = wacom_wac->features.touch_max;
   0x00000000000044d6 <+822>:  41 8b 87 0c 03 00 00  mov    0x30c(%r15),%eax
   0x00000000000044dd <+829>:  41 89 87 98 03 00 00  mov    %eax,0x398(%r15)
   0x00000000000044e4 <+836>:  48 8b 04 24           mov    (%rsp),%rax
   0x00000000000044e8 <+840>:  4c 8b b8 a8 19 00 00  mov    0x19a8(%rax),%r15
~~~

This doesn't really make sense though. The loop just above this block derefrerences each `report->field[i]` while it sets up the variables needed for this block to work. The only thing I can think of is that either another thread is modifying `report->field` under our noses or that we've somehow switched from receiving touch reports with "contact count" data to reports _without_ that data -- this would cause the values of `cc_report`, `cc_index` and `cc_value_index` to be stale.

In case its the latter, I've uploaded a test commit which should work around the issue. Please run `git clone https://github.com/jigpu/input-wacom.git -b stale-data-workaround` to clone a copy of my repository at the correct branch. Follow the regular process to build/install from git. If you see the crash again, please reply back with a copy of the crash from dmesg and the compiled wacom.ko file again.

vladlukianov, 2021-12-13:
Good day, 
we've built your last patch and it works fine and stopped crashing.
I've looked to "git diff" on your last patch and it seems to me that you fixed the problem, but in a strange way. I would rather add check before line 2697 (in your message above) that field->value[] container(vector?) contains hid_data->cc_value_index, at least has size >= than that. Maybe, going out of field->value[] bounds causes the problem.
Never the less, thanks a lot!

jigpu, 2021-12-13:
Glad to hear the patch worked. As the commit states, this was more of a work-in-progress patch to verify the root cause. Now that we know the crash is caused by the driver dealing with stale data, we can write a more proper fix. I'll try to have a more proper fix in the coming days.

jigpu, 2022-01-21:
Fix has been accepted upstream and should be available in the next RC release of the Linux kernel. Stable releases of the kernel should also get the fix "soon". I'll update again once I have version numbers.

jigpu, 2022-01-24:
Queued for stable (5.16.x, 5.15.x, 5.10.x, 5.4.x, 4.19.x, 4.14.x, 4.9.x)

jigpu, 2022-02-02:
Patch accepted by upstream and stable:
* v5.17-rc1~20^2
* v5.16.3~1026
* v5.15.17~835
* v5.10.94~556
* v5.4.174~312
* v4.19.226~213
* v4.14.263~166
* v4.9.298~141

closed 2022-02-02 by jigpu