linuxwacom/input-wacom #283: Bluetooth - Wacom Intuos Pro M weird writing

omennia, 2021-11-13:
Good evening,
I have a Wacom Intuos Pro M, that I use mostly for university lectures.
When I have the tablet plugged in, I can use the touch function and writing/drawing simultaneously. I mean, the tablet seems to ignore the touch input whenever the pen is close enough. And this is great for me.
Unfortunately it does not happen when I am connected through Bluetooth, the writing gets all awkward and I am forced to turn off the touch function, although I would make nice use of it. It's bothersome to have to plug the tablet it when on the go.

If anyone can help me with this issue, I would be very thankful.
I am using Fedora 35, Gnome 41.1, X11

Thanks

omennia, 2021-11-18:
I tried to reproduce the issue with Ubuntu, and could not. Ubuntu worked great. So I performed a clean installation of fedora and the problem still persists.
Is it fedora related? If someone could please point me in the right direction I would appreciate it.

Thanks

omennia, 2021-11-30:
Hi Jlgpu, thanks for your reply.
1. The model is PTH-660.
2. I tested it with Ubuntu 21.04 with gnome and X11 as the display manager, and also tried it with Linux Mint 20.2 Cinnamon Edition. Both work fine.
3. The side button works as expected, although the desired effect is to leave it on and have the touch function work with bluetooth as it works when connected by USB.

Things I have noticed:
On ubuntu, toggling off the laptop's touchpad does not toggle off the tablets touchpad whereas in fedora it does.
On ubuntu, the tablet's touchpad allows me to zoom in / out with finger gestures whereas fedora doesn't.

Joshua-Dickens, 2021-12-02:
Hi Omennia, I'm currently working on trying to replicate your issue but can't seem to replicate it. Do you mind telling me your exact process from starting the tablet to getting to the issue? I'll keep attempting different ways to trigger this issue in the meantime.

omennia, 2021-12-02:
Hey Joshua, thanks for your reply.
So I boot the computer, log in and connect the tablet by bluetooth.
If I have the pen in proximity with the tablet, for example when trying to write, the following happens:
[https://youtu.be/4iIh7uo-T8c](url)
I am going to install fedora into another laptop and check if the issue persists.
I will update the post with this information.
Thanks for your time.

omennia, 2021-12-02:
Update:
I have installed fedora 35 in another laptop, clean installation, only thing I did was changing it from Wayland to X11.
Unfortunately the problem persists.

Joshua-Dickens, 2021-12-03:
Okay so I still have been unable to replicate the issue so I have a few more questions.
How long has this issue been present, has this been an issue since you got the tablet or did it just recently appear after normal use?
If you use wayland instead of X11 does the issue still present itself?
And last in the example video are you attempting to write with the pen or your finger?
Just trying to narrow down the possibilities as much as possible.

omennia, 2021-12-03:
The issue has been present since I got the tablet. At the time I was using fedora 34, with gnome 40, x11.
When using wayland, the strange behaviour does not happen, and there are two cursors(i remember reading somewhere that was a design choice, although it is a little annoying) but it works. Unfornately I want to use x11, because it works better with everything else I use, especially screen sharing.
In the video I was writing with the pen and making a circular motion with my fingers just to intensify the issue. Sometimes it scrolls up and down amidst the writing.
Thanks for your time!

Pinglinux, 2021-12-03:
@omennia Let's see if we can figure out the root cause by investigating the actual tablet events. Please run "libinput record" to record the events that triggered the issue and share it with us here.

omennia, 2021-12-03:
Hey @Pinglinux, I am not too sure how libinput record works, so I selected both the Wacom Pad and the Wacom Finger from the device list, and recorded them for a minute or so each.
Let me know if this was the intended request
[IntuosFinger.txt](https://github.com/linuxwacom/input-wacom/files/7652792/IntuosFinger.txt)
[IntuosPad.txt](https://github.com/linuxwacom/input-wacom/files/7652793/IntuosPad.txt)

Thank you!

Pinglinux, 2021-12-03:
Thanks @omennia for the quick feedback. Do you see Wacom Stylus/Pen or something like that? We don't need Pad, which is for the tablet buttons. We need the stylus/pen, which is what you used to draw.

omennia, 2021-12-04:
Hey @Pinglinux I should have posted it straight away with the others but here it goes
[WacomPen.txt](https://github.com/linuxwacom/input-wacom/files/7652966/WacomPen.txt)

Thanks for the reply

Pinglinux, 2021-12-04:
Thank you for the info. Your WacomPen tells me a lot details. What I am interested in is the BTN_TOUCH event. There are many switch between 1 to 0 in your file. BTN_TOUCH=1 means pen is in-prox. BTN_TOUCH=0 means it is out. When it is out, touch is activated immediately since no pen is in prox. You may not realize that your pen has been moved out of the proximity of the active area. But, when you draw/write, you most likely subconsciously moved the pen out of range.

If you don't expect touch to do anything, you probably should not leave your figure on the surface (or should turn the touch off while you draw). Only turn touch on when you need to use touch features. It's a hardware, it can only do what we tell it to do...

omennia, 2021-12-04:
Hey @Pinglinux
Thanks for the advice.
The problem is if the touch function is enabled, even if I don't have my finger on the surface, the writing gets awkward because in order to write my hand is resting on the tablet as I write, like you would do in a regular piece of paper. The aforementioned "bug" happens even if the pen is touching the surface at all time. 
Furthermore it does not happen when it is connected by USB or when using a different OS.
I have a regular Intuos tablet, and I bought this one specifically because of the touch function. If I have to leave it off at all times, it is counter productive and the Intuos Pro brings nothing new to the table, at least for me. 
I don't think migrating to another OS entirely should be the solution.
So it is either a bluetooth related issue, a driver issue, or a issue coming from the OS.
Anyway I thank you for your time, and I hope there is a way to turn this around.

jigpu, 2021-12-07:
Please run `sudo dnf install hid-replay` to install a tool to record the low-level hardware events and then follow these steps:

1. Connect the tablet over Bluetooth
2. Run `sudo hid-recorder | tee recording.log`
3. A single device should be listed for your tablet (e.g. "CL IntuosPro M"). Type in its number and press Enter.
    * You should see a bunch of data printed to the screen beginning at this step. More should be printed out whenever events are received.
4. Make three short pen strokes on the device without touching your hand to the tablet.
5. Make three short single-finger swipes on the device without the pen in proximity.
6. Use the pen normally for several seconds, allowing your hand to rest on the tablet.
7. When you're done, press CTRL+C to quit recording.
8. Attach the log to this issue for review.

omennia, 2021-12-07:
Hi @jigpu 
Thanks for your reply.
I reproduced the steps above and the resulting log is the following:
[recording.log](https://github.com/linuxwacom/input-wacom/files/7671306/recording.log)

jigpu, 2021-12-09:
Thanks. When replaying it on my main Arch dev box I see that arbitration is being handled within the X11 driver rather than the kernel driver. This is a bit unexpected (we usually perform arbitration in-kernel), but still should work on Fedora 35. I'm setting up a fully-updated Fedora 35 install to see what happens when I replay the events over there.

omennia, 2021-12-10:
Thank you @jigpu
Please let me know how it goes.

Kind regards

jigpu, 2021-12-10:
Replaying your logs on Fedora 35 I see occasional touch events bleeding through the kernel arbitration. As I mentioned earlier, these are normally taken care of by our X driver, but it seems that Fedora 35 actually re-assigns the touchpad to the xf86-input-libinput driver. This prevents userspace arbitration from working.

You can re-assign the touchpad to our driver by creating a file named `/etc/X11/xorg.conf.d/99-user-wacom-override.conf` with the following contents and rebooting:
~~~
# Override Fedora 35's decision to have libinput control Wacom
# touchpads. This restores pen arbitration. 
Section "InputClass"
        Identifier "Wacom USB touchpad user override"
        MatchUSBID "056a:*"
        MatchDevicePath "/dev/input/event*"
        MatchIsTouchpad "true"
        Driver "wacom"
EndSection
~~~

omennia, 2021-12-10:
Hello @jigpu 
Thank you very much for all the effort.
I can confirm that now everything pretty much works like it did on Ubuntu or Mint.
So I consider the problem solved.
Once again, thank you!

closed 2021-12-10 by omennia