linuxwacom/input-wacom #285: How to adjust polling interval (Intuos4M)

hendersj, 2021-12-01:
I've been trying to figure out how to look at/adjust the polling interval. I use an Intos4 M with openSUSE Leap 15.3 (previously with 15.2 and earlier releases).

I ran into an issue about 6 weeks ago when using my device with Stadia in Chrome; part of the troubleshooting that I'm being asked to do is to look at the polling interval and adjust it.  Typically this is done by modifying the configuration for the usbhid module, but the changes don't seem to work with the wacom driver.  (https://wiki.archlinux.org/title/mouse_polling_rate is a generic enough reference).

The mouse moves around the screen normally outside Stadia, but once in-game, the mouse is dragged to the edges of the screen and is very difficult to control in any meaningful way.  This is said to be typical of having a misconfigured polling rate.

(I've asked about this in the openSUSE forums as well as the Stadia community - nobody has any ideas, so I thought I'd "go to the source" for the kernel driver).

If there's a better place for a question like this, please let me know.

hendersj, 2021-12-01:
Here's the relevant output from /sys/kernel/debug/usb/devices

```
T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=12   MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=056a ProdID=00b9 Rev= 1.04
S:  Manufacturer=Tablet
S:  Product=PTK-640
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=300mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=  10 Ivl=4ms

```
The setting that's been recommended is for the Ivl to be set to 8, as I recall - but knowing how to change that is what would be most helpful.

jigpu, 2021-12-07:
The `usbdev.mousepoll` option should take effect for any USB HID device whose first collection is of type `HID_GD_MOUSE`. I don't see an Intuos4 in our wacom-hid-descriptors repository so I can't be 100% sure that your device's HID descriptor is set up like this, though it _is_ common of other Wacom tablets from that era. I'd recommend following the "contributing" instructions at https://github.com/linuxwacom/wacom-hid-descriptors#contributing so we can have a bit more information about the device to confirm.

Testing with my Cintiq 24HDT (which has a mouse as its first collection), I can see that the interval does change but that there's no effect on the report rate as measured by the `evhz` command. This isn't entirely unexpected, however, since Wacom tablets should only be capable of reporting at ~133Hz regardless of what the polling interval is set to. Note that the changed polling rate is _not_ reflected in the output from `/sys/kernel/debug/usb/devices` -- I can only verify the change in polling rate by checking the output from Wireshark...

I suspect the polling rate of the Intuos4 is being changed by the mousepoll setting, but has no practical effect on the actual rate at which events are sent by the kernel.

hendersj, 2021-12-07:
Thanks for the response, Jason.  Attached is the output from the sysinfo script.

[sysinfo-wacom.tar.gz](https://github.com/linuxwacom/input-wacom/files/7672253/sysinfo-wacom.tar.gz)

jigpu, 2021-12-08:
Thanks! It looks like the Intuos4 also has a mouse as its first collection, so the mousepoll command should be working.

If you want to 100% confirm the actual USB polling rate you can try the following:
1. Run `lsusb -d 056a:` to find the bus and device number for your tablet (2 and 21 for the example below)
2. Run `sudo modprobe usbmon` to load the USB monitoring module
3. Run `sudo awk -vbus=<BUSNUM> -vdev=<DEVNUM> 'BEGIN {regex = sprintf("C Ii:%d:%03d", bus, dev)} {if ($0 ~ regex) {print $5}}' /sys/kernel/debug/usb/usbmon/0u` with the BUSNUM and and DEVNUM from step 1
4. Touch the pen to the tablet
5. You should see two colon-seperated numbers. The USB polling rate is the second number (for my Cintiq it is normally 1ms, but it now shows a value of 8ms after having set mousepoll to 8)
6. Run `sudo modprobe -r usbmon` to unload the kernel module

Example:
~~~
$ lsusb -d 056a:
Bus 002 Device 021: ID 056a:00f8 Wacom Co., Ltd DTH-2400 [Cintiq 24HD touch] tablet
Bus 002 Device 020: ID 056a:00f6 Wacom Co., Ltd DTH-2400 [Cintiq 24HD touch] touchscreen
Bus 002 Device 019: ID 056a:00f5 Wacom Co., Ltd Cintiq 24HDT Monitor Control

$ sudo modprobe usbmon

$ sudo awk -vbus=2 -vdev=21 'BEGIN {regex = sprintf("C Ii:%d:%03d", bus, dev)} {if ($0 ~ regex) {print $5}}' /sys/kernel/debug/usb/usbmon/0u
0:8
0:8
0:8
0:8
0:8
0:8
0:8
0:8
0:8
0:8
0:8
0:8
^C

$ sudo modprobe -r usbmon
~~~

hendersj, 2021-12-08:
Here's the output I generated:

```
[jhenderson@artifactovol ~]$ lsusb -d 056a:
Bus 002 Device 002: ID 056a:00b9 Wacom Co., Ltd PTK-640 [Intuos4 (6x9)]
[jhenderson@artifactovol ~]$ sudo modprobe usbmon
[jhenderson@artifactovol ~]$ sudo awk -vbus=2 -vdev=2 'BEGIN {regex= sprintf("C Ii:%d:%03d", bus, dev)} {if ($0 ~ regex) {print $5}}' /sys/kernel/debug/usb/usbmon/0u
0:4
0:4
0:4
0:4
0:4
0:4
0:4
[...]
```

I'll try making the change again and see if it updates correctly - this is very useful information - thanks!

hendersj, 2021-12-08:
Just tried adding:

```
options usbhid mousepoll=8
```
to /etc/modprobe.d/usbhid.conf and reloaded the wacom and usbhid drivers with:

```
modprobe -r wacom usbhid && modprobe wacom usbhid
```

Unfortunately, the output from the awk script was the same repeated value of 0:4.  No change in behavior in Stadia either.

jigpu, 2021-12-09:
It might be necessary to unplug the tablet and reconnect it, or for you to restart for the change to usbhid.conf to take effect.

You can also try running `echo 8 | sudo tee /sys/module/usbhid/parameters/mousepoll` and then unplugging and replugging the tablet. This is the method I used when testing with my Cintiq above.

hendersj, 2021-12-09:
Thanks for that additional info.  I did just try unplugging and replugging the tablet - I noted that the `/sys/module/usbhid/parameters/mousepoll` value is 8 already.  Still seeing 0:4 repeated when I look (now with the new device ID).

jigpu, 2021-12-09:
Ah, well that's no good. Giving a second look at the sysinfo output you provided, it appears the USB port that you're plugged into is off of an xHCI USB controller. The Arch wiki has a section for "Polling rate not changing" which says that the xhci driver doesn't honor the polling interval. It suggests trying to connect the device to other ports on your system (in case one of them is just USB 2.0) or blacklisting the xhci-hcd kernel module. You might also try putting a USB 2.0 hub between the tablet and the PC, though I'm not sure if that would do anything since xhci would still be in control of polling the hub...

hendersj, 2021-12-09:
Somehow I missed that as well - I'll see about plugging into a different port - looks like the system has two EHCI controllers as well.

Pinglinux, 2022-07-25:
Not a tablet or driver issue

closed 2022-07-25 by Pinglinux