linuxwacom/input-wacom #287: touch screen monitor causes x-server to lock up

eebender, 2021-12-03:
I am trying to connect a touch screen monitor to my linux laptop and it causes the x-server to freeze when I try to connect.

Laptop:  system76 Serval with AMD ryzen  7 3700x cpu and nvidia rtx 2070 gpu.

Monitor:  Innocn n1f pro with 1920x1080 resolution. This particular model can be found on amazon at https://www.amazon.com/gp/product/B098NKQ5J6/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

The operating system seems to be invoking the wacom driver.  Any recommendations?

Pinglinux, 2021-12-04:
> The operating system seems to be invoking the wacom driver. Any recommendations?

Where did you see the message? Can you share the whole "dmesg" with us?

eebender, 2021-12-04:
Here is some of the output I got from the log files when working with
System76 support.  I output data from dmesg, syslog, and sys76 logs (I
didn't include the sys76 log here since that seems vendor specific).  My
goal was to capture the output of the log files when the x-server was in
the process of locking up. I describe the procedure I used to capture this
output below. The output from the dmesg and syslog files are attached.
There does not seem to be any output to the dmesg log during this process
and all the information regarding invoking the driver seems to be in the
syslog log.

To do this I used two machines : 'daffy' which is my old system76 kudu
laptop, and 'taz' my new serval laptop.  I attempted to connect the monitor
to  taz.  Before doing this I opened three remote ssh sessions from daffy
to taz.  In each terminal session I monitored a logfile in the /var/log
directory.  The three logfiles I monitored were /var/log/syslog,
/var/log/dmesg, and /var/log/system76.log.  using the tail -f command.  The
specific commands for each ssh session was

tail -f /var/log/syslog | tee syslog.txt

tail -f /var/log/dmesg | tee dmesg.txt

tail -f /var/log/system76.log | tee sys76.txt

With the monitor disconnected I ran the three tail -f commands in the three
terminal windows (approximate time stamp 19:59:45).  Most of the output to
the logfiles was to syslog so I will describe it next.

At 20:30:45 I connected the HDMI cable to taz.  Everything is working at
this point. syslog looks OK to me (not that I know what I am looking for)

At 20:32:43 I opened a gnome terminal on the touch screen with a mouse.  It
opens successfully and I can access the terminal on the touch screen.
syslog registers opening the terminal.

At 20:34:09 I attempt to connect the USB data cable to taz.  The cursor
disappears from the screen.  syslog shows the wacom driver (??) being
activated and a bunch of errors follow.

At 20:36:20 I try to move the terminal window with the touch screen and X
locks up, syslog registers more errors with the wacom driver.

Let me know if you need any more information

Thanks,

Erich

[   14.900416] kernel: audit: type=1400 audit(1637459846.056:61): apparmor="DENIED" operation="open" profile="/snap/snapd/13640/usr/lib/snapd/snap-confine" name="/etc/pop-os/os-release" pid=1291 comm="snap-confine" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   14.904708] kernel: audit: type=1400 audit(1637459846.060:62): apparmor="DENIED" operation="open" profile="/snap/snapd/13640/usr/lib/snapd/snap-confine" name="/etc/pop-os/os-release" pid=1291 comm="snap-confine" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   14.906518] kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlp5s0: link becomes ready
[   15.079001] kernel: audit: type=1400 audit(1637459846.236:63): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=2105 comm="apparmor_parser"
[   15.155147] kernel: bpfilter: Loaded bpfilter_umh pid 2110
[   15.155364] unknown: Started bpfilter
[   15.199721] kernel: Initializing XFRM netlink socket
[   15.550439] kernel: audit: type=1400 audit(1637459846.708:64): apparmor="DENIED" operation="open" profile="snap-update-ns.docker" name="/etc/pop-os/os-release" pid=1987 comm="5" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[   15.697771] kernel: loop22: detected capacity change from 0 to 8
[   15.840439] kernel: overlayfs: missing 'lowerdir'

Nov 20 19:59:45 taz systemd[2665]: Started Tracker metadata database store and lookup manager.
Nov 20 20:00:16 taz tracker-store[6439]: OK
Nov 20 20:00:16 taz systemd[2665]: tracker-store.service: Succeeded.
Nov 20 20:00:19 taz systemd[1]: Started Session 6 of user erich.
Nov 20 20:00:20 taz dbus-daemon[2682]: [session uid=1000 pid=2682] Activating via systemd: service name='org.freedesktop.Tracker1' unit='tracker-store.service' requested by ':1.1' (uid=1000 pid=2678 comm="/usr/libexec/tracker-miner-fs " label="unconfined")
Nov 20 20:00:20 taz systemd[2665]: Starting Tracker metadata database store and lookup manager...
Nov 20 20:00:20 taz dbus-daemon[2682]: [session uid=1000 pid=2682] Successfully activated service 'org.freedesktop.Tracker1'
Nov 20 20:00:20 taz systemd[2665]: Started Tracker metadata database store and lookup manager.
Nov 20 20:00:50 taz tracker-store[6528]: OK
Nov 20 20:00:50 taz systemd[2665]: tracker-store.service: Succeeded.
Nov 20 20:04:29 taz PackageKit: daemon quit
[... 460 lines cut ...]
Nov 20 20:38:09 taz systemd[1]: Starting Network Manager Script Dispatcher Service...
Nov 20 20:38:09 taz dbus-daemon[1167]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Nov 20 20:38:09 taz systemd[1]: Started Network Manager Script Dispatcher Service.
Nov 20 20:38:10 taz goa-daemon[2736]: secret_password_lookup_sync() returned NULL
Nov 20 20:38:11 taz goa-daemon[2736]: secret_password_lookup_sync() returned NULL
Nov 20 20:38:19 taz systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Nov 20 20:38:29 taz NetworkManager[1168]: <info>  [1637462309.8697] manager: NetworkManager state is now CONNECTED_GLOBAL
Nov 20 20:38:29 taz dbus-daemon[1167]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.15' (uid=0 pid=1168 comm="/usr/sbin/NetworkManager --no-daemon " label="unconfined")
Nov 20 20:38:29 taz systemd[1]: Starting Network Manager Script Dispatcher Service...
[... cut ...]
Nov 20 20:38:36 taz whoopsie[1894]: [20:38:36] The default IPv4 route is: /org/freedesktop/NetworkManager/ActiveConnection/1
Nov 20 20:38:36 taz whoopsie[1894]: [20:38:36] Not a paid data plan: /org/freedesktop/NetworkManager/ActiveConnection/1
Nov 20 20:38:36 taz whoopsie[1894]: [20:38:36] Found usable connection: /org/freedesktop/NetworkManager/ActiveConnection/1
Nov 20 20:38:37 taz whoopsie[1894]: [20:38:37] online
Nov 20 20:38:37 taz goa-daemon[2736]: secret_password_lookup_sync() returned NULL

jigpu, 2021-12-07:
The touchscreen device uses a Wacom device ID (056a:8191) but then later declares itself to have been manufactured by "WingCool Inc." I've never seen this before and I have to wonder if something fishy isn't going on; especially with all the trouble the X11 and kernel drivers are having with the device.

Would you mind following the instructions at https://github.com/linuxwacom/wacom-hid-descriptors/#contributing to contribute a description of this hardware to our wacom-hid-descriptors project? You'll want to have the monitor plugged in before running its sysinfo script. This should give us a much better idea of what the hardware is.

~~~
Nov 20 20:34:28 taz kernel: [ 2237.031725] usb 5-2.4: new full-speed USB device number 8 using xhci_hcd
Nov 20 20:34:28 taz kernel: [ 2237.151706] usb 5-2.4: New USB device found, idVendor=056a, idProduct=8191, bcdDevice= 2.16
Nov 20 20:34:28 taz kernel: [ 2237.151711] usb 5-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 20 20:34:28 taz kernel: [ 2237.151713] usb 5-2.4: Product: TouchScreen
Nov 20 20:34:28 taz kernel: [ 2237.151715] usb 5-2.4: Manufacturer: WingCool Inc.
Nov 20 20:34:28 taz kernel: [ 2237.151717] usb 5-2.4: SerialNumber: 9LQ0172005164
Nov 20 20:34:28 taz kernel: [ 2237.318872] input: Wacom TouchScreen Pen as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.0/0003:056A:8191.0007/input/input41
Nov 20 20:34:28 taz kernel: [ 2237.319092] input: Wacom TouchScreen Finger as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.0/0003:056A:8191.0007/input/input42
Nov 20 20:34:28 taz kernel: [ 2237.375861] wacom 0003:056A:8191.0007: hidraw3: USB HID v1.10 Keyboard [WingCool Inc. TouchScreen] on usb-0000:09:00.3-2.4/input0
Nov 20 20:34:28 taz kernel: [ 2237.382817] wacom 0003:056A:8191.0008: Unknown device_type for 'WingCool Inc. TouchScreen'. Ignoring.
Nov 20 20:34:28 taz kernel: [ 2237.390959] wacom 0003:056A:8191.0009: Unknown device_type for 'WingCool Inc. TouchScreen'. Ignoring.
~~~

If this turns out to not be a real Wacom device, I suspect the only thing we can do is to force the device to use the libinput drivers instead. Even then though, there's a high chance that things won't work right because the kernel driver would still be incorrect. I really need the data added to wacom-hid-descriptors if you can submit it...

eebender, 2021-12-08:
I'm assuming that the monitor should be connected when I run this script?
The reason I ask is that the X-server will probably freeze before I run the
script.  In that case I can run it in an ssh session remotely from my other
laptop.

jigpu, 2021-12-08:
Yes, the monitor will need to be connected prior to running the script. It will be probing the device for information (e.g. HID descriptor, kernel properties, etc.).

eebender, 2021-12-11:
I've attached the tar file you requested.

Here is the stdout from the script

***@***.***:~/stuff/tmp/monitor-test$ sudo ./sysinfo.sh
Gathering system and tablet information. This may take a few seconds.
  * General host information...
  * Kernel driver information...
  * Kernel device information...
     -
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:42/FUJ7401:00...
     -
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:42/MSFT0001:00...
     -
/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.0/0003:056A:8191.0007...
     -
/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.1/0003:056A:8191.0008...
     -
/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.2/0003:056A:8191.0009...
     - udev...
  * Unbinding devices...
  * Rebinding devices...
  * Userspace driver information...
  * Userspace device information...
  * Device display information...
  * System logs...
  * System config files...
  * Desktop configuration data...
  * Removing identifying information...
  * Tarball generation...
Finished. Data available in '/tmp/sysinfo.unuZKdtImQ.tar.gz'

Erich

eebender, 2021-12-14:
Here is the tar file you requested

[sysinfo.unuZKdtImQ.tar.gz](https://github.com/linuxwacom/input-wacom/files/7708069/sysinfo.unuZKdtImQ.tar.gz)

eebender, 2021-12-14:
testing...

eebender, 2021-12-14:
Tell me if you got the tar file - having some trouble with sending messages

eebender, 2021-12-14:
> Unfortunately, Github does not include attachments that are sent over
> email. You'll need to make a new comment using the Github interface at
> #287 <https://github.com/linuxwacom/input-wacom/issues/287> and attach
> the tarball by drag & drop onto the comment box.

Tell me if -  got the tar file on github - not sure if it got through.

eebender, 2021-12-20:
Have you been able to access the sysinfo tar file that I sent over github?

eebender, 2022-01-07:
Did you receive the tar from your script I sent on Dec. 8th?

eebender, 2022-02-09:
Is this issue still open?

jigpu, 2022-02-09:
My apologies; this slipped off my radar after the new year.

Dumping the device's HID descriptor I see it contains four application collections: pen, touchscreen, keyboard, and digitizer configuration. The pen collection is basically identical to that used by the "[Wacom One](https://www.wacom.com/en-us/products/pen-displays/wacom-one)" while the touchscreen doesn't appear to be similar to any particular device. The touchscreen report appears well defined, so I'd imagine our kernel driver could handle it without too much trouble. The crash that occurs when you touch the screen appears to be caused by some disagreement about the touch ID numbers.

Please try saving the following snippet as `/etc/X11/xorg.conf.d/90-innocn.conf` and rebooting. This will cause all of the monitor's devices to be handled by the libinput driver rather than our own. As mentioned earlier, this might have a better chance of working since the libinput driver is designed for non-Wacom hardware like this.
~~~
Section "InputClass"
        Identifier "Innocn Pen Touch Override"
        MatchUSBID "056a:8191"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection
~~~

If you still have problems, we might need to have you make a recording of the raw events being sent by the device since at that point the issue may be in the kernel driver rather than the X driver.

eebender, 2022-02-12:
Hi Jason, I figured you were probably busy for the holidays - no problem.
I tried putting the .conf file in file that you sent me.  The directory
/etc/X11/xorg.conf.d did not exist on my machine so I created it and then
put the .conf file in there as you suggested.  It didn't seem to do
anything - I still get the same behavior.  If you look at the syslog file I
attached it seems that the system is still trying to access the wacom
driver but hey what do I know about this stuff. Any suggestions?

Thanks,

Erich

Pinglinux, 2022-07-25:
As Jason pointed out, the touchscreen is not a Wacom device although it claimed with the Wacom Vendor ID. From your dmesg output, we see hid-generic tried to pick it up then passed it to Wacom driver. Both driver could not parse the HID events. There is a company called: http://www.wingcool.cn/index.php?m=content&c=index&a=lists&catid=20. Please contact their technical support: http://www.wingcool.cn/index.php?m=content&c=index&a=lists&catid=4

Nov 20 20:34:25 taz kernel: [ 2233.892643] input: **WingCool Inc. TouchScreen** Keyboard as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.0/0003:056A:8191.0004/input/input30
Nov 20 20:34:25 taz kernel: [ 2233.952127] hid-generic 0003:056A:8191.0004: input,hiddev0,hidraw3: USB HID v1.10 Keyboard [WingCool Inc. TouchScreen] on usb-0000:09:00.3-2.4/input0
Nov 20 20:34:25 taz kernel: [ 2233.958909] hid-generic 0003:056A:8191.0005: hiddev1,hidraw4: USB HID v1.10 Device [WingCool Inc. TouchScreen] on usb-0000:09:00.3-2.4/input1
Nov 20 20:34:25 taz kernel: [ 2233.966973] input: **WingCool Inc. TouchScreen** as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.2/0003:056A:8191.0006/input/input31
Nov 20 20:34:25 taz kernel: [ 2233.967176] hid-generic 0003:056A:8191.0006: input,hidraw5: USB HID v1.10 Mouse **[WingCool Inc. TouchScreen**] on usb-0000:09:00.3-2.4/input2
Nov 20 20:34:25 taz mtp-probe: checking bus 5, device 5: "/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4"
Nov 20 20:34:25 taz mtp-probe: bus: 5, device: 5 was not an MTP device
Nov 20 20:34:25 taz kernel: [ 2234.068546] usb 5-2: USB disconnect, device number 3
Nov 20 20:34:25 taz kernel: [ 2234.068551] usb 5-2.3: USB disconnect, device number 4
Nov 20 20:34:25 taz kernel: [ 2234.068698] usb 5-2.4: USB disconnect, device number 5
Nov 20 20:34:25 taz kernel: [ 2234.184169] wacom 0003:056A:8191.0004: wacom_get_report: ran out of retries (last error = -19)
Nov 20 20:34:25 taz kernel: [ 2234.184175] wacom 0003:056A:8191.0004: wacom_feature_mapping: could not get HID_DG_CONTACTMAX, defaulting to 16
Nov 20 20:34:25 taz kernel: [ 2234.184226] input: Wacom TouchScreen Pen as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.0/0003:056A:8191.0004/input/input32
Nov 20 20:34:25 taz kernel: [ 2234.184324] input: Wacom TouchScreen Finger as /devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/5-2.4:1.0/0003:056A:8191.0004/input/input33
Nov 20 20:34:25 taz kernel: [ 2234.184404] wacom 0003:056A:8191.0004: hidraw3: USB HID v1.10 Keyboard [**WingCool Inc. TouchScreen**] on usb-0000:09:00.3-2.4/input0
Nov 20 20:34:25 taz kernel: [ 2234.184577] wacom 0003:056A:8191.0005: Unknown device_type for '**WingCool Inc. TouchScreen**'. Ignoring.
Nov 20 20:34:25 taz kernel: [ 2234.288028] wacom 0003:056A:8191.0006: Unknown device_type for '**WingCool Inc. TouchScreen**'. Ignoring.
Nov 20 20:34:25 taz fido_id[7555]: 5-2.4: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/5-2/5-2.4/report_descriptor': No such file or directory
Nov 20 20:34:25 taz fido_id[7567]: usb5: Failed to open report descriptor at '/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb5/report_descriptor': No such file or directory

closed 2022-07-25 by Pinglinux