linuxwacom/input-wacom #288: Regression: Dell Active Pen PN579X + XPS 9310 2-in-1 not working anymore under linux-5.15.6.

bionade24, 2021-12-06:
Additional info will follow. Works fine under 5.15.5.

bionade24, 2021-12-07:
`libinput list-devices under Kernel 5.15.5`:
```
Device:           Video Bus
Kernel:           /dev/input/event5
Group:            1
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Power Button
Kernel:           /dev/input/event1
Group:            2
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Lid Switch
Kernel:           /dev/input/event0
Group:            3
Seat:             seat0, default
Capabilities:     switch
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Integrated_Webcam_HD: Integrate
Kernel:           /dev/input/event8
Group:            4
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
[... 180 lines cut ...]
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           PS/2 Generic Mouse
Kernel:           /dev/input/event28
Group:            12
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a
```

jigpu, 2021-12-07:
Can you provide some more information about what exactly is not working when using 5.15.6? Since you've posted the "list-devices" output from the earlier 5.15.5 kernel, I assume the device does not show up at all in 5.15.6?

I would suggest running our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) under both 5.15.5 and 5.15.6. Attach the tarball generated by each version to this issue so I can compare them for any obvious differences.

Linux 5.15.6 did introduce one bugfix to the Wacom kernel driver, but it shouldn't cause the device to not be detected if that's the issue:
~~~
$ git log --oneline v5.15.5..v5.15.6 -- drivers/hid
25bbaa3ae179 HID: magicmouse: prevent division by 0 on scroll
6341c9ccb29a HID: input: set usage type to key on keycode remap
740dd84229a5 HID: input: Fix parsing of HID_CP_CONSUMER_CONTROL fields
d048d3eb3ca7 HID: wacom: Use "Confidence" flag to prevent reporting invalid contacts
~~~

bionade24, 2021-12-07:
> Can you provide some more information about what exactly is not working when using 5.15.6? Since you've posted the "list-devices" output from the earlier 5.15.5 kernel, I assume the device does not [...]

I'm super busy currently so I didn't have time to actually go back to 5.15.6 and check. I'll ping you once I actually upgraded the kernel again and checked libinput. For now, the only thing I can tell you is that https://git.sr.ht/~sircmpwn/wev hasn't shown any input events from the pen on 5.15.6.

janicsc, 2021-12-07:
I think I have the same issue. I use a Thinkpad X1 Yoga 4th gen and a Bamboo Ink pen.

With kernel 5.15.6 the pen and touchscreen work initially, but after using the pen for about a minute, the pen and the touchscreen stop working. The timing of the failure seems to be random, but always when using the pen. After a reboot everything starts working again.

5.15.5 runs as described without any problems. I've run sysinfo.sh for 5.15.5, 5.15.6 with working pen input and for 5.15.6 after the pen stopped working.

[sysinfo_5.15.5.tar.gz](https://github.com/linuxwacom/input-wacom/files/7671174/sysinfo_5.15.5.tar.gz)
[sysinfo_5.15.6_working.tar.gz](https://github.com/linuxwacom/input-wacom/files/7671175/sysinfo_5.15.6_working.tar.gz)
[sysinfo_5.15.6_nonworking.tar.gz](https://github.com/linuxwacom/input-wacom/files/7671176/sysinfo_5.15.6_nonworking.tar.gz)

JoalpaLe, 2021-12-08:
I have the exact same issue as @janicsc with my ThinkPad X1 Yoga Gen. 6 and `kernel-5.15.6-200.fc35.x86_64` when using the included Wacom stylus. For me, in most cases the issue seems to happen when I press one of the two buttons on the stylus while moving the pen away from the screen. After that, I'm no longer able to use the pen. Touch, however, works fine even after the issue occurred.

bionade24, 2021-12-08:
> With kernel 5.15.6 the pen and touchscreen work initially, but after using the pen for about a minute, the pen and the touchscreen stop working. The timing of the failure seems to be random, but always happens when using the pen. After a reboot everything starts working again.

Can confitm that sometimes the stylus worked for me too under 5.15.6 although only small timeframes. Can confirm that's the same issue.

jigpu, 2021-12-09:
Please install the hid-recorder tool (e.g. `dnf install hid-replay` for Fedora) and perform the following steps in the 5.15.6 kernel:

1. Run `sudo hid-recorder | tee output.log`
2. Select the device with "WCOM" in its name and press Enter.
    * You should see a bunch of data printed out on the screen, and even more data whenever pen / touch events are being received by the system.
3. Begin using your pen / touch normally.
4. When the pen stops working, take note of the timestamp currently displayed in the hid-recorder output (it is the first number printed out on each line)
5. Use the pen and touch for a few more seconds.
6. Press CTRL+C to stop recording
7. Attach the generated output.log file to this issue and let me know the timestamp you noted in step 4.

I will be able to use the log file to simulate the device locally on my own machine so I begin debugging the 5.15.6 kernel locally.

janicsc, 2021-12-10:
Since I had two Wacom devices to select from, I recorded both. Device 0 is the touchscreen and device 1 is the pen-input. While recording I noticed that the failure happens, when the palm rests on the touchscreen while I'm writing or starting to write at same time

The pen first stopped working at around 114 seconds. But it started working again after a few additional strokes, sadly i didn't catch the timestamp, when it started working again, but I guess it took around 5 to 10 seconds. Touch worked the whole time.

The second time the pen stopped working is at around 165 seconds. This time touch stopped working too and it stayed broken for the rest of the time.

[output.log](https://github.com/linuxwacom/input-wacom/files/7691532/output.log)

jigpu, 2021-12-10:
Quick update: it looks like the touch sensor is getting confused and sending out strange data. This section of the recording, for example, shows a single finger being removed at t=163.988784 and then in the very next report a single finger with ID 2 (instead of the expected ID 1, though this shouldn't be a problem) coming into contact. At t=164.812796 this contact also disappears, but the sensor does not have its "Touch Valid" bit set when this happens. The result is that the kernel thinks that the finger is still in contact, leading to strange arbitration issues.

~~~
E: 163.954745 40 0c 00 01 05 01 00 62 2a 44 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 e6
E: 163.964815 40 0c 00 01 05 01 00 62 2a 44 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 84 e6
E: 163.978754 40 0c 00 01 05 01 00 62 2a 44 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e8 e6
E: 163.988784 40 0c 00 01 04 01 00 62 2a 44 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4c e7
E: 164.785293 40 0c 00 01 05 02 00 58 26 25 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1c 08
E: 164.795286 40 0c 00 01 05 02 00 58 26 25 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 08
E: 164.802821 40 0c 00 01 05 02 00 58 26 25 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e4 08
~~~

The Linux 5.15.6 kernel includes a change to how we handle the "Touch Valid" bit to fix another tablet. It seems we have a bit more thought ahead of us on how to not break when encountering this confused condition. Pinging @Joshua-Dickens since he will also be interested in this bug.

mastermax1988, 2021-12-12:
I am having the same issue with my Lenovo L13 yoga. Posted this first in the arch forum, before finding this thread: https://bbs.archlinux.org/viewtopic.php?id=272000

bionade24, 2021-12-12:
[output_bionade24.log](https://github.com/linuxwacom/input-wacom/files/7698917/output_bionade24.log)
Here is mine. `206.302273` is the 2nd time it stopped working in Xournalpp. Interestingly, touching Alacritty's window always fixes the problem. I switched between AES and MPP after the Occurance.

bionade24, 2021-12-12:
@jigpu: When hid-recorder is running and I touch the terminal where it is running in with the pen, it sudenly works again. If hid-recorder doesn't run, the only way to recover is reboot. Hope that helps.

Pinglinux, 2021-12-14:
Before digging too deep into the events/data, l wonder if the last patch is/isn't the root cause. Can one of you build and install the latest input-wacom, without the last patch, under kernel 5.15.6 to see if the issue occurs. Please follow the steps below:
 1. git clone https://github.com/linuxwacom/input-wacom
 2. cd input-wacom
 3. git reset --hard c2dbbbaf4df4267dcb46b858afff684c9be744b1
 4. follow the Prerequisites and Build / Install sections at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source to build and install the driver
 5. test as usual

mastermax1988, 2021-12-18:
I did a full system update and testet it there under kernel 5.15.10. The bug still exists. After a few seconds, the stylus stopps working.

```
Imax@yoga:~% uname -a
Linux yoga 5.15.10-arch1-1 #1 SMP PREEMPT Fri, 17 Dec 2021 11:17:37 +0000 x86_64 GNU/Linux

Imax@yoga:~% grep "" /sys/module/wacom*/version
v2.00-0.48.0.13.gc2dbbba

Imax@yoga:~% xinput
[...]
⎜   ↳ Wacom Pen and multitouch sensor Finger  	id=12	[slave  pointer  (2)]
[...]
⎜   ↳ Wacom Pen and multitouch sensor Pen Pen (0xb050e709)	id=18	[slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Pen Eraser (0xb050e709)	id=19	[slave  pointer  (2)]
[...]

Imax@yoga:~% lsusb
[...]
Bus 001 Device 005: ID 056a:51f6 Wacom Co., Ltd Pen and multitouch sensor
[...]
```

mastermax1988, 2021-12-26:
https://bbs.archlinux.org/viewtopic.php?pid=2011490#p2011490
The patched kernel mentioned in this post fixes this bug for me.

jigpu, 2022-01-04:
Thanks for the update @mastermax1988. Something may have gone wrong when testing Ping's suggestion since the package from the linked thread would have a similar effect (the author says it simply reverts the suspected commit, which is what we wanted to test).

Reverting the commit is not an ideal fix since it will break other users' devices, but something similar might have to be the bandage while I figure out something that works for all devices.

jigpu, 2022-01-12:
I've just pushed a branch which should hopefully fix this issue. I would appreciate if someone could test it and report back. You can get the branch by running `git clone https://github.com/jigpu/input-wacom.git -b fix-288` and then following the standard [build instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source#build--install) inside the downloaded input-wacom directory.

After installing / rebooting you can verify that the fixed version of the driver is currently in use by running `cat /sys/module/wacom/version` and ensuring it reports back `v2.00-0.48.0.17.g24e94f2`

tattsan, 2022-01-13:
I've built this module on linux-5.16.0-zen1-1 of Arch Linux , and it works fine so far.

Danacus, 2022-01-13:
It's working fine for me too so far. Might also have fixed some issues with touch and pen input acting weird with palm detection.

mastermax1988, 2022-01-13:
Works great on my L13 yoga running 5.16.0-arch1-1 kernel, thanks @jigpu for your effort!

enriqueartal, 2022-01-13:
It works also on my HP EliteBook x360 830 G6 with kernel 5.15.13-200.fc35.x86_64. Thanks!

XpioWolf, 2022-01-15:
Hi, I currently have a Lenovo IdeaPad (14ARE05) running FerenOS (an Ubuntu based distro with KDE plasma). I have been using the included pen for about a year now, but as the pen has worn out so I got a wacom Bamboo ink stylus. The new pen would work for a few minutes and then the computer would stop detecting it, just as the others described. I have been looking for solutions ever since I got the pen and today I found an issue (lost the issues number) that talked about kfifo, I ran dmesg and there were lot's of `"kfifo has filled, starting to drop events"` messages. I searched how to fix that and found #299 which mentioned this issue, so I cloned and built the patch and the pen worked for much longer and the are no longer and `kfifo has filled....` messages when I run dmseg but it has stopped working again (no kfifo messages now though). 

So, if anyone has any ideas how to solve this I'd love to hear your thoughts and of course run any tests you need me to. 

```Operating System: Feren OS 2021.10
KDE Plasma Version: 5.22.5
KDE Frameworks Version: 5.86.0
Qt Version: 5.15.3
Kernel Version: 5.11.0-46-generic (64-bit)
Graphics Platform: X11
Processors: 6 × AMD Ryzen 5 4500U with Radeon Graphics
Memory: 15.0 GiB of RAM
Graphics Processor: AMD RENOIR 
```

Thanks!

Danacus, 2022-01-16:
@XpioWolf Can you confirm that this is not a hardware issue? Sometimes this happens to me too and it's fixed after reseating the battery (before the new patch this would not always fix the issue). Does `hid-recorder` show anything when the pen stops working?

XpioWolf, 2022-01-17:
@Danacus Hi, I tried getting hid-recorder to work but there is some package missing and I haven't had the time to try and fix it, but I did use `xinput test` and the input stopped once the pen stopped working. Sometimes taking the battery out in putting it in again helps, but not for long. I will go and buy a new battery soon and see if it fixes the issue, but I'm not hopeful.

Danacus, 2022-01-17:
@XpioWolf as far as I know the only way to know if it's a hardware issue or a software issue is to use `hid-recorder`, because it records raw hardware events before they are sent to the kernel (if I understand correctly). If `xinput` and `libinput` stop showing messages it might also be a problem with the kernel driver.

XpioWolf, 2022-01-17:
Ok, so I'll try and get it working. Other than that I tried using the pen again, it worked very well until I noticed it was starting to glitch so I looked at `dmesg` and the were many `kfifo has filled...` entries, once I replaced the battery nothing changed and the were new `kfifo has filled...` entries (I know they were new because I cleared `dmesg`). So I'll report back later with a recording, and we'll see from there.

Thanks for taking the time and helping me, much much appreciated!!

(btw, if I need to open a new issue let me know, I just thought as I was having what seemed to be the same issue as the OP maybe I'd use this one)

b1tt1, 2022-01-17:
> I've just pushed a branch which should hopefully fix this issue. I would appreciate if someone could test it and report back. 

The branch fixed my similar stylus problems on my Thinkpad Yoga X380 running Opensuse Tumbleweed (kernel 5.16.0). Thanks!

adggit, 2022-01-18:
Different machines/pens and different LTS kernel series 5.10 (via Debian stable), but all my stylus pens started seeing the same behavior above (no response after a few minutes) starting from kernel 5.10.83, which is where the same "Confidence" flag commit was introduced.  Before this, the stylus pens had been working fine with all 5.4.x/5.10.x kernel versions that I had used. (Multiple machines: Lenovo Yoga Flex 5 14 ARE.  Pens: Lenovo Active Pen.)

Should I open a separate issue?

adggit, 2022-01-18:
Forgot to mention that in my case (kernel 5.10 series, Lenovo Flex 5 14 ARE with Active Pen) when the stylus stops working (for 5.10.83 or later), the finger touch still works fine (always).  I have downgraded to pre-5.10.82 kernel in my work machines, but have started testing the fix-288 branch in some other spare machines and will report soon.

adggit, 2022-01-18:
> I've just pushed a branch which should hopefully fix this issue. I would appreciate if someone could test it and report back.
> ...
> After installing / rebooting you can verify that the fixed version of the driver is currently in use by ... ensuring it reports back `v2.00-0.48.0.17.g24e94f2`

I built/installed the `fix-288` branch (`24e94f2`) on my Lenovo Flex 5 14 machine running kernel 5.10.84, with the issue present.  I then reloaded the `wacom` module _without rebooting_ and `modinfo wacom` showed that the `fix-288` branch module (`v2.00-0.48.0.17.g24e94f2`) got loaded.  And the problem seems to have gone away - thanks @jigpu!

goeba, 2022-01-18:
For me, fix-288 resolved the issue which seems to be the same as described here:
https://github.com/linuxwacom/input-wacom/issues/290
I use a Lenovo Yoga L380 with 5.16.0-zen1-1-zen kernel on Arch.

jigpu, 2022-01-21:
Fix has been accepted upstream. It should be appearing in the next RC release, as well as stable kernels "soon". I'll update with specific version numbers when available.

jigpu, 2022-01-24:
Queued for stable:

5.16.x, 5.15.x, 5.10.x, 5.4.x, 4.19.x:
* Reset expected and received contact counts at the same time
* Ignore the confidence flag when a touch is removed

4.14.x:
* Ignore the confidence flag when a touch is removed

inducer, 2022-01-30:
Looks like this made 5.16.4: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/?h=v5.16.4&ofs=1000

jigpu, 2022-02-02:
Patches accepted in upstream and stable:
* v5.17-rc1
* v5.16.3
* v5.15.17
* v5.10.94
* v5.4.174
* v4.19.226
* v4.14.263

closed 2022-02-02 by jigpu

lejono, 2023-01-19:
Hi, thanks for all these efforts. I also have this problem (X1 yoga, gen 5, Ubuntu 22.04 (with kernel5.15.0-58-generic) , using a Wacom Bamboo Ink-Active Stylus (2nd gen). My log is filled up with entries "wacom_wac_queue_insert: kfifo has filled, starting to drop events". 
 
I try to install the latest input-wacom-0.49 (which I presume includes the bug fix) following the instructions at  [https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source#build--install](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source#build--install) but the new modules do not load. I think this may be because during the build process I get some SSL errors? 
[wacom-input-bugfix.txt](https://github.com/linuxwacom/input-wacom/files/10452511/wacom-input-bugfix.txt). I attach the output of my build. My dmesg contains the line "Lockdown: systemd: /dev/mem,kmem,port is restricted; see man kernel_lockdown.7" but that doesn't seem to match the messages I've seen on this forum which indicate that unsigned modules will not load.

When I try to load the modules manually, I get

jono@mycomputer: ~/src/input-wacom-0.49.0$ sudo insmod 4.5/wacom.ko
insmod: ERROR: could not insert module 4.5/wacom.ko: Operation not permitted
jono@mycomputer:~/src/input-wacom-0.49.0$ find /lib/modules/$(uname -r)/extra -type f -print0 | xargs -0 md5sum | grep wacom
7c47a44d933cdb1a9b8b3d5d2c4dc957  /lib/modules/5.15.0-58-generic/extra/wacom_i2c.ko
0e4bb70a61ace49d4f7087f925460280  /lib/modules/5.15.0-58-generic/extra/wacom_w8001.ko
a2b28f71cd92c5e0ca3a304bbd4f8005  /lib/modules/5.15.0-58-generic/extra/wacom.ko