linuxwacom/input-wacom #289: Wacom Intuos S (CTL-4100 with Ubuntu 20.04: device not detected

Basdorsman, 2021-12-10:
Hello there! 

I've done the install of input-wacom 0.48.0, but after following the installation instructions of the kernel driver and rebooting, my device is still not detected. The device does work with other PCs. Is my installation failing? (I'm new to Ubuntu.)

```
bas@bas-XPS-13-9310:~$ sudo apt-get install build-essential autoconf linux-headers-$(uname -r)
[sudo] password for bas: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
autoconf is already the newest version (2.69-11.1).
build-essential is already the newest version (12.8ubuntu1.1).
linux-headers-5.10.0-1052-oem is already the newest version (5.10.0-1052.54).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
bas@bas-XPS-13-9310:~$ cd Downloads/input-wacom-0.48.0/
bas@bas-XPS-13-9310:~/Downloads/input-wacom-0.48.0$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/bas/Downloads/input-wacom-0.48.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/5.10.0-1052-oem/build
checking kernel version... 5.10.0-1052-oem
checking RHEL7 minor release... 0
checking power supply version... v4.1+
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... 
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/5.10.0-1052-oem/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/5.10.0-1052-oem/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... yes
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... yes
checking for module signing key... /var/lib/shim-signed/mok/MOK.priv
checking for module signing certificate... /var/lib/shim-signed/mok/MOK.der

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/5.10.0-1052-oem/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/bas/Downloads/input-wacom-0.48.0'
Making all in 4.5
make[2]: Entering directory '/home/bas/Downloads/input-wacom-0.48.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/5.10.0-1052-oem/build M=/home/bas/Downloads/input-wacom-0.48.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-5.10.0-1052-oem'
make[3]: Leaving directory '/usr/src/linux-headers-5.10.0-1052-oem'
make[2]: Leaving directory '/home/bas/Downloads/input-wacom-0.48.0/4.5'
make[2]: Entering directory '/home/bas/Downloads/input-wacom-0.48.0'
make[2]: Leaving directory '/home/bas/Downloads/input-wacom-0.48.0'
make[1]: Leaving directory '/home/bas/Downloads/input-wacom-0.48.0'
Making install in 4.5
make[1]: Entering directory '/home/bas/Downloads/input-wacom-0.48.0/4.5'
make -C /lib/modules/5.10.0-1052-oem/build M=/home/bas/Downloads/input-wacom-0.48.0/4.5 modules_install mod_sign_cmd='"/lib/modules/5.10.0-1052-oem/build/scripts/sign-file" "sha512" "/var/lib/shim-signed/mok/MOK.priv" "/var/lib/shim-signed/mok/MOK.der"'
make[2]: Entering directory '/usr/src/linux-headers-5.10.0-1052-oem'
  INSTALL /home/bas/Downloads/input-wacom-0.48.0/4.5/wacom.ko
  INSTALL /home/bas/Downloads/input-wacom-0.48.0/4.5/wacom_w8001.ko
  DEPMOD  5.10.0-1052-oem
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.10.0-1052-oem'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
echo "override wacom_w8001 * extra" >> /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 5.10.0-1052-oem
/usr/sbin/update-initramfs -u -k 5.10.0-1052-oem
update-initramfs: Generating /boot/initrd.img-5.10.0-1052-oem
W: Possible missing firmware /lib/firmware/i915/tgl_huc_7.5.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_huc_7.5.0.bin for module i915
make[1]: Leaving directory '/home/bas/Downloads/input-wacom-0.48.0/4.5'
make[1]: Entering directory '/home/bas/Downloads/input-wacom-0.48.0'
make[2]: Entering directory '/home/bas/Downloads/input-wacom-0.48.0'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/bas/Downloads/input-wacom-0.48.0'
make[1]: Leaving directory '/home/bas/Downloads/input-wacom-0.48.0'

```
After installation the module is not found if I search for it using grep. the troubleshooting section suggests I run:

```
sudo rmmod wacom
sudo rmmod wacom_w8001
sudo modprobe wacom
sudo modprobe wacom_w8001
```

Funnily enough, after doing this, the module IS found! But after a reboot, it is lost again.

Any help would be much appreciated!

jigpu, 2021-12-10:
These installation logs look normal enough. Please perform the following steps to produce a clean set of logs to help us debug:
1. Disconnect the tablet and reboot the system
2. Once logged back in, connect the tablet
3. Verify the tablet does not work and then run `sudo rmmod wacom` and then `sudo modprobe wacom`
4. Run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh)
5. Attach the generated tarball to this issue for review.

Following this sequence of steps should let us clearly see exactly what happens when the tablet is connected, what versions of the driver are installed, and what version of the driver is eventually loaded by modprobe.

Basdorsman, 2021-12-11:
Dear Jigpu,

It looks like your step 1 and 2: disconnect -> reboot -> reconnect, resolved the issue! The built-in LED of the tablet lights and the tablet is recognized from startup. It works now! Thank you!

closed 2021-12-11 by Basdorsman

jigpu, 2021-12-13:
Just as one final confirmation, can you let me know if the tablet still works correctly if you reboot with it connected?

Basdorsman, 2021-12-15:
Hi again! I just performed that test, and it still works if rebooting with device connected.

irodionzaytsev, 2025-04-06:
My Wacom Tablet works, shows up in the settings, but says "no tablet detected". I am also on Ubuntu 20.04. How can I make sure it shows up in the settings correctly so that I can assign the corresponding buttons etc.?