linuxwacom/input-wacom #290: Wacom in X1 Yoga events lock up (kfifo has filled)

TheJJ, 2021-12-15:
I'm running Linux 5.15.8 with libwacom 1.12, Xorg 1.20.13, xf86-input-wacom 0.40.0.

It seems that wacom input events somehow can lock up.
Right now my stylus no longer works (no event in `xev`), but touch does work perfectly.
In a previous occurence the also stylus didn't work any more, but also one touch point "froze", so touching with one finger resulted in a two-finger-gesture.

Suspend-unsuspend doesn't resolve this, `usbreset` doesn't help.

Xorg-log doesn't say anything, but in the kernel log there's this:

```
[Dec15 18:00] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.015025] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004987] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.015007] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004008] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.010994] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.009985] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.009002] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004994] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.015109] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004922] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.014999] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004980] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.005023] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004998] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.010000] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.005003] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.005029] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.004960] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[  +0.010034] wacom 0003:056A:50B8.0008: wacom_wac_queue_insert: kfifo has filled, starting to drop events
```

Any hints on how to further debug this?

jigpu, 2021-12-16:
The `kfifo has filled, starting to drop events` message will be printed if we never receive a pen ID from the device. There seem to be a number of causes for this, and some of them we don't fully understand. Two of the easiest to check are battery level and electrical interference. If your pen can be recharged, make sure it has had adequate time to be fully charged; if it takes a battery, try to find a replacement. Also try to operate the device away from sources of electrical interference (e.g. phone, wifi router, other computers, TV, etc.).

TheJJ, 2021-12-17:
It's a X1 yoga thinkpad pen pro (the small one). I don't think the cause is the charge or interference, since a reboot of the laptop did resolve the problem. I haven't checked reloading the kernel module yet.

mashuptwice, 2022-01-03:
I got the exact same issue with my Yoga 260. 
It seems the issue only occurs if I rest my palm on the display. If only the stylus is used, without any touchscreen events, I could write for hours.

mashuptwice, 2022-01-03:
Installing input-wacom-0.48.0 from source solved the issue in my case, the pen is working as expected with no lockups.

wltjr, 2022-01-11:
@mashuptwice where did you find 0.48.0? The latest tag/release is 0.40.0. I built it from the latest git master, but soon as I rest my hand on the screen, I get the error as first reported. It does not seem to have anything to do with pen charge level as mentioned in https://github.com/linuxwacom/input-wacom/issues/290#issuecomment-996216365

goeba, 2022-01-18:
I,
I have the same issue on a Lenovo Yoga L380, installed input-wacom, latest git (should I have taken 0.48.0?), the kfifo messages went away, but the issue persists (only without messages now). It definitely seems related to putting the hand on the screen and then drawing, can be reproduced both with xournal++ and styluslabs write.

I use 5.16.0-zen1-1-zen kernel on Arch.

wltjr, 2022-01-18:
Yes same here on my X1 Yoga 5th Gen. I get like a timeout now with the update, but it does not let me put my hand on the screen and write at the same time. That is interesting about older kernels, I may have to try one, I have been using 5.15.11-gentoo, with 0.48.0 wacom driver.

Thank you to all those doing the development and working on this issue!

goeba, 2022-01-19:
Hi,
fix-288 in the thread I mentioned above fixes the issue for me.

tiwai, 2022-01-19:
We've got a similar bug report for openSUSE Tumbleweed on ThinkPad X1 Yoga Gen 1:
  https://bugzilla.suse.com/show_bug.cgi?id=1194667

According to the reporter, this looks like a regression.  5.15.5 worked, and later version doesn't.  5.15.6 got the recent upstream fix (commit 7fb0413baa7f, HID: wacom: Use "Confidence" flag to prevent reporting invalid contacts), and this seems triggering the bug.

Interestingly, though, a kernel with the revert (on top of 5.16) showed the same kfifo error at some time later.  So that might not be the only cause, but the upstream fix appears bringing troubles on certain devices.

drskalman, 2022-01-20:
I can confirm that I have the same issue with X1 Yoga 5th Gen with kernel 5.15.13-arch1-1. As long as there is no touch the pen works and shortly after my hand touches the screen the stylus stops working.

wltjr, 2022-01-21:
> Hi, fix-288 in the thread I mentioned above fixes the issue for me.

I can confirm this fixes the issue for me as well. I can rest my entire arm on the screen while writing with the pen. Thank you!

jigpu, 2022-02-08:
Closing this issue since it is most likely the same issue as #288 as mentioned by others.

closed 2022-02-08 by jigpu