linuxwacom/input-wacom #291: Cannot connect using bluetooth - Intuos Pro L PTH-860

rafaelnco, 2021-12-17:
### Issue:
hello guys, I'm trying to fix a Intuos Pro L PTH-860 on Arch Linux 5.15.7 and it works flawlessly but currently I'm having this problem in which the I Cannot connect using bluetooth

### Scenario:
Device is Intuos Pro L PTH-860 on Arch Linux 5.15.7. Device battery is full, I turn it on, the touchring leds blink and then only one becomes lighted. using bluetoothctl i am able to connect to the `LE IntuosPro L` device but it is not recognized as a tablet and after a couple of seconds it auto disconnects (something like 10 seconds)  

### Packages:
```
input-wacom-dkms 0.48.0-1
kcm-wacomtablet 1:3.2.0-4
libwacom 1.12-1
xf86-input-wacom 0.40.0-2
tuhi-git 20210208-1
```

jigpu, 2021-12-17:
The Intuos Pro can be connected to over Bluetooth classic or Bluetooth LE. The Bluetooth LE connection that you're using is only for the device's "Paper Mode" and does not work as a general tablet (see the "Tuhi" program if you're actually interested in paper mode). You'll instead want to connect with Bluetooth classic to the device named "CL IntuosPro L".

To enter classic pairing mode, simply turn the tablet on while it is disconnected from USB (do _not_ hold down the center touchring button). The light next to the USB port should _not_ be blue. After a few seconds you should see the "CL IntuosPro L" device appear in the list of Bluetooth devices.

rafaelnco, 2021-12-17:
@jigpu thank you very much for feedback, i just tried your steps but the CL collection is never appearing no matter how long i wait, is there something more i need to do in order to trigger the right mode? there is no blue led near the usb cable, i just touched the power button and no other button was touched neither during powering nor after. i tried listing bluetooth connections both on computer and smartphone.

rafaelnco, 2021-12-27:
hello @jigpu ! i have been able to find the `BT IntuosPro L` you told me about! thank you very much for pointing me on that direction! right now i have the same problem in which after connecting to the tablet using bluetooth it disconnects after a couple of seconds, while connected `xsetwacom --list devices` won't list any device, leds on the tablet are turned on as normal as if it were running

would you please tell me if there are additional steps required to recognize the tablet on arch linux under bluez packages?

Pinglinux, 2022-08-12:
Hi @rafaelnco, I don't have a Intuos Pro L. I tested an Intuos Pro M. So, as Jason pointed out, "BT IntuodPro" instead of "LE IntuosPro" is what we need to pair with. I think I reproduced your connect/disconnect issue. When I connected the first time, it disconnected after a while. Cursor didn't move. Then I charged it through USB (and made sure USB worked). After it is fully charged, I paired it again. It worked without problem. Maybe you just need to charge the tablet before using it?

rafaelnco, 2022-08-13:
hey @Pinglinux thanks for your feedback! i only use this tablet on usb mode, do you think it can charge while it is being used? i  will try out the  BT InstuosPro bluetooth connection right away!

Pinglinux, 2022-08-13:
Yes, you can use the tablet while it is charging/connected through USB.

rafaelnco, 2022-08-13:
hey @Pinglinux alright! i have not been able to test it yet but ill try it soon!

Pinglinux, 2022-08-29:
@rafaelnco I am closing the issue. Feel free to open it again if it still doesn't work for you.

closed 2022-08-29 by Pinglinux