linuxwacom/input-wacom #292: Device will restart intermittently - Intuos Pro L PTH-860

rafaelnco, 2021-12-17:
### Issue:
hello guys, I'm trying to fix a Intuos Pro L PTH-860 on Arch Linux 5.15.7 and it works flawlessly but currently I'm having this problem in which the device will restart itself intermittently. Already checked all hardware and it works on Mac and Windows and I think it could be something related to my OS config? Would you guys please give me details about how to debug this?

### Scenario:
Device is Intuos Pro L PTH-860 on Arch Linux 5.15.7. Device is connected on USB cable, [currently I am not able to connect to it with bluetooth](https://github.com/linuxwacom/input-wacom/issues/291). All hardware is working without problems. All below packages are installed and all device features work flawlessly. Along the time using the tablet the Device will restart intermittently without any plausible explanation. Already checked all hardware and everything is perfect. Only thing I have noticed is sometimes the table "resets" while also by CPU profile goes super high

### Packages:
```
input-wacom-dkms 0.48.0-1
kcm-wacomtablet 1:3.2.0-4
libwacom 1.12-1
xf86-input-wacom 0.40.0-2
tuhi-git 20210208-1
```

jigpu, 2021-12-17:
Please provide more information on what you mean by "the Device will restart intermittently without any plausible explanation". What happens when the device restart? Does the hardware show any signs of restarting (e.g. lights briefly blink), or is it just software (e.g. pen briefly stops responding or settings are lost)?

Until I know more, the only suggestion I have is to run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) immediately after such a "device restart" and attach its generated tarball to this issue.

rafaelnco, 2021-12-17:
[sysinfo.kHzuFNd5ll.tar.gz](https://github.com/linuxwacom/input-wacom/files/7737977/sysinfo.kHzuFNd5ll.tar.gz)

rafaelnco, 2021-12-17:
Hello! So it just happened and here is the tarball! Thank you very much on helping me with this. For "restarting" it seems like it is really something on the software side, it will not lose any configuration though, the led will not blink as if the table would be starting up. The lights turn off and tablet is unresponsive for 2.5s minimum (sometimes up to 5s or more) then it will resume as is it as before.

rafaelnco, 2021-12-17:
just happened again, another tarball here!
[sysinfo.BxVZe6cxYt.tar.gz](https://github.com/linuxwacom/input-wacom/files/7737997/sysinfo.BxVZe6cxYt.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.whtlA1zNXJ.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738014/sysinfo.whtlA1zNXJ.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.oB6pNX1qbp.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738059/sysinfo.oB6pNX1qbp.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.7d3lVcyrgv.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738060/sysinfo.7d3lVcyrgv.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.cw53x5cqNv.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738093/sysinfo.cw53x5cqNv.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.zrsLJzHZOI.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738146/sysinfo.zrsLJzHZOI.tar.gz)

rafaelnco, 2021-12-17:
[Uploading sysinfo.YHDwcIobol.tar.gz…]()

rafaelnco, 2021-12-17:
[sysinfo.YHDwcIobol.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738208/sysinfo.YHDwcIobol.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.KYuDp9xAjP.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738211/sysinfo.KYuDp9xAjP.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.UfvBR8nJNH.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738306/sysinfo.UfvBR8nJNH.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.rP3WQisiYa.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738318/sysinfo.rP3WQisiYa.tar.gz)

rafaelnco, 2021-12-17:
[sysinfo.uLnBkMh2xB.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738331/sysinfo.uLnBkMh2xB.tar.gz)

rafaelnco, 2021-12-18:
[sysinfo.jB6an1hb69.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738441/sysinfo.jB6an1hb69.tar.gz)
[sysinfo.vLjY4Bp6W4.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738442/sysinfo.vLjY4Bp6W4.tar.gz)
[sysinfo.8axmSflNru.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738443/sysinfo.8axmSflNru.tar.gz)
[sysinfo.kB0B35EGnh.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738444/sysinfo.kB0B35EGnh.tar.gz)
[sysinfo.E1oZI1Uz2J.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738445/sysinfo.E1oZI1Uz2J.tar.gz)
[sysinfo.tsDW5F8bH3.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738446/sysinfo.tsDW5F8bH3.tar.gz)
[sysinfo.c1NF8pmpnG.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738447/sysinfo.c1NF8pmpnG.tar.gz)
[sysinfo.ofy3IABAmg.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738448/sysinfo.ofy3IABAmg.tar.gz)
[sysinfo.0s2q0HvZhW.tar.gz](https://github.com/linuxwacom/input-wacom/files/7738449/sysinfo.0s2q0HvZhW.tar.gz)

rafaelnco, 2021-12-18:
[sysinfo.yHqOfXSaqY.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739092/sysinfo.yHqOfXSaqY.tar.gz)
[sysinfo.gIM8OTy0u2.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739093/sysinfo.gIM8OTy0u2.tar.gz)
[sysinfo.J2y652gf8A.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739094/sysinfo.J2y652gf8A.tar.gz)
[sysinfo.8ZzgGV0bV8.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739095/sysinfo.8ZzgGV0bV8.tar.gz)
[sysinfo.ZNnCvlIx5n.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739096/sysinfo.ZNnCvlIx5n.tar.gz)
[sysinfo.bhPqZ2ygcX.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739097/sysinfo.bhPqZ2ygcX.tar.gz)
[sysinfo.6KScEM3Gef.tar.gz](https://github.com/linuxwacom/input-wacom/files/7739098/sysinfo.6KScEM3Gef.tar.gz)

rafaelnco, 2021-12-19:
[sysinfo.5SlJEh2xMu.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740962/sysinfo.5SlJEh2xMu.tar.gz)
[sysinfo.HaPxCTEGZi.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740963/sysinfo.HaPxCTEGZi.tar.gz)
[sysinfo.j2ofNf7mJK.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740964/sysinfo.j2ofNf7mJK.tar.gz)
[sysinfo.JRGTNOXgPH.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740965/sysinfo.JRGTNOXgPH.tar.gz)
[sysinfo.KKxj5tQ0iz.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740966/sysinfo.KKxj5tQ0iz.tar.gz)
[sysinfo.KLoLozihRP.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740967/sysinfo.KLoLozihRP.tar.gz)
[sysinfo.LzPrQymjjE.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740968/sysinfo.LzPrQymjjE.tar.gz)
[sysinfo.mm3IEFvwAD.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740969/sysinfo.mm3IEFvwAD.tar.gz)
[sysinfo.NtbLwnX9G3.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740970/sysinfo.NtbLwnX9G3.tar.gz)
[sysinfo.pq1AScSR5K.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740971/sysinfo.pq1AScSR5K.tar.gz)
[sysinfo.S5IPTVqRRr.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740972/sysinfo.S5IPTVqRRr.tar.gz)
[sysinfo.sboU8Gos4f.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740973/sysinfo.sboU8Gos4f.tar.gz)
[sysinfo.Sd2wC595GX.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740974/sysinfo.Sd2wC595GX.tar.gz)
[sysinfo.uOyHBPUliq.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740975/sysinfo.uOyHBPUliq.tar.gz)
[sysinfo.UvCg0uzPBH.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740976/sysinfo.UvCg0uzPBH.tar.gz)
[sysinfo.xyoycSJa68.tar.gz](https://github.com/linuxwacom/input-wacom/files/7740977/sysinfo.xyoycSJa68.tar.gz)

rafaelnco, 2021-12-22:
guys please help me, the problem is getting worse, the hangs sometimes takes 10 seconds, i have already tried downgrading the versions of dependencies to libwacom-0.33-1-x86_64 and  xf86-input-wacom-0.37.0-1-x86_64 but still no luck

also i have noticed the table still shows  up on lsusb and xsetwacom --list devices while hanging

also i have cloned the input-wacom driver and installed it locally modifiying the supports_autosuspend to 0 but still no luck, if anyone could point out the right repo/file to look for i would really appreciate, thankk you very much!!!

rafaelnco, 2021-12-22:
i have just restarted my computer and the hangs become low in frequency as if the problem gets worse along hours of the driver running

rafaelnco, 2021-12-26:
[sysinfo.0KjvGWswRL.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777200/sysinfo.0KjvGWswRL.tar.gz)
[sysinfo.olGEceeUxv.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777201/sysinfo.olGEceeUxv.tar.gz)
[sysinfo.dzRM3ytjIw.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777202/sysinfo.dzRM3ytjIw.tar.gz)
[sysinfo.TSRKALln3I.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777203/sysinfo.TSRKALln3I.tar.gz)
[sysinfo.AleMSnIwvR.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777204/sysinfo.AleMSnIwvR.tar.gz)
[sysinfo.8PPrEvhLlk.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777205/sysinfo.8PPrEvhLlk.tar.gz)

rafaelnco, 2021-12-26:
[sysinfo.JceqNkMFey.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777373/sysinfo.JceqNkMFey.tar.gz)
[sysinfo.CsTumWzTxC.tar.gz](https://github.com/linuxwacom/input-wacom/files/7777374/sysinfo.CsTumWzTxC.tar.gz)

Pinglinux, 2022-08-12:
From your dmesg, I see hid-generic driver picked up the tablet before wacom driver took it over. I can't tell why hid-generic made that decision. For end users, that action could be translated as "device restarted". Are you willing to try another kernel or to upgrade your Linux?

`[    1.892257] hid-generic 0003:056A:0358.0002: input,hiddev96,hidraw1: USB HID v1.10 Mouse [Wacom Co.,Ltd. Wacom Intuos Pro L] on usb-0000:00:1d.0-1.2/input0`
`[    1.896801] hid-generic 0003:056A:0358.0003: hiddev97,hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom Intuos Pro L] on usb-0000:00:1d.0-1.2/input2`
`[    4.070586] input: Wacom Intuos Pro L Pen as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:056A:0358.0002/input/input19`
`[    4.070773] input: Wacom Intuos Pro L Pad as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:056A:0358.0002/input/input21`

rafaelnco, 2022-08-13:
hey @Pinglinux thank you for your feedback! i made sure i was using latest versions of everything! maybe i have found out the real issue behind this problem and it could be that i was using an old notebook that doesnt support usb 3.0 and i think actually the device was being restarted because of high power usage

rafaelnco, 2022-08-13:
i am now using the same tablet on a newer notebook which has usb 3.0 and it is working perfectly! the only issue now is i cannot use Proximity>33 or the cursor will jump on relatve mode! i have opened an issue for that on another thread and would be willing to help solving :)

Pinglinux, 2022-08-13:
Ok, that's good news. I'll close this issue. I need to understand what you mean by "cannot use Proximity>33 or the cursor will jump on relative mode" in the other issue.

closed 2022-08-13 by Pinglinux

rafaelnco, 2022-08-13:
hey @Pinglinux alright! i have detailed this problem in another issue but right now i am unable to find the link, ill ping when i find it!