linuxwacom/input-wacom #293: Jumpy cursor with Relative mode - Intuos Pro L PTH-860

rafaelnco, 2021-12-17:
### Issue:
hello guys, I'm trying to fix a Intuos Pro L PTH-860 on Arch Linux 5.15.7 and it works flawlessly but currently I'm having this problem in which the device will have a Jumpy cursor with Relative mode, I am able to circumvent the problem by setting CursorProximity to half it's maximum value but I would like to ask if [is there any chance using those patches to create a solution for my scenario?](https://github.com/linuxwacom/input-wacom/issues/54)

### Scenario:
Device is Intuos Pro L PTH-860 on Arch Linux 5.15.7. Device is connected on USB cable, [currently I am not able to connect to it with bluetooth](https://github.com/linuxwacom/input-wacom/issues/291). All hardware is working without problems. All below packages are installed and all device features work flawlessly. Along the time using the tablet the Device will have a Jumpy cursor with Relative mode. I am able to circumvent the problem by setting CursorProximity to half it's maximum value. 

### Reproduction:
It may occur more often if the following steps are done by tapping the stylus on the tablet. Basically just jump the stylus on the tablet letting the tip enter it's energy field rapidly  so that the tablet interprets it as a cursor movement. Do this movements varying the distance between each gesture.

### Packages:
```
input-wacom-dkms 0.48.0-1
kcm-wacomtablet 1:3.2.0-4
libwacom 1.12-1
xf86-input-wacom 0.40.0-2
tuhi-git 20210208-1
```

jigpu, 2021-12-17:
I suggest continuing to reduce CursorProximity until you reach a point where the jumps do not occur.

The jumps are caused by moving the pen inside of a zone where it is both close enough that the tablet detects it but too far for us to receive any position updates. If the pen is in relative mode and lifted into this zone and then placed back down some distance away, the driver sees this as a sudden large movement since it was unable to track the pen while inside the zone.

The CursorProximity setting adjusts how far away a pen can be before we ignore the tablet's own proximity bit and no longer consider it active. You need to set it low enough that the zone describe above does not exist. That is, there is *no* point where we consider the pen active and yet is too far to receive position updates. Unfortunately, the ideal value of CursorProximity will vary between devices, so we can't just set it to a static value in the driver. You'll need to tune it by hand to reduce the jumps without making the hover height so low that you can't move the cursor at all.

closed 2021-12-17 by jigpu

rafaelnco, 2021-12-17:
hey @jigpu ! thank you for your attention! i am currently trying to implement your method and it really works however do you think there would be any chance for me to open a PR with some customization of the algorithm that could help resolve this issue? i love wacom and thats why i`m opening this issues and would really love to help building a smoother experience for everyone

pepper-jelly, 2021-12-23:
Some tablets don't have this broken behavior. But for those that do either adjust CursorProximity with xsetwacom or patch function similar to this:
``` patch
--- a/wacom_wac.c
+++ b/wacom_wac.c
@@ -2987,8 +2987,8 @@
 	if (data[0] != WACOM_REPORT_PENABLED)
 	    return 0;
 
-	range = (data[1] & 0x80) == 0x80;
 	prox = (data[1] & 0x40) == 0x40;
+	range = prox;
 	rdy = (data[1] & 0x20) == 0x20;
 
 	wacom->shared->stylus_in_proximity = range;
```

rafaelnco, 2021-12-27:
hey @pepper-jelly you are AWESOME thank you very much, gonna try that tomorrow! man i really need your help please, are you available just hinting me in the right path to solve an issue with a PTH-860 on linux? #292 the problem is it keeps usb resetting a lot really and i am afraid there is some hardware issue with the tablet, i`m desperate man

rafaelnco, 2021-12-29:
hello @pepper-jelly ! merry christmas! may i ask for your help again? i have just tried your snippet but it seems like it does not resolve the issue here, i have assured the custom driver is installed and loaded and rebooted the machine

pepper-jelly, 2021-12-30:
@rafaelnco happy holidays, that snippet was for tablet I don't use anymore as example. For intuos its on line 1293 except prepended with bool. If that won't work your only option lowering cursor proximity.

rafaelnco, 2022-03-29:
hey @pepper-jelly thank you very much for feedback! sorry for the delay, i am going to try the 1293 line config!