linuxwacom/input-wacom #294: CTL-472 unresponsive except after reboot from Windows

liamtruter, 2021-12-17:
Hello. I am experiencing a very unusual issue with my Wacom One CTL-472 tablet.

If I start up the machine in Linux, the tablet shows up correctly using ```xsetwacom --list```. However the tablet is unable to move the cursor and the buttons don't respond.

If I start up the machine in Windows, the tablet works as expected. If I reboot into Linux after booting into Windows, it continues to work as expected until it is disconnected and reconnected.

I did an HID recording of the device working [normally](https://github.com/linuxwacom/input-wacom/files/7732679/hid_rec_responsive.txt) and of the device [not responding](https://github.com/linuxwacom/input-wacom/files/7732689/hid_rec_unresponsive.txt). When it is working normally, the HID packets are all 10 bytes. When it is not responding the HID packets are 10 bytes at idle and 4 bytes while using the pen.

I have been searching for several hours for a solution, but nothing has worked for me so far. Let me know if there is any other information I should attach. Any help would be greatly appreciated.

jigpu, 2021-12-17:
The recordings you've provided look like those from #242 when it stops responding. We haven't yet been able to determine how / why the tablet gets into this "unresponsive" mode. Can you run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) in both the "normal" and "not responding" cases and attach the generated tarballs here?

liamtruter, 2021-12-18:
Here are my sysinfo tarballs:
[unresponsive](https://github.com/linuxwacom/input-wacom/files/7738654/sysinfo_unresponsive.tar.gz)
[working](https://github.com/linuxwacom/input-wacom/files/7738655/sysinfo_working.tar.gz)

liamtruter, 2022-01-03:
Update:
I did nothing, but it now seems to be working fine. No idea what fixed it, but I'll just close this issue for now and open another if something breaks again.

liamtruter, 2022-01-06:
Unfortunately it was too good to last and it is now back to its previous behavior. I am going to try using this tablet on other systems to see if it works on other hardware configurations. I'll update if I find out anything that could be useful to others.

jigpu, 2022-01-07:
There seem to be some curious differences in how the device responds when it is first discovered on the USB bus. Specifically, it seems that in the "unresponsive" case the tablet is able to report back its "product" string (i.e. "CTL-472") but not in the "working" case. The missing product string occurs both in the dmesg output as well as the verbose lsusb output.

The next difference is that in the "unresponsive" case, once our our driver loaded we're unable to command the tablet into its normal operating mode: `wacom 0003:056A:037A.0008: wacom_set_report: ran out of retries (last error = -110)` Being in the wrong mode would explain why we're getting weird data, but then the question becomes why the command fails. The error number "-110" is a timeout, usually indicating the device doesn't like the request. It could be some kind of firmware issue given the difference in behavior when first discovering the device...

Given the similarity to #242 (especially with the device "sometimes" working properly even without rebooting from Windows) I'm going to close this issue and have you start reporting further information over there. I'll include a copy of these findings over there for reference as well.

closed 2022-01-07 by jigpu