linuxwacom/input-wacom #296: Is it possible to use Expresskey Remote?

Scheherazade, 2021-12-28:
I'm using a Wacom Cintiq Pro 24 on Linux Mint. Most things are working fine but I can't configure any of the buttons on the Expresskey Remote. I've seen videos of people using the Graphics Tablet interface to make custom keybindings for other tablets but I don't have the 'Map Buttons' option. I ran libwacom-list-local-devices and it looks like it recognises the Expresskey remote. The remote pairs with the dongle just fine, and I've tried unpairing and pairing with it again. I checked the kernel and I'm pretty sure that's right too.

Am I doing something wrong or is what I'm trying to do not possible? I am very new to Linux.

jigpu, 2022-01-07:
I don't know if the Cinnamon desktop used by Linux Mint supports configuration of the ExpressKey remote. Its Wacom control panel is based off of a very old GNOME control panel that I don't believe supported the EKR. I would suggest either trying to configure the EKR with the `xsetwacom` comand or exploring other desktop environments. The modern GNOME environment should support the EKR, but I don't know about any others. A third option could be to see if the [wacom-gui](https://github.com/tb2097/wacom-gui) tool is able to configure the EKR or not.

Scheherazade, 2022-01-09:
Thanks for your reply. Xsetwacom worked and I even ended up writing my first shell script.

For anyone as new as me, I followed these articles: 
How to use xsetwacom
https://www.linux-magazine.com/Issues/2018/212/xsetwacom
https://www.systutorials.com/docs/linux/man/1-xsetwacom/
I found the name of the ExpressKey Remote from the `--list devices` command and then the list of buttons from the command  `xsetwacom -s --get 'DEVICE-NAME (ie name of expresskey remote)' all`. Then it was a case of copying from the list of buttons and changing the `"button +1"` etc to `key +ctrl +z` for example.
As the article notes, the button configurations don't save when you reboot your computer so you have to write a shell script and run it each time you boot up.

How to write a shell script
https://www.linuxjournal.com/content/how-create-shell-script-linux

I pretty much just put `#! /bin/bash` and then the button configurations from xsetwacom into the Text Editor, saved it as a .sh file and then whenever I boot up Linux Mint I double-click on that script to run it. I think there's a way to make it run automatically on startup as well but I haven't looked into that.

I did find I had to configure the Touch Ring as well to make the zoom work so I ended up changing AbsWUp to "key +" and AbsWDown to "key -".

jigpu, 2022-01-11:
Glad to hear that you were able to get xsetwacom working for you! :) Every desktop seems to have its own way to launch things at startup; I imagine it shouldn't be too difficult to find out how its done in Cinnamon. E.g., you might find some startup settings in its control panel.

closed 2022-01-11 by jigpu