linuxwacom/input-wacom #297: Thinkpad X201 Tablet Stylus works, but not touch

danjboyd, 2021-12-30:
I just put a fresh install of Debian 11 Bullseye on my X201 tablet. The stylus works great, but touch doesn't work at all. I'm not sure which debug output would be most helpful, so I'm not including any, but happy to respond to any comments if there is some command I can run with useful debug output.

jigpu, 2022-01-04:
Can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach its output to this issue? It will provide us with more information about your sensor and OS.

danjboyd, 2022-01-04:
[sysinfo.bs9rnQde4R.tar.gz](https://github.com/linuxwacom/input-wacom/files/7808706/sysinfo.bs9rnQde4R.tar.gz)

Output attached.

danjboyd, 2022-01-10:
Did some additional troubleshooting. If I run 
> sudo cat /dev/input/event14
I get some output when I use my stylus

But i get nothing from the touch interface:
> sudo cat /dev/input/event15

Similarly, if I run the libinput debug-gui, I get no response from the touch screen.

Pinglinux, 2022-07-05:
It's a serial tablet. Both pen and touch are recognized by the driver:

[37343.715125] input: Wacom Serial Penabled 1FG Touchscreen Pen as /devices/pnp0/00:05/tty/ttyS0/serio6/input/input21
[37343.715439] input: Wacom Serial Penabled 1FG Touchscreen Finger as /devices/pnp0/00:05/tty/ttyS0/serio6/input/input22

If you don't get any touch data, it is probably not a driver issue. Thinkpad X 201 is more than 15 years old, I guess...

jigpu, 2022-07-15:
Try running `sudo hexdump -C /dev/ttyS0` and touching the screen. This should print out the raw data being sent by the hardware to the driver. If there are no events printed out while touching the screen, then the problem is with the hardware or its configuration. Otherwise, it implies a problem with the driver not processing the data for some reason.

Pinglinux, 2022-08-03:
Not a Wacom kernel driver issue

closed 2022-08-03 by Pinglinux