linuxwacom/input-wacom #298: Wacom Bamboo Ink Plus: Bluetooth connection established, but button doesn't work

sebastian-387, 2021-12-31:
Hello dear developers,

first of all, thank you for supporting Wacom on Linux! :)

I have just bought a Bamboo Ink Plus, which otherwise works out of the box with my Ubuntu 20.04 LTS (latest development version).

After the pen is put into Bluetooth pairing mode by pressing and holding the top button for 6 sec., I can successfully see and pair the device. Furthermore, after initial pairing is done, the Bluetooth connection is successfully activated and deactivated each time I press the top button - so no problems until here!

Sadly, when I try to assign a custom shortcut to the pen's button (by going to GNOME Settings -> Keyboard -> View and Customize Shortcuts), no matter what I try, the pressing of the button is simply not recognized.

I'm running Linux kernel 5.13.0-19-generic, and GNOME 41.1 on Wayland. My convertible Laptop uses MPP 2.0. I can use the pen and it's two regular buttons just fine, only the top Bluetooth button refuses to work, sadly.

Since I'm a teacher, I would love to use my new Wacom pen for navigating through my presentations in LibreOffice. A fix would therefore be very much appreciated. :) 

Thanks and have a happy new year!
Sebastian

jigpu, 2022-01-07:
Happy new year to you as well! Unfortunately there hasn't been much traction on getting this particular feature to work. These are the issues that I'm aware of related to the problem:
* https://github.com/linuxwacom/input-wacom/issues/49
* https://gitlab.gnome.org/GNOME/mutter/-/issues/432
* https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/638

Part of the problem is that some of the keyboard shortcuts that get sent by the Bluetooth button interfere with other shortcuts that had to be shoe-horned to work with X11. It might be possible to produce yet-another workaround on top of the current mess, but everyone is understandably hesitant to dive into the muck to get it working. It can't hurt to comment on the issues to show that there's interest, but I sadly cannot provide much hope :(

sebastian-387, 2022-01-09:
If only the pen's button were recognized as XF86AudioMicMute! (as my last cheap China pen did)
Then I could simply re-map it via "key-mapper" under Wayland.

Instead, when I press the pen's button, it creates a Bluetooth connection to my laptop, and then immediatelly disables it again. As far as I can detect using GNOME Settings, no key or key combo is sent.

Any help in this matter would be really appreciated.

Pinglinux, 2022-07-15:
As described by RussianNeuroMancer at https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/638, the BT button is claimed to a one key keyboard, which isn't associated with a specific pen/tablet. The functionality of the button is beyond this project's scope.

closed 2022-07-15 by Pinglinux