linuxwacom/input-wacom #299: Pen input stops working randomly

Danacus, 2022-01-02:
After using the pen for a while, it would occasionally stop working. `libinput` and the kernel stop sending events for pen input (`evemu-record` shows no events from pen input), while finger input keeps working. However, hardware events are still being sent to the kernel when I capture a HID recording.

[recording.log.gz](https://github.com/linuxwacom/input-wacom/files/7798905/recording.log.gz)

It will start working again when I reboot the device.

I'm not sure if this is related, but I did notice a lot of "kfifo has filled, starting to drop events" messages.

Device info: https://github.com/linuxwacom/wacom-hid-descriptors/tree/master/Lenovo%20YOGA%20520-14IKB

Let me know if there is anything I can do to help debugging this problem.

Danacus, 2022-01-02:
`rmmod wacom` and `modprobe wacom` also seems to fix the issue without having to reboot. I've also switched to the latest git version of the module. Maybe that will fix it?

enriqueartal, 2022-01-03:
It is also a temporary firx for me; it may happen to stop again, but repeating the orders works

jigpu, 2022-01-04:
This could potentially be a duplicate of https://github.com/linuxwacom/input-wacom/issues/288 -- does the issue occur if you are careful to never touch the screen?

Danacus, 2022-01-04:
I'm not sure if it's the same issue. I can't test if it occurs if I never touch the screen, because I can't consistently reproduce the issue. It can work well for hours or days and then suddenly stop working until I reload the kernel module.

I might try reverting c2dbbbaf4df4267dcb46b858afff684c9be744b1 as others have proposed and see if that fixes the problem.

Danacus, 2022-01-07:
It might be too early to tell, but so far I haven't had any issues since I reverted that commit. This is probably a duplicate of #288, but with less severe problems.

jigpu, 2022-01-07:
Good to know. I'll close this issue for now, but please feel free to re-open it if the problem re-appears even with your revert of c2dbbba in place.

closed 2022-01-07 by jigpu