linuxwacom/input-wacom #300: Wacom cintiq not recognised by Ubuntu 20.04.3

13arry, 2022-01-05:
Since I upgraded from Ubuntu 18.?? to version 20.04 my Wacom Cintiq is not recognised when booting. I found a workaround by switching the screen off and back on again to get it to work. Having just built a new installation with Ubuntu 20.04.3 this no longer works. The screen display does not come back on at all and remains blank. I followed the instructions on GitHub to install the kernel and driver but when I run the rmmod I get errors. Below is a copy of the terminal i/o. Any suggestions???
barry@double13.uk

barry@Terrabox:~$ modinfo wacom | grep version
version:        v2.00-0.48.0
srcversion:     9069AD5A48C1B017DE804CA
vermagic:       5.11.0-44-generic SMP mod_unload modversions
barry@Terrabox:~$ sudo sh -c "apt-get update && apt-get install xserver-xorg-input-wacom$(dpkg -S$(which Xorg) | grep -Eo -- "-hwe-[^:]*")"
dpkg: error: unknown option -/

Type dpkg --help for help about installing and uninstalling packages [*];
Use 'apt' or 'aptitude' for user-friendly package management;
Type dpkg -Dhelp for a list of dpkg debug flag values;
Type dpkg --force-help for a list of forcing options;
Type dpkg-deb --help for help about manipulating *.deb files;

Options marked [*] produce a lot of output - pipe it through 'less' or 'more' !
[sudo] password for barry:
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Hit:2 http://gb.archive.ubuntu.com/ubuntu focal InRelease                                                   
Get:3 http://gb.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                                  
Hit:4 http://archive.canonical.com/ubuntu focal InRelease                                                              
Fetched 228 kB in 0s (500 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree      
Reading state information... Done
xserver-xorg-input-wacom is already the newest version (1:0.39.0-0ubuntu1).
xserver-xorg-input-wacom set to manually installed.
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
barry@Terrabox:~$ sudo rmmod wacom
rmmod: ERROR: Module wacom is not currently loaded
barry@Terrabox:~$ sudo rmmod wacom_W8001
rmmod: ERROR: Module wacom_W8001 is not currently loaded
barry@Terrabox:~$ sudo modprobe wacom
modprobe: ERROR: could not insert 'wacom': Operation not permitted
barry@Terrabox:~$ sudo modprobe wacom_8001
modprobe: FATAL: Module wacom_8001 not found in directory /lib/modules/5.11.0-44-generic

13arry, 2022-01-07:
Update The driver wacom.ko is created and searching with files is located in /usr/lib/modules/5.11.0-44-generic (the current update of ubuntu 20.04.3) but rmmod shows wacom is not loaded and modprobe reports insertion of wacom is not permitted - see terminal output below

barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ modinfo wacom | grep version
version:        v2.00-0.48.0
srcversion:     9069AD5A48C1B017DE804CA
vermagic:       5.11.0-44-generic SMP mod_unload modversions 
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo rmmod wacom
[sudo] password for barry: 
rmmod: ERROR: Module wacom is not currently loaded
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo rmmod wacom_w8001
rmmod: ERROR: Module wacom_w8001 is not currently loaded
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo modprobe wacom
modprobe: ERROR: could not insert 'wacom': Operation not permitted
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo modprobe wacom_w8001
modprobe: ERROR: could not insert 'wacom_w8001': Operation not permitted
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$

The latest version of the X-driver is installed

barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo sh -c "apt-get update && apt-get install xserver-xorg-input-wacom$(dpkg -S $(which Xorg) | grep -Eo -- "-hwe-[^:]*")"
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Hit:2 http://archive.canonical.com/ubuntu focal InRelease                     
Hit:3 http://gb.archive.ubuntu.com/ubuntu focal InRelease                     
Get:4 http://gb.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Fetched 228 kB in 0s (598 kB/s)                                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
xserver-xorg-input-wacom is already the newest version (1:0.39.0-0ubuntu1).
0 to upgrade, 0 to newly install, 0 to remove and 10 not to upgrade.
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$

build-essential is the latest version, but libwacom seems to be a problem - 

barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo apt-get install build-essential
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.8ubuntu1.1).
0 to upgrade, 0 to newly install, 0 to remove and 10 not to upgrade.
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$ sudo apt-get build-dep libwacom
Reading package lists... Done
E: You must put some 'deb-src' URIs in your sources.list
barry@Terrabox:~/Downloads/wacom/input-wacom-0.48.0$

I'm not sure this is relevant. libwacom is listed in many places throughout my files - in /usr/share and several snap applications. My wacom device Cintiq 25HD is listed in my settings.
My guess is that the wacom driver is not being installed. How can I fix this please.

13arry, 2022-01-07:
Problem solved (ish - I'm no expert)
The issue is caused by secure boot. There is a solution for a similar problem at the link below - it worked for me too.
https://askubuntu.com/questions/1114867/operation-not-permitted-when-trying-to-modprobe-xpad
With secure boot switched off I ran modprobe successfully. The wacom is not activated at boot, but is if I switch the monitor off and on again. Not perfect but good enough.
A word of caution - 
I have left my secure boot switched off - not sure if this is sensible but it is a private computer that only I have access to, so I'm okay with the (completely unquantified) risk

closed 2022-01-07 by 13arry

jigpu, 2022-01-07:
Glad you were able to find the source of the modprobe issue. Note that if you (or anyone else reading this thread in the future) wants to use the input-wacom driver with secure boot enabled, we have instructions on our [Secure Boot wiki article](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot) that explain how to set up the necessary signing keys.

As for the issue with the device not activating at boot unless you power it off and on again, that's been reported as issue #104. Unfortunately we've never been able to isolate the cause of the issue to fix it :frowning: The USB connection needs to be reset after boot, either by powering off/on, unplugging/replugging, or poking the kernel with a script.