linuxwacom/input-wacom #305: Lenovo Ideapad Flex 5 - unable to fget touchscreen to work

jasonday, 2022-02-05:
Unable to get the Wacom touchscreen on my Lenovo Ideapad Flex 5 to work. 
Kernel: 5.13.0-28 bits: 64 compiler: gcc v: 9.3.0 Desktop: Cinnamon 5.2. 

I've tried a bunch of things at the suggestion of the Linuxmint forums: https://forums.linuxmint.com/viewtopic.php?f=49&t=367244&p=2134570#p2134570

And I just installed the 0.48.0 kernel driver with no change.

```
$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ELAN0650:01 04F3:304B Mouse             	id=11	[slave  pointer  (2)]
⎜   ↳ ELAN0650:01 04F3:304B Touchpad          	id=12	[slave  pointer  (2)]
⎜   ↳ Wacom HID 50E9 Pen stylus               	id=13	[slave  pointer  (2)]
⎜   ↳ Wacom HID 50E9 Finger touch             	id=14	[slave  pointer  (2)]
⎜   ↳ Wacom HID 50E9 Pen eraser               	id=17	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ EasyCamera: EasyCamera                  	id=10	[slave  keyboard (3)]
    ↳ Ideapad extra buttons                   	id=15	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=16	[slave  keyboard (3)]

```

```
cat /var/log/Xorg.0.log
...
[    40.480] (II) config/udev: Adding input device Wacom HID 50E9 Finger (/dev/input/event7)
[    40.480] (**) Wacom HID 50E9 Finger: Applying InputClass "libinput touchscreen catchall"
[    40.480] (**) Wacom HID 50E9 Finger: Applying InputClass "Wacom USB touchscreen class"
[    40.480] (**) Wacom HID 50E9 Finger: Applying InputClass "Wacom touchscreen class"
[    40.480] (II) Using input driver 'wacom' for 'Wacom HID 50E9 Finger'
[    40.480] (**) Wacom HID 50E9 Finger: always reports core events
[    40.480] (**) Option "Device" "/dev/input/event7"
[    40.480] (EE) Wacom HID 50E9 Finger: Invalid type 'stylus' for this device.
[    40.480] (EE) Wacom HID 50E9 Finger: Invalid type 'eraser' for this device.
[    40.480] (EE) Wacom HID 50E9 Finger: Invalid type 'cursor' for this device.
[    40.480] (II) Wacom HID 50E9 Finger: type not specified, assuming 'touch'.
[    40.480] (II) Wacom HID 50E9 Finger: other types will be automatically added.
[    40.481] (**) Wacom HID 50E9 Finger touch: panscroll is 0
[    40.481] (**) Wacom HID 50E9 Finger touch: panscroll modified to 13
[    40.481] (--) Wacom HID 50E9 Finger touch: maxX=13820 maxY=7772 maxZ=0 resX=40000 resY=40000 
[    40.481] (II) Wacom HID 50E9 Finger touch: hotplugging dependent devices.
[    40.481] (EE) Wacom HID 50E9 Finger touch: Invalid type 'stylus' for this device.
[    40.481] (EE) Wacom HID 50E9 Finger touch: Invalid type 'eraser' for this device.
[    40.481] (EE) Wacom HID 50E9 Finger touch: Invalid type 'cursor' for this device.
[    40.481] (EE) Wacom HID 50E9 Finger touch: Invalid type 'pad' for this device.
[    40.481] (II) Wacom HID 50E9 Finger touch: hotplugging completed.
[    40.518] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-WCOM50E9:00/0018:056A:50E9.0002/input/input12/event7"
[    40.518] (II) XINPUT: Adding extended input device "Wacom HID 50E9 Finger touch" (type: TOUCH, id 14)
[    40.519] (**) Wacom HID 50E9 Finger touch: (accel) keeping acceleration scheme 1
[    40.519] (**) Wacom HID 50E9 Finger touch: (accel) acceleration profile 0
[    40.519] (**) Wacom HID 50E9 Finger touch: (accel) acceleration factor: 2.000
[    40.519] (**) Wacom HID 50E9 Finger touch: (accel) acceleration threshold: 4
[    40.521] (II) config/udev: Adding input device Wacom HID 50E9 Finger (/dev/input/mouse3)
[    40.522] (II) No input driver specified, ignoring this device.
[    40.522] (II) This device may have been added with another device file.

```
```
$ libinput list-devices

Device:           Power Button
...

Device:           Video Bus
...

Device:           Lid Switch
...

Device:           Video Bus
...
Device:           Power Button
...
Device:           EasyCamera: EasyCamera
...
Device:           ELAN0650:01 04F3:304B Mouse
Kernel:           /dev/input/event4
Group:            6
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   *button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

Device:           ELAN0650:01 04F3:304B Touchpad
Kernel:           /dev/input/event5
Group:            6
Seat:             seat0, default
Size:             100x66mm
Capabilities:     pointer gesture
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    *button-areas clickfinger 
Disable-w-typing: enabled
Accel profiles:   none
Rotation:         n/a

Device:           Wacom HID 50E9 Pen
Kernel:           /dev/input/event6
Group:            7
Seat:             seat0, default
Size:             346x194mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   none
Rotation:         n/a

Device:           Wacom HID 50E9 Finger
Kernel:           /dev/input/event7
Group:            7
Seat:             seat0, default
Size:             346x194mm
Capabilities:     touch 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Ideapad extra buttons
Kernel:           /dev/input/event10
Group:            8
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           HDA Intel PCH Mic
...
Device:           HDA Intel PCH Headphone
...
Device:           HDA Intel PCH HDMI/DP,pcm=3
...
Device:           HDA Intel PCH HDMI/DP,pcm=7
...
Device:           HDA Intel PCH HDMI/DP,pcm=8
..
Device:           HDA Intel PCH HDMI/DP,pcm=9
...
Device:           HDA Intel PCH HDMI/DP,pcm=10
...
Device:           AT Translated Set 2 keyboard
...
```

jasonday, 2022-02-06:
I've tried several distros (ubuntu, fedora, endeavour) with no change.

Using `libwacom-list-local-devices` I get the following:

```
name: 'Wacom ISDv4 50e9'
bus: 'i2c'
vid: '0x056a'
pid: '0x50e9'
nodes: 
- /dev/input/event10
```

In endeavourOS's Power Manager, I can see the the device is recognized, but it lists it as "unknown state"

jasonday, 2022-02-07:
Note: The Lenovo stylus works correctly

Pinglinux, 2022-07-25:
Please run one of the tools https://github.com/linuxwacom/input-wacom/wiki/Analysing-kernel-events to see if the touch port (/dev/input/event7 was shown in your output) posts any touch events. libwacom-list-local-devices is for Gnome control center. It doesn't tell us what happened to the touch events

Pinglinux, 2022-08-04:
If finger worked before stylus touched the tablet, it could be the issue that Jason fixed recently: [Force pen out of prox](https://github.com/linuxwacom/input-wacom/commit/3fe3a4540bf8afd529c4fe3b859494c628bc763b).

Pinglinux, 2022-08-15:
The issue is platform specific. Since we don't get further details, I'm closing the request.

closed 2022-08-15 by Pinglinux