linuxwacom/input-wacom #312: Build Failed on Kali 5.15

artificer-ilari, 2022-03-17:
I'm somewhat new to Linux so please forgive me if I'm missing info.

I have a Wacom One tablet which my computer isn't recognising as a tablet. The computer is an ASUS ROG Zephyrus 14, running Kali via Debian 5.15.5-2kali2 (2021-12-22), and it isn't identifying the tablet as a tablet so I can't give you the serial from terminal. The tablet pen works (sort of) but is calibrated very poorly, such that (0,0) is about right but the cursor moves much further up and to the right than is correct. When I try to run:

```
sudo apt-get install build-essential autoconf linux-headers-$(uname -r) 
```

This is the feedback it gives me:

```
└─$ sudo apt-get install build-essential autoconf linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.15.0-kali2-amd64
E: Couldn't find any package by glob 'linux-headers-5.15.0-kali2-amd64'
E: Couldn't find any package by regex 'linux-headers-5.15.0-kali2-amd64'
```

It seems something isn't updated for the current Kali distro but I can't figure out how to step back an update and see if the install still works. Thanks for your help!

jigpu, 2022-04-11:
Please update your system (e.g. `sudo apt-get update && sudo apt-get upgrade`), reboot, and then try again. This error can occur if your distro has released a new kernel since the last time you updated. The `linux-headers-$(uname -r)` dependency requests the headers for your running kernel, but those won't be available if your kernel is out-of-date.

I've just tried installing these dependencies on Kali 2022.1 and had no issue after updating.

closed 2022-04-11 by jigpu

artificer-ilari, 2022-04-16:
Thanks! My computer won't support the update. Is there a way to install it on 5.15.0, like I asked?

jigpu, 2022-04-18:
My best suggestion would be to try downloading and installing the required package manually. It looks like you can find `linux-headers-5.15.0-kali2-amd64_5.15.5-2kali2_amd64.deb` in the Kali repository: http://http.kali.org/pool/main/l/linux/

Once you have the deb package, you should be able to install it with `sudo dpkg -i <debfile>` and then retry the input-wacom install process.