linuxwacom/input-wacom #318: Touchscreen detects pen, but not fingers

h4-n1, 2022-05-21:
(Originally posted this issue [here](https://github.com/linuxwacom/xf86-input-wacom/issues/268), realized this repo is a better place for it)

The built-in touchscreen on my Lenovo Thinkpad S1 (20CD) will detect the wacom pen but not my fingers.

I'm running Swaywm and GNOME (both wayland) on EndeavourOS. I've also tried Ubuntu and Fedora live USBs, as well as running GNOME under X. Updating the BIOS doesn't help either. This worked on windows (although oddly it won't work now on a windows live usb), so I don't think it can be a fault in the hardware.

All the devices that should be there are detected and assigned input event nodes:

<details>
  <summary>devices detected by `libinput record`</summary>

```
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	ThinkPad Extra Buttons
/dev/input/event5:	Video Bus
/dev/input/event6:	Wacom ISDv4 EC Pen
/dev/input/event7:	ELAN Touchscreen
/dev/input/event8:	ELAN Touchscreen UNKNOWN
/dev/input/event9:	ELAN Touchscreen UNKNOWN
/dev/input/event10:	HDA Intel HDMI HDMI/DP,pcm=3
/dev/input/event11:	HDA Intel HDMI HDMI/DP,pcm=7
/dev/input/event12:	HDA Intel HDMI HDMI/DP,pcm=8
/dev/input/event13:	HDA Intel HDMI HDMI/DP,pcm=9
/dev/input/event14:	HDA Intel HDMI HDMI/DP,pcm=10
/dev/input/event15:	Integrated Camera: Integrated C
/dev/input/event16:	HDA Intel PCH Mic
/dev/input/event17:	HDA Intel PCH Front Headphone
/dev/input/event18:	Synaptics TM2941-001
/dev/input/event19:	TPPS/2 IBM TrackPoint
```
</details>

`libinput record`, `evemu record`, and `intercept` all don't detect *any* input coming from ELAN Touchscreen. Running `sudo modprobe -r wacom; sudo modprobe wacom` as in #310 doesn't help. The symptoms of my issue seem to be the same as in #305 and #297.

Here are the [results of `sysinfo.sh`](https://github.com/linuxwacom/xf86-input-wacom/files/8737811/sysinfo.xNSf4HkzsI.tar.gz) from a few days ago.

Please let me know if there's any other info you'd like me to provide.

h4-n1, 2022-05-29:
Since the pen device is `Wacom ISDv4 EC Pen`, would the appropriate kernel module be `wacom_w8001` instead of `wacom`? In that case should that module handle both the pen and the touchscreen? I think that might be the issue. Loading `wacom_w8001`after boot doesn't work, however, even with unloading other modules. I'm not sure how to test this.

Pinglinux, 2022-07-05:
wacom_w8001 is for ISDv4 serial devices. If your device is [ISDv4 EC] (https://github.com/linuxwacom/libwacom/blob/master/data/isdv4-ec.tablet), yours is a USB device. Pen can be driven by wacom.ko. But, the the touch should be driven by hid-generic since it is an Elan Touchscreen.

jigpu, 2022-07-15:
Closing this issue since the generic HID driver is responsible for the Elan touchscreen, not our driver. You'll need to file an issue either on your [distro's bug tracker](https://forum.endeavouros.com/c/arch-based-related-questions/bug-reports/64) or the upstream [Kernel bugzilla](https://bugzilla.kernel.org/).

closed 2022-07-15 by jigpu