linuxwacom/input-wacom #321: Pad coordinates perfectly match screen coordinates

ZinaKojima, 2022-06-02:
While I was using Windows, I got used to that the pointer does not jump across the screen to the point, which coordinates clearly match those I stroke on my pad.
After changing to Arch, it disturbs a lot. How to make the pointer be merely dragged by stokes but not teleported?

Pinglinux, 2022-07-05:
Which tablet do you use? What version of Arch do you run, especially the kernel version of your Arch? We need those details to troubleshoot the issue.

ZinaKojima, 2022-07-06:
I'm not sure what "arch version" means (desktop env? Manjaro gnome), but kernel version is 5.18.0-arch1-1.
The tablet model is CTE-650.

ZinaKojima, 2022-07-06:
It is in absolute mode, yes.
I actually switched to 5.15.32-1-MANJARO, and the problem remained.
I suppose, I need to switch the mode, right? How could that be done?

Pinglinux, 2022-07-06:
You want to stay at absolute mode, unless you use the tablet as a mouse replacement. You could change the stylus mode under "Wacom Tablet" in Gnome Settings. What do you use the tablet for, drawing or something else? Do you see the issue inside an app or under the whole desktop environment? TBH, I still don't know the exact issue you face. Can you show the issue by a screenshot?

ZinaKojima, 2022-07-07:
[image]
For the first screenshot I've placed the stilus around the center of my pad. Thus, the cursor occured
 around the center of my screen.
[image]
For the second screenshot I've taken the stilus away, moved it leftward about a quarter of the pad's width, and lowered it to the pad's surface. In response, the cursor moved to the left by a quarter of the screen.

The behaviour I would see is that the cursor didn't move, since I hadn't run the stylus over the tablet.

Now, I do what you say. I change the tablet mode, and I get the behaviour I want. Nice and easy.
[image]

The issue is solved, thank you for your help.

closed 2022-07-11 by Pinglinux