linuxwacom/input-wacom #322: Wacom Intuos not recognized in Ubuntu 20.04

jgoodman8, 2022-06-26:
After installing this package from the source code my tablet stopped working. It was previously recognised, but I cannot interact with it anymore.
When installing this package I get this output:

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/jguzman/tools/input-wacom-0.48.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/5.13.0-51-generic/build
checking kernel version... 5.13.0-51-generic
checking RHEL7 minor release... 0
checking power supply version... pre-v4.1
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... 
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... on
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/5.13.0-51-generic/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/5.13.0-51-generic/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... yes
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... yes
checking for module signing key... /var/lib/shim-signed/mok/MOK.priv
checking for module signing certificate... /var/lib/shim-signed/mok/MOK.der

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/5.13.0-51-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/jguzman/tools/input-wacom-0.48.0'
Making all in 4.5
make[2]: Entering directory '/home/jguzman/tools/input-wacom-0.48.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/5.13.0-51-generic/build M=/home/jguzman/tools/input-wacom-0.48.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-5.13.0-51-generic'
make[3]: Leaving directory '/usr/src/linux-headers-5.13.0-51-generic'
make[2]: Leaving directory '/home/jguzman/tools/input-wacom-0.48.0/4.5'
make[2]: Entering directory '/home/jguzman/tools/input-wacom-0.48.0'
make[2]: Leaving directory '/home/jguzman/tools/input-wacom-0.48.0'
make[1]: Leaving directory '/home/jguzman/tools/input-wacom-0.48.0'
Making install in 4.5
make[1]: Entering directory '/home/jguzman/tools/input-wacom-0.48.0/4.5'
make -C /lib/modules/5.13.0-51-generic/build M=/home/jguzman/tools/input-wacom-0.48.0/4.5 modules_install mod_sign_cmd='"/lib/modules/5.13.0-51-generic/build/scripts/sign-file" "sha512" "/var/lib/shim-signed/mok/MOK.priv" "/var/lib/shim-signed/mok/MOK.der"'
make[2]: Entering directory '/usr/src/linux-headers-5.13.0-51-generic'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.13.0-51-generic/extra/wacom.ko
  SIGN    /lib/modules/5.13.0-51-generic/extra/wacom.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /lib/modules/5.13.0-51-generic/extra/wacom_w8001.ko
  SIGN    /lib/modules/5.13.0-51-generic/extra/wacom_w8001.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  /lib/modules/5.13.0-51-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.13.0-51-generic'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
echo "override wacom_w8001 * extra" >> /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 5.13.0-51-generic
/usr/sbin/update-initramfs -u -k 5.13.0-51-generic
update-initramfs: Generating /boot/initrd.img-5.13.0-51-generic
make[1]: Leaving directory '/home/jguzman/tools/input-wacom-0.48.0/4.5'
make[1]: Entering directory '/home/jguzman/tools/input-wacom-0.48.0'
make[2]: Entering directory '/home/jguzman/tools/input-wacom-0.48.0'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/jguzman/tools/input-wacom-0.48.0'
make[1]: Leaving directory '/home/jguzman/tools/input-wacom-0.48.0'

```

Now, if I run `xsetwacom --list devices` I cannot find it.

I'm attaching the output of the sysinfo,

[sysinfo.Eyl76l9RBs.tar.gz](https://github.com/linuxwacom/input-wacom/files/8986893/sysinfo.Eyl76l9RBs.tar.gz)

Pinglinux, 2022-07-05:
From your sysinfo file, I see your Intuos is driven by hid-generic driver:

[   81.143925] hid-generic 0005:056A:03C6.000D: unknown main item tag 0x0
[   81.144040] hid-generic 0005:056A:03C6.000D: hidraw10: BLUETOOTH HID v1.00 Device [Wacom Intuos PBS] on 14:18:c3:ea:ed:4e

Was there anything wrong with your built-in wacom.ko? That is, what triggered you to build the wacom.ko from source?

Also, from your attached make; make install, I don't see wacom.ko built under 4.5.

Pinglinux, 2022-07-18:
Wacom kernel driver isn't driving the device, which is either an OS issue or an user change.

closed 2022-07-18 by Pinglinux

jigpu, 2022-07-19:
There might be an issue with your kernel module signing setup. I'm seeing `unsigned module loading is restricted; see man kernel_lockdown.7` messages immediately after every device probe by hid-generic. Your build output also has a pair of `sign-file: certs/signing_key.pem: No such file or directory` errors that seem to indicate the driver could not be signed. You may want to reference our [Secure Boot](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot) wiki article for information on how to either bypass or satisfy the signing requirements.