linuxwacom/input-wacom #325: After upgrade to Ubuntu 22.04 any movement of Intuos2 mouse crashes Xorg

slalomsk8er, 2022-08-18:
As the title states, the logon experience right after upgrade wasn't to good and it took me a while to figure out, that it wasn't the NVIDIA driver but the "mouse".

```
Operating System: Ubuntu 22.04.1 LTS              
          Kernel: Linux 5.15.0-46-lowlatency
    Architecture: x86-64

```

```
[  1229.938] (EE) Backtrace:
[  1229.938] (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x139) [0x55f97a8696e9]
[  1229.939] (EE) 1: /lib/x86_64-linux-gnu/libc.so.6 (__sigaction+0x50) [0x7f8496066520]
[  1229.939] (EE) 2: /lib/x86_64-linux-gnu/libc.so.6 (pthread_kill+0x12c) [0x7f84960baa7c]
[  1229.940] (EE) 3: /lib/x86_64-linux-gnu/libc.so.6 (raise+0x16) [0x7f8496066476]
[  1229.941] (EE) 4: /lib/x86_64-linux-gnu/libc.so.6 (abort+0xd3) [0x7f849604c7f3]
[  1229.941] (EE) unw_get_proc_name failed: no unwind info found [-10]
[  1229.941] (EE) 5: /lib/x86_64-linux-gnu/libc.so.6 (?+0x0) [0x7f849604c71b]
[  1229.942] (EE) 6: /lib/x86_64-linux-gnu/libc.so.6 (__assert_fail+0x46) [0x7f849605de96]
[  1229.942] (EE) unw_get_proc_name failed: no unwind info found [-10]
[  1229.942] (EE) 7: /usr/lib/xorg/modules/input/wacom_drv.so (?+0x0) [0x7f84958b3dbf]
[  1229.942] (EE) unw_get_proc_name failed: no unwind info found [-10]
[  1229.942] (EE) 8: /usr/lib/xorg/modules/input/wacom_drv.so (?+0x0) [0x7f84958b460f]
[  1229.942] (EE) unw_get_proc_name failed: no unwind info found [-10]
[  1229.943] (EE) 9: /usr/lib/xorg/modules/input/wacom_drv.so (?+0x0) [0x7f84958b7114]
[  1229.943] (EE) unw_get_proc_name failed: no unwind info found [-10]
[  1229.943] (EE) 10: /usr/lib/xorg/modules/input/wacom_drv.so (?+0x0) [0x7f84958ab2f8]
[  1229.943] (EE) 11: /usr/lib/xorg/Xorg (input_unlock+0x2e7) [0x55f97a867997]
[  1229.943] (EE) 12: /usr/lib/xorg/Xorg (OsCleanup+0x571) [0x55f97a86a0d1]
[  1229.943] (EE) 13: /usr/lib/xorg/Xorg (input_unlock+0x109) [0x55f97a8677b9]
[  1229.944] (EE) 14: /lib/x86_64-linux-gnu/libc.so.6 (pthread_condattr_setpshared+0x513) [0x7f84960b8b43]
[  1229.944] (EE) 15: /lib/x86_64-linux-gnu/libc.so.6 (__xmknodat+0x230) [0x7f849614aa00]
[  1229.944] (EE) 
[  1229.944] (EE) 
Fatal server error:
[  1229.944] (EE) Caught signal 6 (Aborted). Server aborting
[  1229.944] (EE) 
[  1229.944] (EE) 
```

```
cat /sys/module/wacom/version
v2.00
```

```
nala search wacom
kde-config-tablet 3.2.0-4build1 [Ubuntu/jammy universe]
├── ist installiert
└── implements a KDE configuration GUI for the Wacom drivers

kde-config-tablet 3.2.0-4build1 [Ubuntu/jammy universe]
├── ist installiert
└── implements a KDE configuration GUI for the Wacom drivers

libwacom-bin 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library -- binaries

libwacom-bin 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library -- binaries

libwacom-bin 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library -- binaries

libwacom-bin:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library -- binaries

libwacom-bin:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library -- binaries

libwacom-bin:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library -- binaries

libwacom-common 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library (common files)

libwacom-common 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library (common files)

libwacom-common 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library (common files)

libwacom-dev 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library (development files)

libwacom-dev 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library (development files)

libwacom-dev 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library (development files)

libwacom-dev:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library (development files)

libwacom-dev:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library (development files)

libwacom-dev:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library (development files)

libwacom9 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library

libwacom9 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library

libwacom9 2.2.0-1 [Ubuntu/jammy main]
├── ist installiert
└── Wacom model feature query library

libwacom9:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library

libwacom9:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library

libwacom9:i386 2.2.0-1 [Ubuntu/jammy main]
└── Wacom model feature query library

xserver-xorg-input-wacom 1:1.0.0-3ubuntu1 [Ubuntu/jammy main]
├── ist installiert
└── X.Org X server -- Wacom input driver

xserver-xorg-input-wacom 1:1.0.0-3ubuntu1 [Ubuntu/jammy main]
├── ist installiert
└── X.Org X server -- Wacom input driver

xserver-xorg-input-wacom 1:1.0.0-3ubuntu1 [Ubuntu/jammy main]
├── ist installiert
└── X.Org X server -- Wacom input driver

xserver-xorg-input-wacom-hwe-18.04 3:14.6 [Ubuntu/jammy universe]
└── Transitional package for xserver-xorg-input-wacom-hwe-18.04

xserver-xorg-input-wacom-hwe-18.04 3:14.6 [Ubuntu/jammy universe]
└── Transitional package for xserver-xorg-input-wacom-hwe-18.04
```

Pinglinux, 2022-08-19:
This is probably a X driver issue. Let's figure out the root cause first. Can you share the whole Xorg.0.log and dmesg with the tablet connected to the system and without touching the Intuos 2 by pen or finger?

slalomsk8er, 2022-08-20:
[_usr_lib_xorg_Xorg.0.crash.txt](https://github.com/linuxwacom/input-wacom/files/9387215/_usr_lib_xorg_Xorg.0.crash.txt)
I think all you could ask for is in this file and sorry for reporting in the wrong repository

Pinglinux, 2022-08-23:
Where did you get the content? The file is loaded with hex coredump which I can't read. The part I can read shows X driver wasn't happy. You have the old version of Intuos 2, which should work out of box. Did the device work on older Ubuntu? Did you try Wayland/libinput?

slalomsk8er, 2022-08-23:
I think the file is a Ubuntu thing and repesents the basis of there automatic bug reporting. It looks to me, that it contains every thing from xorg log to dmesg to kernel dump.

Yes, it worked out of the box before the upgrade and on every Linux distribution I used over the years.

I haven't played with Wayland/libinput yet as all of my PCs have Nvidia GPUs.

Pinglinux, 2022-08-23:
I see. You need to stay with X. Can you stay at Ubuntu 20.04 or try to install, instead of upgrade to, 22.04? From your shared info, "xserver-xorg-input-wacom-hwe-18.04 3:14.6" was mentioned. I want to rule out the possibility of issues unrelated to 22.04.

slalomsk8er, 2022-08-23:
> I see. You need to stay with X. Can you stay at Ubuntu 20.04 or try to install, instead of upgrade to, 22.04? From your shared info, "xserver-xorg-input-wacom-hwe-18.04 3:14.6" was mentioned. I want to rule out the possibility of issues unrelated to 22.04.

I already upgraded so no going back there but I could use a external drive to test with 20.04 and 22.04 and thus rule out a upgrade problem.

> I'm seeing the following assertion warning in the log:
> 
> ```
> AssertionMessage: Xorg: ./WacomInterface.h:206: wcmAxisValue: Assertion `snprintf(buf, len, \"%d\", val) < len' failed.`
> ```
> 
> In theory this should only fail if the buffer we provided to snprintf was too small to contain the string representation of `val`. However, there is a bug in versions 0.99.3 through 1.0.0 which passes the wrong value for `len`. The incorrect value is smaller than it should be, which can cause the assertion to incorrectly fail.
> 
> This issue was fixed by this commit in version 1.1.0: [linuxwacom/xf86-input-wacom@571fbce](https://github.com/linuxwacom/xf86-input-wacom/commit/571fbced64a51259ea7825c737f1703a0eb6bdf5)
> 
> I suspect if you update xf86-input-wacom to the latest version it should resolve the issue.

This sounds more promising to me and I will try this first.

slalomsk8er, 2022-08-24:
Switched from version 1.0.0 to 1.1.0 and no crash yet, while before X crashed as soon as I touched the mouse or after a while if I didn't (can't rule out accidental micro movement of mouse).

slalomsk8er, 2022-08-24:
How to close as fixed and not as completed?

closed 2022-08-24 by slalomsk8er