linuxwacom/input-wacom #329: Mouse issues

hendersj, 2022-10-01:
I've had a few issues over the years with my Intuos 4M mouse, but generally not significant enough to warrant reporting an issue (the biggest one is the QT scroll wheel issue, which I understand y'all are pointing to the QT project for progress on).

I'm now seeing that scroll wheel issue in GNOME apps as well, but not consistently.  Running openSUSE Tumbleweed with GNOME 3 on Xorg, and things like the GNOME settings app don't scroll when I dial the scroll wheel.  A regular bluetooth mouse works fine (which makes it seem more like it's something not being translated by the Wacom driver correctly).  But other GNOME/GTK apps work fine.  Chrome scrolls fine, as does Firefox.  It's just really, really inconsistent.

I also had noticed that when attempting to use the Wacom mouse in Stadia, the mouse pointer would be dragged hard to the edges of the screen, making it completely unusable.  While Stadia is (as announced yesterday) being shut down in January, that issue is still just another example of something weird in how the mouse device is handled differently from a bluetooth or wired mouse.

Looking for any guidance on how to resolve this issue definitively.  It *seems* like the driver just isn't sending the same kind of info along as the normal mouse drivers (bluetooth, USB), but I don't know a lot about the internals of any of those drivers.

Pinglinux, 2022-10-05:
Hi @hendersj, this project is for the kernel driver of the device. If you see issues with the raw (x,y) value of the mouse, we can look into it and figure out the root cause. If it is specific to some apps/program/desktop environments, you'll need to get help from those specific open source projects.

hendersj, 2022-10-05:
It seems to be the difference between what's presented by the driver to the kernel and what a regular mouse is - I don't think it's tied to the specific applications or desktop environments, but something more fundamental than that.  A regular bluetooth mouse (I have an Apple Magic Mouse here on my desk that I use with my work system that I've tested with), and that works flawlessly with all applications.  If this was a one-off app issue, I could see reporting it there, but it's not - it's many apps, some of which have no commonality with each other (ie, not all QT apps, not all GTK apps).  I really struggle with the idea of reaching out to each of those projects to say "hey, my mouse doesn't work with your app, can you fix this?" when the common starting point would be the driver for the device.  Were I one of those app developers, my inclination would be check with my mouse, see that it works, and say "not my problem, get a 'real' mouse".

In talking with some of the folks in the openSUSE forums, they've suggested that maybe trying using libinput rather than the wacom driver for comparison (obviously I would lose the pen's pressure info by doing that, so it's not a long-term solution) just to try to narrow things down a bit.

It really seems like there's just something not being passed back from the driver that a bluetooth mouse (or USB-connected mouse, I've tried one of those as well) is handling.  I keep coming back to the common thing being the wacom driver.

Happy to run whatever diagnostics are needed to figure this out - not sure where to start or what information is useful.  Let me know, and I'll happily do that.

Pinglinux, 2022-10-07:
> suggested that maybe trying using libinput

That's a good suggestion. libinput supports all the features that a tablet provides, which include pressure sensitivity.

hendersj, 2022-10-07:
What functionality would I lose, then?

Pinglinux, 2022-10-12:
libinput is the future of xf86-input-wacom, which is the current X driver that reports tablet values to userland apps. So, in theory, libinput supports the functionality the X driver supports. If you find anything missing with libinput, report it to the project since it covers the future.

Another fact I'd like to mention about pucks/mouse on tablet is they report absolute values. It is the responsibility of userland drivers/clients to translate the absolute values into appropriate relative values. The kernel driver, input-wacom, reports the raw tablet events without modification. If you compared mouse events with tablet puck events in the kernel, they should be different. Your issue is in the userland.

Pinglinux, 2022-10-31:
This is not a driver issue.

closed 2022-10-31 by Pinglinux