linuxwacom/input-wacom #330: stylus / finger fail in small vertical band in middle of screen

gvb1234, 2022-10-22:
Hi, long time user of wacom on linux, I'd like first to thank you for all the work.

I have a Lenovo X1 yoga (4th gen) which has a wacom layer for both stylus and finger. This setup has worked very well for the last 2+ years until this week.

A lenovo tech has had to change the screen for an unrelated reason. With the new screen, there is a vertical band of about 1cm in the middle of the screen where stylus and finger are not recognized. Everything else works just fine in all other regions of the screen.

Specifically, if I move a finger on the screen from side to side, the cursor follows my finger until I reach the the middle where it stops folliowing, and same if I use the stylus. With the stylus, the two messages below appear when the stylus is close to the offending region:

"wacom_idleprox_timeout: tool appears to be hung in-prox. forcing it out." 

and 

 "wacom_wac_queue_insert: kfifo has filled, starting to drop events"

I'd be tempted to say it's a hardware problem, but the "cursor stops following finger" does not happen in the BIOS (stylus does not work at all in BIOS), which makes me doubt the hardware is faulty.

Any idea?

gvb1234, 2022-10-26:
If that helps, I've recorded two such events.

First one is [here](https://github.com/linuxwacom/input-wacom/files/9869103/finger.txt). I was moving a finger on the screen from left to right. At ~0.685 (I think), the cursor stopped following in the middle of the screen.

The second one is [here](https://github.com/linuxwacom/input-wacom/files/9869107/stylus.txt). I was successfully using the stylus to draw a vertical line off center until 1.08, then from 7.33 was trying to draw a vertical line in the non-working region of the screen.

Pinglinux, 2022-10-28:
If nothing else changed except the screen, it must be a hardware issue. Nothing else could cause the issue since they all worked togther.

gvb1234, 2022-10-28:
Thanks Ping, that was my first thought as well. However, the problem does not happen in the BIOS, at least for the touchscreen (the stylus is not recognized at all in BIOS).

closed 2022-10-28 by gvb1234