linuxwacom/input-wacom #331: Tablet works but xsetwacom doesnt...

MAA28, 2022-11-06:
Im trying to use my tablet but im unable too configure it...
I installed input-wacom with `sudo pamac install xf86-input-wacom` and i plugged in the intuos pro and then tried `xsetwacom list devices` and it didnt return anything.

Pinglinux, 2022-11-13:
This issue doesn't belong here. xsetwacom is part of xf86-input-wacom.

With that said, I think you are either not running X server or using xf86-input-libinput instead of xf86-input-wacom. Please take a look at this one https://github.com/linuxwacom/xf86-input-wacom/issues/105 to see if there is anything similar to your case.

Pinglinux, 2023-03-31:
One of my suggestions may be the root cause.

closed 2023-03-31 by Pinglinux