linuxwacom/input-wacom #332: how to make wacom working automatically at startup?

kyon328, 2022-11-17:
my wacom not working at startup, no response when I moving pens on tablet, can not move cursor
but Debian can detect it, it shows in Setting -> Input Devices -> Graphic Tablet
and after I plug out then re-plug in usb, it works perfect
but after reboot, I have to do that again make it work...
 so is there anyway to make it work automatically at startup?

Pinglinux, 2022-11-22:
Which version of Debian do you have the issue with? Did you update/change the system yourself? What kernel version do you use?

kyon328, 2022-11-23:
> Which version of Debian do you have the issue with? Did you update/change the system yourself? What kernel version do you use?

already solved, it seems to be debian's compatible bugs with qt, because if I use kde or lxqt, after update nvidia driver and reboot, wacom not working, but if I use gnome or xfce, it works fine after update nvidia driver

closed 2022-11-23 by kyon328