linuxwacom/input-wacom #334: Wacom Tablet and Steam Deck - Update

rickleone, 2022-11-26:
I'm sorry to open a second post on the same issue, but I still don't know how to properly delete or edit the previous one, since a couple of things have changed, I've been able to remove the read-only mode of the Steam Deck for example, and I've managed to install at least the required libraries for the Kernel.
For context, I'm trying to make a freshly bought **Wacom One (CPT133)** work with the **Steam Deck**. They're connected via a type-C USB hub. So far, the Deck recognizes it as a second screen, but the **pen function is entirely missing**.
I've done this thanks to a Linux techie on Discord, who gave me all the commands to type in the terminal, but even they can't quite figure out the installation of the Kernel, being the Deck such a curious hybrid between open-source and proprietary software.
So, what happened so far? So, for starters, they made me **update the keyring**.
```
(deck@steamdeck ~)$ sudo pacman-key --init
(deck@steamdeck ~)$ sudo pacman-key --populate archlinux
==> Appending keys from archlinux.gpg...
==> Updating trust database...
gpg: next trustdb check due at 2022-12-01 
```
And this successfully installed the libraries. Even if they say that some were already installed.
 ```
(deck@steamdeck ~)$ sudo pacman -S archlinux-keyring
[sudo] password for deck: 
warning: archlinux-keyring-20220125-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) archlinux-keyring-20220125-1

Total Download Size:   1.01 MiB
Total Installed Size:  1.43 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 archlinux-keyrin...  1036.5 KiB   602 KiB/s 00:02 [###] 100%
(1/1) checking keys in keyring                     [###] 100%
(1/1) checking package integrity                   [###] 100%
(1/1) loading package files                        [###] 100%
(1/1) checking for file conflicts                  [###] 100%
(1/1) checking available disk space                [###] 100%
:: Processing package changes...
(1/1) reinstalling archlinux-keyring               [###] 100%
==> Appending keys from archlinux.gpg...
gpg: public key DB323392796CA067 is 3037 days newer than the signature
==> Updating trust database...
gpg: next trustdb check due at 2022-12-01
==> Updating trust database...
gpg: next trustdb check due at 2022-12-01
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
(deck@steamdeck ~)$ ^C
(130)(deck@steamdeck ~)$ sudo pacman-key --init; sudo pacman-key --populate archlinux
==> Appending keys from archlinux.gpg...
==> Updating trust database...
gpg: next trustdb check due at 2022-12-01
(deck@steamdeck ~)$ sudo pacman-key --init
(deck@steamdeck ~)$ sudo pacman-key --populate archlinux
==> Appending keys from archlinux.gpg...
==> Updating trust database...
gpg: next trustdb check due at 2022-12-01
(deck@steamdeck ~)$ sudo pacman -S base-devel linux-headers
:: There are 24 members in group base-devel:
:: Repository core
   1) autoconf  2) automake  3) binutils  4) bison
   5) fakeroot  6) file  7) findutils  8) flex  9) gawk
   10) gcc  11) gettext  12) grep  13) groff  14) gzip
   15) libtool  16) m4  17) make  18) pacman  19) patch
   20) pkgconf  21) sed  22) sudo  23) texinfo  24) which

Enter a selection (default=all): 
warning: autoconf-2.71-1 is up to date -- reinstalling
warning: automake-1.16.5-1 is up to date -- reinstalling
warning: binutils-2.36.1-3 is up to date -- reinstalling
warning: bison-3.8.2-1 is up to date -- reinstalling
warning: fakeroot-1.27-1 is up to date -- reinstalling
warning: file-5.41-1 is up to date -- reinstalling
warning: findutils-4.8.0-1 is up to date -- reinstalling
[... 452 lines cut ...]
Net Upgrade Size:       0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                                                                [#######################################################] 100%
(1/1) checking package integrity                                                              [#######################################################] 100%
(1/1) loading package files                                                                   [#######################################################] 100%
(1/1) checking for file conflicts                                                             [#######################################################] 100%
(1/1) checking available disk space                                                           [#######################################################] 100%
:: Processing package changes...
(1/1) reinstalling glibc                                                                      [#######################################################] 100%
warning: directory permissions differ on /var/db/
filesystem: 711  package: 755
Generating locales...
  en_US.UTF-8... done
Generation complete.
:: Running post-transaction hooks...
(1/4) Reloading system manager configuration...
(2/4) Creating temporary files...
/usr/lib/tmpfiles.d/steamos.conf:24: Duplicate line for path "/var/empty", ignoring.
/usr/lib/tmpfiles.d/tmp.conf:12: Duplicate line for path "/var/tmp", ignoring.
/usr/lib/tmpfiles.d/var.conf:19: Duplicate line for path "/var/cache", ignoring.
(3/4) Arming ConditionNeedsUpdate...
(4/4) Updating the info directory file...
```
And we're pretty much stranded here. Seems like the "headers" are missing, whatever the "headers" are. Be kind to me, I'm an Ubuntu user that barely knows how Ubuntu works after 15 years of using only that.

jigpu, 2022-11-29:
> And we're pretty much stranded here. Seems like the "headers" are missing, whatever the "headers" are. Be kind to me, I'm an Ubuntu user that barely knows how Ubuntu works after 15 years of using only that.

That looks about right. The "headers" provide definitions for the functions that are available in the kernel, and we can't build this driver without them. Your logs show that you were able to install the "linux-headers" package, but they appear to be for a different kernel than what is running:
* Installed package: linux-headers-5.16.2.arch1-1.1
* Running kernel: 5.13.0-valve21.3-1-neptune

Since the Steam Deck is running a customized version of Arch it isn't entirely surprising to see that the kernel is different. It may be necessary to install a different package to get the headers for Valve's kernel.

Please run `pacman -Sl | grep linux` and copy / paste the output here. It should list out all of the kernel and support packages available to you.

rickleone, 2022-12-01:
> Please run pacman -Sl | grep linux and copy / paste the output here. It should list out all of the kernel and support packages available to you.
Here's the output
```
$ pacman -Sl | grep linux
jupiter linux-firmware-neptune 20220624.650ab31-1 [installed]
jupiter linux-firmware-neptune-rtw-debug 20220524-1
jupiter linux-neptune 5.13.0.valve21.3-1 [installed]
jupiter linux-neptune-headers 5.13.0.valve21.3-1
jupiter linux-neptune-rtw-debug 5.13.0.valve12.rtwdbg.4-1
jupiter linux-neptune-rtw-debug-headers 5.13.0.valve12.rtwdbg.4-1
jupiter linux-tools-meta 5.13-1
holo linux 5.16.2.arch1-1.1
holo linux-headers 5.16.2.arch1-1.1 [installed]
holo linux-lts 5.15.16-1.1
holo linux-lts-headers 5.15.16-1.1
holo linux-steam-integration-debug 0.7.3-4
core archlinux-keyring 20220125-1 [installed]
core linux 5.16.2.arch1-1
core linux-api-headers 5.12.3-1 [installed]
core linux-docs 5.16.2.arch1-1
core linux-firmware 20220119.0c6a7b3-2
core linux-firmware-bnx2x 20220119.0c6a7b3-2
core linux-firmware-liquidio 20220119.0c6a7b3-2
core linux-firmware-marvell 20220119.0c6a7b3-2
core linux-firmware-mellanox 20220119.0c6a7b3-2
core linux-firmware-nfp 20220119.0c6a7b3-2
core linux-firmware-qcom 20220119.0c6a7b3-2
core linux-firmware-qlogic 20220119.0c6a7b3-2
core linux-firmware-whence 20220119.0c6a7b3-2
core linux-headers 5.16.2.arch1-1 [installed: 5.16.2.arch1-1.1]
core linux-lts 5.15.17-1
core linux-lts-docs 5.15.17-1
core linux-lts-headers 5.15.17-1
core syslinux 6.04.pre2.r11.gbf6db5b4-3
core util-linux 2.37.3-1 [installed]
core util-linux-libs 2.37.3-1 [installed]
extra archlinux-appstream-data 20220107-1 [installed]
extra archlinux-menus 2.5-5
extra archlinux-themes-slim 1.2-6
extra linux-atm 2.5.2-7
extra linux-hardened 5.15.16.hardened1-1
extra linux-hardened-docs 5.15.16.hardened1-1
extra linux-hardened-headers 5.15.16.hardened1-1
extra linux-zen 5.16.2.zen1-1
extra linux-zen-docs 5.16.2.zen1-1
extra linux-zen-headers 5.16.2.zen1-1
community aarch64-linux-gnu-binutils 2.37-1
community aarch64-linux-gnu-gcc 11.2.0-1
community aarch64-linux-gnu-gdb 11.2-1
community aarch64-linux-gnu-glibc 2.34-1
community aarch64-linux-gnu-linux-api-headers 5.8-1
community archlinux-contrib 20210710-1
community archlinux-repro 20220116-1
community archlinux-wallpaper 1.6.1-1
community archlinux-xdg-menu 0.7.6.3-2
community golang-github-linuxdeepin-go-dbus-factory 1.9.22-1
community golang-github-linuxdeepin-go-x11-client 0.6.7-2
community linux-tools-meta 5.16-1
community linuxconsole 1.7.1-1
community linuxdoc-tools 0.9.82-1
community linuxsampler 2.2.0-2
community perl-linux-pid 0.04-14
community playonlinux 4.3.4-4
community python-linux-procfs 0.7.0-1
community python-selinux 0.2.1-7
community riscv64-linux-gnu-binutils 2.36.1-1
community riscv64-linux-gnu-gcc 11.1.0-1
community riscv64-linux-gnu-gdb 11.2-1
community riscv64-linux-gnu-glibc 2.32-1
community riscv64-linux-gnu-linux-api-headers 5.8-1
community ttf-linux-libertine 5.3.0-7
community ttf-linux-libertine-g 20120116-6
community xdm-archlinux 0.7-2
multilib lib32-util-linux 2.37.3-1 [installed]
```

jigpu, 2022-12-02:
It looks like `linux-neptune-headers` is probably the package that you'll need. Try installing that and then re-running the input-wacom install instructions.

Pinglinux, 2023-03-29:
I assume the issue fixed.

closed 2023-03-29 by Pinglinux