linuxwacom/input-wacom #338: input-wacom modules do not load on startup on Nobara Linux 36

CorInABox, 2022-12-26:
Hello, I am currently using a Wacom Cintiq 16 on Nobara Linux 36 (secure boot disabled) and I installed the input-wacom v0.49.0 drivers because the stylus wasn't working (the tablet was detected in settings beforehand). I followed the install from source instructions, however, the modules do not load on startup and I have to run `modprobe wacom` and `modprobe wacom_w8001` on every boot. Afterwards, the stylus works for the current session.

Before `modprobe`:

`dmesg` gives me the following output:

[  111.284908] usb 1-7.1: Product: Cintiq 16
[  111.284909] usb 1-7.1: Manufacturer: Wacom Co.,Ltd.
[  111.284910] usb 1-7.1: SerialNumber: 9HQ00X1005268
[  111.348345] input: Wacom Co.,Ltd. Cintiq 16 System Control as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-7/1-7.1/1-7.1:1.0/0003:056A:0390.0007/input/input23
[  111.400331] input: Wacom Co.,Ltd. Cintiq 16 Stylus as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-7/1-7.1/1-7.1:1.0/0003:056A:0390.0007/input/input24
[  111.400404] input: Wacom Co.,Ltd. Cintiq 16 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-7/1-7.1/1-7.1:1.0/0003:056A:0390.0007/input/input25
[  111.400523] hid-generic 0003:056A:0390.0007: input,hiddev99,hidraw6: USB HID v1.10 Device [Wacom Co.,Ltd. Cintiq 16] on usb-0000:02:00.0-7.1/input0

Going through the troubleshooting section: 

`grep "" /sys/module/wacom*/version`   grep: /sys/module/wacom*/version: No such file or directory
`sudo rmmod wacom`  rmmod: ERROR: Module wacom is not currently loaded
`sudo rmmod wacom_w8001`  rmmod: ERROR: Module wacom_w8001 is not currently loaded

After running `modprobe`:
`grep "" /sys/module/wacom*/version`: v2.00-0.49.0

Thank you!

waffleboi9000, 2023-01-14:
If you're still having the issue, I got some help in the discord. The reason is that opentabletdriver was installed, and in /usr/lib/modprobe.d/opentabletdriver.conf wacom is blacklisted. I deleted opentabletdriver and the issue was gone. Let me know if it helps you too!

CorInABox, 2023-01-15:
Uninstalling opentabletdriver solved the stylus problem! Thank you @waffleboi9000 !

closed 2023-01-15 by CorInABox