linuxwacom/input-wacom #339: Touch screen has a nonexistant battery on ThinkPad Z13

hexchain, 2022-12-29:
It seems that the upstream Wacom driver exposes a "battery" called `wacom_battery_0` for the touch screen on ThinkPad Z13. The battery is always at level 0% and does not have any other information. Is it possible to not generate this power supply?

Kernel version: 6.1.1-arch1-1
Driver version (in-tree): v2.00
Kernel log:
```
% journalctl -k --grep wacom --no-pager
Dec 27 18:42:40 kotori kernel: input: Wacom HID 52E6 Pen as /devices/platform/AMDI0010:01/i2c-1/i2c-WACF2200:00/0018:056A:52E6.0002/input/input26
Dec 27 18:42:40 kotori kernel: input: Wacom HID 52E6 Finger as /devices/platform/AMDI0010:01/i2c-1/i2c-WACF2200:00/0018:056A:52E6.0002/input/input27
Dec 27 18:42:40 kotori kernel: wacom 0018:056A:52E6.0002: hidraw1: I2C HID v1.00 Mouse [WACF2200:00 056A:52E6] on i2c-WACF2200:00
```

jigpu, 2023-03-31:
This sounds like the same issue as https://bugzilla.kernel.org/show_bug.cgi?id=217062 which I'm in the process of resolving. I have a patch which fixes the issue which I'm encouraging affected users to test.

Please follow the [input-wacom build instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) to build and install the driver. However, instead of downloading the latest release, please use the following command to clone the appropriate branch of my git repo:

~~~
$ git clone https://github.com/jigpu/input-wacom.git -b lazy-battery
~~~

After installing the driver, please check the output of `grep "" /sys/module/wacom*/version` to make sure that you're running the patched version. It should report a version number of `v2.00-0.49.0.13.ged57a9a` if the patch has been applied. Please let me know if you have any problems or if the patch functions correctly for you.

hexchain, 2023-03-31:
I built the module, unloaded the in-tree one, and loaded it with `insmod`. It reports a version of `v2.00-0.42.0.278.ged57a9a`.
The battery device `wacom_battery_0` disappeared after this. Touchscreen functions are not affected.

jigpu, 2023-03-31:
Great! Thanks for the report!

closed 2023-04-21 by Pinglinux