linuxwacom/input-wacom #341: Help debugging why pen works, but finger does not

levouh, 2023-01-13:
Similar to #318 it seems, however my device hardware is a bit different. I am using a Lenovo X1 Yoga (4th Generation) with the 2K display option. The output from the `sysinfo.sh` script can be found [here](https://github.com/linuxwacom/input-wacom/files/10413843/sysinfo.eWAVWHO25F.tar.gz).

Output from `libinput record` is as follows:

```
Available devices:
/dev/input/event0:	Sleep Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	TPPS/2 Elan TrackPoint
/dev/input/event5:	Wacom Pen and multitouch sensor Finger
/dev/input/event6:	Wacom Pen and multitouch sensor Pen
/dev/input/event7:	SYNA8004:00 06CB:CD8B Mouse
/dev/input/event8:	SYNA8004:00 06CB:CD8B Touchpad
/dev/input/event9:	Video Bus
/dev/input/event10:	ThinkPad Extra Buttons
/dev/input/event11:	PC Speaker
/dev/input/event12:	Integrated Camera: Integrated C
/dev/input/event13:	sof-hda-dsp Mic
/dev/input/event14:	sof-hda-dsp Headphone
/dev/input/event15:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event16:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event17:	sof-hda-dsp HDMI/DP,pcm=5
```

where I _assume_ that I would be able to do touchscreen input via the `Wacom Pen and multitouch sensor Finger` device at event number 5.

---

At this point, I'm not even sure if my device actually supports touchscreen finger input in the first place.

- Tried using Ubuntu 22.10 and Fedora 37, might boot Windows just to see what happens
- Verified that the device is indeed being recognized (or so it seems) via `dmesg`
- Tried uninstalling and reinstalling all Wacom-related stuff, unloading and reloading kernel modules, installing the drivers (?) /kernel modules from source, etc.
- Flipped the 2-in-1 into "tablet mode" where the track pad and keyboard are disabled, but the finger touch input still does not work in that state

What else can I do to debug why things are not working as expected?

levouh, 2023-01-13:
Booted Windows and have the same result - pen works but finger does not. How can I confirm if the device even supports finger touch on the main screen?

Pinglinux, 2023-03-31:
Your device most likely doesn't have a touchscreen. Their tech spec didn't mention touchscreen: https://www.lenovo.com/us/en/p/laptops/thinkpad/thinkpadx1/x1-yoga-gen-4/22tp2txx14y?orgRef=https%253A%252F%252Fwww.bing.com%252F&cid=us:sem|se|msn|brand_commercial_think_x+series|x1+yoga_4th+gen|lenovo%20x1%20yoga%20gen4|e|402399255|1263339624453011|kwd-78959171475010:loc-190|text|brand&msclkid=ddadbfc47c7a1c8c5946dbe22ae3437e#tech_specs. They did mention pen though.

closed 2023-04-07 by Pinglinux

ferre111, 2024-10-27:
@levouh I also have a Lenovo X1 Yoga (1st generation), and one day my touchscreen stop responding to finger (I'm sure it worked in the past). I would like to ask if you were able to resolve this issue?