linuxwacom/input-wacom #343: Build fails on ubuntu 22.04 kernel 5.10 aarch64

Pedrocasf, 2023-01-29:
Building the main branch with 
if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed" 
fails with several
input-wacom/4.5/wacom_wac.c:4168:23: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
error, forbidden warning:wacom_wac.c:4168
 4168 |         for (i = 0; i < wacom->led.count; i++)

Pinglinux, 2023-03-29:
Can you paste the whole build output with errors here?

Also, I'm curious about the reason you want to build input-wacom on Ubuntu 22.04. Do you try to make a tablet work? If yes, which tablet?

Pinglinux, 2023-06-26:
I see the "warning" treated as error issue. That int

closed 2023-06-28 by Pinglinux