linuxwacom/input-wacom #344: Wacom Cintiq 12WX behaves erratic?

eezacque, 2023-01-30:
When I switch on my Wacom Cintiq 12WX, about 1 out of 3 times I cannot use its stylus, like it is not there, or it seems to be present, but its button is not functioning. I would appreciate any pointers that help me to pin down this extremely annoying issue

Pinglinux, 2023-03-29:
What Linux distribution and version are you using?

eezacque, 2023-03-29:
On 29/03/2023 18:49, Ping Cheng wrote:
>
> What Linux distribution and version are you using?
>
Linux behemoth 5.15.0-67-generic #74-Ubuntu SMP Wed Feb 22 14:14:39 UTC 
2023 x86_64 x86_64 x86_64 GNU/Linux

-- 
Grinnikend door het leven...

Pinglinux, 2023-03-31:
That version should support Cintiq 12WX. Can you post your dmesg here, especially the part that shows Wacom devices. Are you under X or Wayland?

eezacque, 2023-03-31:
On 31/03/2023 05:38, Ping Cheng wrote:
>
> That version should support Cintiq 12WX. Can you post your dmesg here, 
> especially the part that shows Wacom devices. Are you under X or Wayland?
>
It worked flawlessly in the past, and it still works on Windows, so the 
hardware itself seems to be ok. The problem is that, sooner or later, 
wacom crashes my xserver with, I think, a segmentation fault...

-- 
Grinnikend door het leven...

Pinglinux, 2023-04-06:
Please post your segfault message here. Thanks.

eezacque, 2023-05-15:
Looks like something has improved on the Linux side. I haven't seen 
segmentation faults for weeks, although about half of the times I switch 
on my Cintiq, it is not functional, and I need to switch it off and on 
once or twice. So, although there is room for improvement, I can live 
with the current state of affairs.

-- 
Grinnikend door het leven...

closed 2023-06-28 by Pinglinux