linuxwacom/input-wacom #346: No "Mouse" mode supported.

tromas71, 2023-02-16:
On windows, was a setting which I can toogle and tablet behaves like a mouse, by not using an absolute position. But when I select Relative mode, it just maps all tablet area to a smaller area on a computer screen. I need to have and "Mouse mode" in the driver. How could that feature be implemented. Sorry if I am making a problem on a empty place, but I am too stupid in this area to do what I wanted. How can such feature be implemented?

Pinglinux, 2023-03-31:
Ok, I tested "Touchpad (relative)" for Tracking Mode under "Wacom Tablet" on Ubuntu 22.04. I can confirm that the stylus stays at "Tablet (absolute)" Tracking Mode. "Wacom Tablet" settings is part of Gnome project, which doesn't belong here. linuxwacom mainly enables the tablet at low-level, such as drivers. Please report the bug to https://gitlab.gnome.org/groups/GNOME/-/issues.

Pinglinux, 2023-04-21:
It's a Gnome issue.

closed 2023-04-21 by Pinglinux