linuxwacom/input-wacom #347: Mouse randomly moving whenever I try to use the tablet.

sloomy-chan, 2023-02-25:
This has never happened before, and I have no idea what is causing this. The only notable difference between now and then is the fact that I've changed distros. It worked with: Mint cinnamon, Mint XFCE, Kubuntu, Pop OS, Ubuntu. Now I'm using KDE Neon and this happens: https://imgur.com/a/VdOFxDa
I am using the latest version from this repository. I have seen a few threads about a somewhat similar issue, but they are all very old, and seem to have been fixed already by a commit.

sloomy-chan, 2023-02-25:
Update: I tested it on another computer, that was using windows, and it worked just fine, so It's not a hardware issue

khseal, 2023-03-27:
Try xsetwacom --set 11 RawSample 1

Pinglinux, 2023-03-29:
If the tablet worked fine with all the other distros, I'd say there is something wrong with KDE Neon.

jigpu, 2023-04-03:
That almost looks like interference being picked up by the tablet. The tablet is sensitive to nearby electromagnetic fields and can misbehave if placed too near to other electronic devices. Does the issue still occur (or perhaps lessen) if you move the tablet to a different place on your desk, place the tablet in your lap, or switch off nearby lights?

sloomy-chan, 2023-04-03:
Alright, so it's not a driver issue, nor a hardware issue, nor a OS issue. It's the goddamn laptop charger... This started happening when I switched chargers, since the old one was acting up. My fix is to plug in my printer on the same outlet as the charger. Don't ask me why this works, but it definitely works. Thanks everyone for trying to help me.

closed 2023-04-03 by sloomy-chan