linuxwacom/input-wacom #349: Stylus Too Sensitive

linuxgamingcentral, 2023-03-28:
I have the Intuos S. I've discovered the stylus is *way* too sensitive. I can't even draw a straight line with it. Only thing I could dig up was this old [Stack Exchange thread](https://unix.stackexchange.com/questions/179644/how-to-make-a-wacom-tablet-stylus-slower-less-sensitive) and adjusting the device's area. It works...somewhat. But then only a portion of the tablet's surface becomes usable; the mouse cursor won't move past a certain threshold.

Any way to properly adjust the sensitivity? I'm on Fedora with GNOME by the way.

jigpu, 2023-04-03:
The trick suggested by the linked thread seems to work for me and I don't see any unusable portion of the tablet appearing. Can you share the default Area that is set when you first connect your Intuos S, and the values that you are setting it to? Also, please share what model number your Intuos S is (e.g. CTL-4100 --- it can be found on the under side of the tablet).

linuxgamingcentral, 2023-04-03:
Default area for both the Stylus and Eraser is 0 0 15200 9500. After setting them to 45600 28500 the cursor does indeed slow down but only the top-left corner of my desktop becomes usable.

Model is CTL-4100.

By the way, the tablet only works after setting it to Android mode.

Pinglinux, 2023-04-06:
On Linux, it should work without going into Android mode. Which Fedora version do you use?

closed 2023-04-11 by linuxgamingcentral