linuxwacom/input-wacom #350: Spike in CPU usage with Wacom Intuos Pro L connection

josepablo-espinoza, 2023-03-30:
I am experiencing a problem with my Wacom Intuos Pro L tablet that causes a spike in CPU usage. Every time I connect the tablet, a process named "/usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch" runs and lasts for about 3 minutes. During this time, the CPU usage spikes to 100%.

I am running a Linux distribution with kernel version 5.10.0-20-amd64 #1 SMP Debian 5.10.158-2 (2022-12-13) x86_64 GNU/Linux and xsetwacom version 0.34.99. I have checked the Xorg log file (/var/log/Xorg.0.log) and found the following information:

No input driver is specified for the Wacom Intuos Pro L Pen, Pad, and Finger. The device may have been added with another device file.

I would appreciate any help in resolving this issue. Thank you.

/var/log/Xorg.0.log

```
[ 36454.700] (II) config/udev: Adding input device Wacom Intuos Pro L Pen (/dev/input/mouse3)
[ 36454.700] (II) No input driver specified, ignoring this device.
[ 36454.700] (II) This device may have been added with another device file.
[ 36454.702] (II) config/udev: Adding input device Wacom Intuos Pro L Pad (/dev/input/js0)
[ 36454.702] (II) No input driver specified, ignoring this device.
[ 36454.702] (II) This device may have been added with another device file.
[ 36454.703] (II) config/udev: Adding input device Wacom Intuos Pro L Finger (/dev/input/mouse4)
[ 36454.703] (II) No input driver specified, ignoring this device.
[ 36454.703] (II) This device may have been added with another device file.
[ 36454.890] (II) config/udev: Adding input device Wacom Intuos Pro L Pen (/dev/input/event22)
[ 36454.890] (**) Wacom Intuos Pro L Pen: Applying InputClass "libinput tablet catchall"
[ 36454.890] (**) Wacom Intuos Pro L Pen: Applying InputClass "Wacom USB tablet class"
[ 36454.890] (**) Wacom Intuos Pro L Pen: Applying InputClass "Wacom tablet class"
[ 36454.890] (II) LoadModule: "wacom"
[ 36454.890] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[ 36454.892] (II) Module wacom: vendor="X.Org Foundation"
[ 36454.892] 	compiled for 1.20.4, module version = 0.34.99
[ 36454.892] 	Module class: X.Org XInput Driver
[ 36454.892] 	ABI class: X.Org XInput driver, version 24.1
[ 36454.892] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
	Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
	Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
	Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
	CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
	CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
	Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
	PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
	DTU2231, DTU1631, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
	Intuos2 12x12, Intuos2 12x18, Intuos2 6x8 , Volito, PenStation,
	Volito2 4x5, Volito2 2x3, PenPartner2, Bamboo, Bamboo1, Bamboo1 4x6,
	Bamboo1 5x8, Intuos3 4x5, Intuos3 6x8, Intuos3 9x12, Intuos3 12x12,
	Intuos3 12x19, Intuos3 6x11, Intuos3 4x6, Intuos4 4x6, Intuos4 6x9,
	Intuos4 8x13, Intuos4 12x19, Intuos4 WL USB Endpoint,
	Intuos4 WL Bluetooth Endpoint, Intuos5 touch S, Intuos5 touch M,
	Intuos5 touch L, Intuos5 S, Intuos5 M, Intuos Pro S, Intuos Pro M,
	Intuos Pro L, Cintiq 21UX, Cintiq 20WSX, Cintiq 12WX, Cintiq 21UX2,
	Cintiq 24HD, Cintiq 22HD, Cintiq 24HD touch (EMR digitizer),
	Cintiq 13HD, DTK2241, DTH2242, Cintiq 22HDT, Mobilestudio Pro 13,
	MobileStudio Pro 16, MobileStudio Pro 13, MobileStudio Pro 16,
	Cintiq 13 FHD Pro, Cintiq 16 UHD Pro, Cintiq 13 FHD Pro,
	Cintiq 16 UHD Pro, TabletPC 0x90, TabletPC 0x93, TabletPC 0x97,
	TabletPC 0x9A, CapPlus  0x9F, TabletPC 0xE2, TabletPC 0xE3,
	TabletPC 0xE5, TabletPC 0xE6, TabletPC 0xEC, TabletPC 0xED,
	TabletPC 0xEF, TabletPC 0x100, TabletPC 0x101, TabletPC 0x10D,
	TabletPC 0x116, TabletPC 0x12C, TabletPC 0x4001, TabletPC 0x4004,
	TabletPC 0x5000, TabletPC 0x5002, usb:172f:0024, usb:172f:0025,
	usb:172f:0026, usb:172f:0027, usb:172f:0028, usb:172f:0030,
	usb:172f:0031, usb:172f:0032, usb:172f:0033, usb:172f:0034,
	usb:172f:0035, usb:172f:0036, usb:172f:0037, usb:172f:0038,
	usb:172f:0039, usb:172f:0051, usb:172f:0052, usb:172f:0053,
	usb:172f:0054, usb:172f:0055, usb:172f:0056, usb:172f:0057,
	usb:172f:0058, usb:172f:0500, usb:172f:0501, usb:172f:0502,
	usb:172f:0503, usb:1b96:0001, usb:17ef:6004
[ 36454.895] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Pen'
[ 36454.895] (**) Wacom Intuos Pro L Pen: always reports core events
[ 36454.895] (**) Option "Device" "/dev/input/event22"
[ 36454.953] (II) Wacom Intuos Pro L Pen: type not specified, assuming 'stylus'.
[ 36454.953] (II) Wacom Intuos Pro L Pen: other types will be automatically added.
[ 36454.953] (--) Wacom Intuos Pro L Pen stylus: using pressure threshold of 851 for button 1
[... 72 lines cut ...]
[ 36455.178] (**) Wacom Intuos Pro L Finger: Applying InputClass "Wacom tablet class"
[ 36455.178] (**) Wacom Intuos Pro L Finger: Applying InputClass "Wacom touchpad class"
[ 36455.178] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Finger'
[ 36455.178] (**) Wacom Intuos Pro L Finger: always reports core events
[ 36455.178] (**) Option "Device" "/dev/input/event24"
[ 36455.180] (EE) Wacom Intuos Pro L Finger: Invalid type 'stylus' for this device.
[ 36455.181] (EE) Wacom Intuos Pro L Finger: Invalid type 'eraser' for this device.
[ 36455.181] (EE) Wacom Intuos Pro L Finger: Invalid type 'cursor' for this device.
[ 36455.181] (II) Wacom Intuos Pro L Finger: type not specified, assuming 'touch'.
[ 36455.181] (II) Wacom Intuos Pro L Finger: other types will be automatically added.
[ 36455.181] (--) Wacom Intuos Pro L Finger touch: maxX=12400 maxY=8640 maxZ=0 resX=40000 resY=40000 
[ 36455.181] (II) Wacom Intuos Pro L Finger touch: hotplugging dependent devices.
[ 36455.181] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'stylus' for this device.
[ 36455.181] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'eraser' for this device.
[ 36455.181] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'cursor' for this device.
[ 36455.181] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'pad' for this device.
[ 36455.181] (II) Wacom Intuos Pro L Finger touch: hotplugging completed.
[ 36455.205] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:056A:0358.0003/input/input35/event24"
[ 36455.205] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Finger touch" (type: TOUCH, id 20)
[ 36455.206] (**) Wacom Intuos Pro L Finger touch: (accel) keeping acceleration scheme 1
[ 36455.207] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration profile 0
[ 36455.207] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration factor: 2.000
[ 36455.207] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration threshold: 4

```

josepablo-espinoza, 2023-03-30:
Thank you for your response, 

> Are you using a laptop?
yes a thinkpad t480

>Can you disable the IBM TrackPoint to see if it makes a difference?
I disabled IBM Trackpoint, the same process still spikes but I think duration feels shorter but not by much.
below /var/log/Xorg.0.log with the device disabled.

```
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (EE) Wacom Intuos Pro L Finger touch: Error reading wacom device : No such device
[ 21387.781] (II) config/udev: removing device Wacom Intuos Pro L Finger touch
[ 21387.844] (II) UnloadModule: "wacom"
[ 21407.400] (II) config/udev: Adding input device Wacom Intuos Pro L Pen (/dev/input/mouse3)
[ 21407.400] (II) No input driver specified, ignoring this device.
[ 21407.400] (II) This device may have been added with another device file.
[ 21407.403] (II) config/udev: Adding input device Wacom Intuos Pro L Finger (/dev/input/mouse4)
[ 21407.403] (II) No input driver specified, ignoring this device.
[ 21407.403] (II) This device may have been added with another device file.
[ 21407.406] (II) config/udev: Adding input device Wacom Intuos Pro L Pad (/dev/input/js0)
[ 21407.406] (II) No input driver specified, ignoring this device.
[ 21407.406] (II) This device may have been added with another device file.
[ 21407.532] (II) config/udev: Adding input device Wacom Intuos Pro L Pen (/dev/input/event22)
[ 21407.532] (**) Wacom Intuos Pro L Pen: Applying InputClass "libinput tablet catchall"
[ 21407.532] (**) Wacom Intuos Pro L Pen: Applying InputClass "Wacom USB tablet class"
[ 21407.532] (**) Wacom Intuos Pro L Pen: Applying InputClass "Wacom tablet class"
[ 21407.532] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Pen'
[ 21407.532] (**) Wacom Intuos Pro L Pen: always reports core events
[ 21407.532] (**) Option "Device" "/dev/input/event22"
[ 21407.591] (II) Wacom Intuos Pro L Pen: type not specified, assuming 'stylus'.
[ 21407.591] (II) Wacom Intuos Pro L Pen: other types will be automatically added.
[ 21407.591] (--) Wacom Intuos Pro L Pen stylus: using pressure threshold of 851 for button 1
[ 21407.591] (--) Wacom Intuos Pro L Pen stylus: maxX=62200 maxY=43200 maxZ=8191 resX=200000 resY=200000  tilt=enabled
[ 21407.591] (II) Wacom Intuos Pro L Pen stylus: hotplugging dependent devices.
[ 21407.592] (EE) Wacom Intuos Pro L Pen stylus: Invalid type 'touch' for this device.
[ 21407.592] (EE) Wacom Intuos Pro L Pen stylus: Invalid type 'pad' for this device.
[ 21407.592] (II) Wacom Intuos Pro L Pen stylus: hotplugging completed.
[ 21407.639] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0358.0008/input/input49/event22"
[ 21407.639] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Pen stylus" (type: STYLUS, id 16)
[ 21407.642] (**) Wacom Intuos Pro L Pen stylus: (accel) keeping acceleration scheme 1
[ 21407.642] (**) Wacom Intuos Pro L Pen stylus: (accel) acceleration profile 0
[ 21407.642] (**) Wacom Intuos Pro L Pen stylus: (accel) acceleration factor: 2.000
[ 21407.642] (**) Wacom Intuos Pro L Pen stylus: (accel) acceleration threshold: 4
[ 21407.702] (**) Wacom Intuos Pro L Pen eraser: Applying InputClass "libinput tablet catchall"
[ 21407.702] (**) Wacom Intuos Pro L Pen eraser: Applying InputClass "Wacom USB tablet class"
[ 21407.702] (**) Wacom Intuos Pro L Pen eraser: Applying InputClass "Wacom tablet class"
[ 21407.702] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Pen eraser'
[ 21407.702] (**) Wacom Intuos Pro L Pen eraser: always reports core events
[ 21407.702] (**) Option "Device" "/dev/input/event22"
[ 21407.702] (**) Option "Type" "eraser"
[ 21407.702] (--) Wacom Intuos Pro L Pen eraser: maxX=62200 maxY=43200 maxZ=8191 resX=200000 resY=200000  tilt=enabled
[ 21407.715] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0358.0008/input/input49/event22"
[ 21407.715] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Pen eraser" (type: ERASER, id 17)
[ 21407.717] (**) Wacom Intuos Pro L Pen eraser: (accel) keeping acceleration scheme 1
[ 21407.717] (**) Wacom Intuos Pro L Pen eraser: (accel) acceleration profile 0
[ 21407.717] (**) Wacom Intuos Pro L Pen eraser: (accel) acceleration factor: 2.000
[ 21407.717] (**) Wacom Intuos Pro L Pen eraser: (accel) acceleration threshold: 4
[ 21407.719] (**) Wacom Intuos Pro L Pen cursor: Applying InputClass "libinput tablet catchall"
[ 21407.719] (**) Wacom Intuos Pro L Pen cursor: Applying InputClass "Wacom USB tablet class"
[ 21407.719] (**) Wacom Intuos Pro L Pen cursor: Applying InputClass "Wacom tablet class"
[ 21407.719] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Pen cursor'
[ 21407.719] (**) Wacom Intuos Pro L Pen cursor: always reports core events
[ 21407.719] (**) Option "Device" "/dev/input/event22"
[... 37 lines cut ...]
[... cut ...]
[ 21407.915] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:056A:0358.0009/input/input53/event24"
[ 21407.915] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Finger touch" (type: TOUCH, id 20)
[ 21407.917] (**) Wacom Intuos Pro L Finger touch: (accel) keeping acceleration scheme 1
[ 21407.917] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration profile 0
[ 21407.917] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration factor: 2.000
[ 21407.917] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration threshold: 4

```

Pinglinux, 2023-03-30:
If disabling the TrackPoint doesn't fix the issue, I'd think something else in the X space is competing with the tablet. BTW, input-wacom is the Wacom kernel driver. Your issue is in userland, not in the kernel.

I wonder if you are willing to build and install the latest X driver by following the steps at [https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver]. The latest version is 1.1.0 [https://github.com/linuxwacom/xf86-input-wacom/releases].   Since we didn't figure out the root cause of the issue, I'm not confident that updating the X driver would fix the issue. Another step you can take is to update your Linux to the latest, which may fix more system-wide issues than updating the driver itself.

josepablo-espinoza, 2023-03-31:
Thanks @Pinglinux for your time, I'm fishing wide to narrow my options down , my problem at the moment is just an inconvenience updating may be worth trying in the future. Is there another command or file worth looking at to get more information about this problem?

Pinglinux, 2023-03-31:
What you described doesn't sound like an isolated issue. I suspect the root cause is either in the system or in the OS. Is this the first time you use a Wacom tablet on the system? Can you try the tablet on another system, such as Windows, to rule out the hardware issue (I doubt it is a hardware)?

Pinglinux, 2024-05-17:
Reopen it if necessary.

closed 2024-05-17 by Pinglinux