linuxwacom/input-wacom #351: initialization problem with Cintiq 22HD?

Nuliel, 2023-04-02:
Hello,
I try to help someone with a Wacom Cintiq 22HD, he is on Ubuntu 22.04.
The tablet is not usable if he boots on Ubuntu with tablet plugged before booting, but if he unplugs and plugs the tablet after Ubuntu boot, tablet becomes usable. 
Reloading wacom module with modprobe doesn't help.
Note that reloading wacom module when tablet is usable makes it unusable. 
Is there an initialization problem for this model?
Thanks

Pinglinux, 2023-04-06:
What do you see in the dmesg output, when the tablet is connected before you disconnect and reconnect the tablet?

Pinglinux, 2023-04-21:
demsg looks fine. If power cycle or reconnect the device worked, it could be something in the userland. Which Windows system, Xorg or Wayland, are you using?

Nuliel, 2023-04-23:
I use Xorg, because with wayland xsetwacom list says "Wayland devices found but this tool is incompatible with Wayland"

Pinglinux, 2023-04-25:
Then, please post your Xorg.0.log here, after you disconnected or power cycled the tablet. Basically, after you know the tablet is working...

closed 2023-06-28 by Pinglinux