linuxwacom/input-wacom #355: Cintiq 13HD behaving erratically and crashing to login screen

ardemd, 2023-04-19:
Hi, I'm having trouble with my cintiq 13HD touch and would appreciate some pointers on how to investigate what is going on:

It doesn't seem to obey the command to turn off the touch control, or it does, only for a limited time and it gets back on.

A few seconds after switching on the tablet, the mouse pointer goes invisible everywhere on other monitors/apps, and the way to get it back to normal is to touch the tablet screen either with the stylus OR finger...reinforcing the weird touch behavior described above. I can set Touch off again...but this all loops after some 30 seconds.

while drawing on the tablet, it stops recognizing the stylus input randomly from time to time...requiring a power cycle on the tablet.

The worst of all are the crashes to login screen, also random.

This is all new, it worked well for many years. 
My machine has 2 regular monitors and an intuos 4 that behaves normally.
running latest Mint, ryzen 5 3600, nvidia 1060 6Gb.

Pinglinux, 2023-04-21:
Those random issues are the hardest to troubleshoot. If the device worked well before, I'd think it's something in the latest update. Which command did you use to disable touch? If you are under Xorg server, please post your Xorg.0.log here. Also, post your dmesg as well

ardemd, 2023-05-06:
I think I've got some connection issue that causes the cintiq to randomly disconnect and immediately reconnect the usb. That explains why setting touch off via xsetwacom works, but only until the tablet reconnects again with the default behavior being touch on.
The thing is, depending on the distro, this hiccup would range from barely noticeable (manjaro), to catastrophic (mint) for some reason.
Once I added Option "Touch" "off" to the code block "Wacom USB touchscreen class" in  /usr/share/X11/xorg.conf.d/70-wacom.conf  the tablet seems to be working as expected again, even the connection issues seem to be gone.

Pinglinux, 2023-06-28:
The issue sounds like some kind of hardware/connection issue, which is beyond this project's scope.

closed 2023-06-28 by Pinglinux