linuxwacom/input-wacom #362: 'Double free' crashes GNOME

enkidu-smith, 2023-05-31:
Ubuntu 23.04
gnome-shell 44.0-2ubuntu3
libwacom 2.6.0-1
xserver-xorg-input-wacom 1:1.1.0-1ubuntu1
Wacom Intuos S

Intermittently Gnome crashes. Adjacent to the log message are messages about the tablet and the 'double free' message. Gnome restarts and everything is fine until the next time. The Wacom Intuos S behaves normally. The tablet is plugged in, but not being used when the crash occurs.

Note: XOrg is being used, not Wayland.

Log extract:
2023-05-31T16:03:30.086313+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (EE) Wacom Intuos S Pad pad: Error reading wacom device : No such device
2023-05-31T16:03:30.092299+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (II) config/udev: removing device Wacom Intuos S Pad pad
2023-05-31T16:03:30.092389+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (II) UnloadModule: "wacom"
2023-05-31T16:03:30.092435+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (II) systemd-logind: releasing fd for 13:70
2023-05-31T16:03:30.160492+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (EE) Wacom Intuos S Pen stylus: Error reading wacom device : No such device
2023-05-31T16:03:30.166884+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (II) config/udev: removing device Wacom Intuos S Pen stylus
2023-05-31T16:03:30.168782+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (II) UnloadModule: "wacom"
2023-05-31T16:03:30.168888+12:00 bumblebee /usr/libexec/gdm-x-session[10220]: (II) systemd-logind: releasing fd for 13:69
2023-05-31T16:03:30.175976+12:00 bumblebee gnome-shell[10435]: free(): double free detected in tcache 2
2023-05-31T16:03:30.176159+12:00 bumblebee gnome-shell[10435]: GNOME Shell crashed with signal 6
2023-05-31T16:03:30.176235+12:00 bumblebee gnome-shell[10435]: == Stack trace for context 0x55f6e13b3cb0 ==

Pinglinux, 2023-06-27:
Was there any Wacom/Intuos related messages in the kernel? Run dmesg from a console and share it here, if you can tell anything. Your error message was X/Gnome related.

Was the tablet connected to the system all the time? Did the system go to sleep mode while "not being used"? "(EE) ... : Error reading wacom device : No such device" could be a valid message since the device may not available. If that's true, the "double free" issue is in Gnome.

enkidu-smith, 2023-06-28:
I have switched back from Wayland to XOrg and the issue appears to be gone. I am unable to provide the dmesg output, sorry.

You are probably correct when you suggest that it was a Gnome issue, as I note that the version of Gnome on my system is now 44.1.0ubuntu1 but it was gnome-shell 44.0-2ubuntu3 when I was having issues.

I will close this issue now, and if I have further problems I will report them and reference this issue. Thanks!

closed 2023-06-28 by enkidu-smith