linuxwacom/input-wacom #365: https://github.com/linuxwacom/input-wacom/tree/for-6.4 wacom_i2c.c:281:27: note: (near initialization for ‘wacom_i2c_driver.<anonymous>.probe’)

ohquait, 2023-06-03:
> 
make -C /lib/modules/6.3.5-pclos1/build M=/var/lib/dkms/input-wacom/1.0.0-1pclos2023/build modules
make[1]: Entering directory '/usr/src/kernel-devel-6.3.5-pclos1'
  CC [M]  /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_wac.o
  CC [M]  /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_sys.o
  LD [M]  /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom.o
  CC [M]  /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_w8001.o
  CC [M]  /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_i2c.o
/var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_i2c.c:281:27: error: initialization of ‘int (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *, const struct i2c_device_id *)’ [-Werror=incompatible-pointer-types]
  281 |         .probe          = wacom_i2c_probe,
      |                           ^~~~~~~~~~~~~~~
/var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_i2c.c:281:27: note: (near initialization for ‘wacom_i2c_driver.<anonymous>.probe’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build/wacom_i2c.o] Error 1
make[1]: *** [Makefile:2004: /var/lib/dkms/input-wacom/1.0.0-1pclos2023/build] Error 2
make[1]: Leaving directory '/usr/src/kernel-devel-6.3.5-pclos1'
make: *** [Makefile:16: all] Error 2

from rel-tarball,
i will next provide result from build with git-cloned source

ohquait, 2023-06-03:
all works good on kernel-5.15...series on VBOX
[https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/Input-Wacom-1.0.0-Screenshot_2023-06-03_09-34-33.png](url)

jigpu, 2023-06-07:
The Linux 6.3 kernel that you're building against has made changes that break the current input-wacom 1.0.0 release. See #367. These issues should be fixed shortly.

Pinglinux, 2023-06-27:
Jason's fix has been merged (https://github.com/linuxwacom/input-wacom/pull/368). Please clone the code from the current source by:

$ git clone https://github.com/linuxwacom/input-wacom.git
$ cd input-wacom
$ ./autogen.sh

then follow the steps at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source to build the driver.

Pinglinux, 2023-09-26:
Fix is in 1.1.0.

closed 2023-09-26 by Pinglinux