linuxwacom/input-wacom #370: verify installation with gpg

earthfail, 2023-06-10:
Hi, I'm following the instruction in [Wiki](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-sourcel).
After downloading [input-wacom-1.0.0.tar.bz2](https://github.com/linuxwacom/input-wacom/releases/download/v1.0.0/input-wacom-1.0.0.tar.bz2) and [input-wacom-1.0.0.tar.bz2.sig](https://github.com/linuxwacom/input-wacom/releases/download/v1.0.0/input-wacom-1.0.0.tar.bz2.sig) I get an error message after running the command
`gpg --verify input-wacom-1.0.0.tar.bz2.sig input-wacom-1.0.0.tar.bz2`

> Can't check signature: No public key

I can't seem to find this public key anywhere. After searching the web I understood that it is not necessary to the installation but recommended to verify the file.

Joshua-Dickens, 2023-06-14:
Hi earthfail, the public key should be available to fetch from the [ubuntu keyserver](https://keyserver.ubuntu.com/) to fetch. If that doesn't work please let me know so I can help fix any issues.

earthfail, 2023-06-14:
Thanks Joshua for the reply. I'm still new to the process of downloading and verifying packages from github so it took me a while.
I will write what worked for me for documentation:
1. search for your name on [ubuntu keyserver](https://keyserver.ubuntu.com/) , Joshua Dickens.
2. copied key id , a8af906d9307fbad
3. run `gpg --keyserver hkp://keyserver.ubuntu.com --recv-keys a8af906d9307fbad` to import key
4. verify package with `gpg --verify file.sig file`

Thank you for your help.

Pinglinux, 2023-06-15:
Nice to see you found the key and resolved the issue. Job well done @earthfail!

closed 2023-06-15 by Pinglinux