linuxwacom/input-wacom #371: Wacom Intuos BT M - system unresponsive

JustCryen, 2023-06-13:
Hi, I have a weird issue with this wacom tablet, it's the `CTL-6100WL` model to be exact.

So in short, seconds after pen hovers over the pad surface all the system input stops working. I'm using wayland as my display server with gnome as the DE. 
I still can see the system notifications and BT icon reacting to me shutting off the tablet and turning it back on, so overall the system is running fine… but all input stops working.
So no keyboard, mouse, touchpad and wacom tablet input works at all… at this point the only thing I can do is basically force a reboot by holding down the power button.

It did work a while back and only occasionally entering this state but now, I get this behaviour basically all the time when trying to use the tablet.
I mostly use it in the BT wireless mode, for now I'm not sure if it also freezes the input when wired (I'll update this in the comments when I get to trying it in wired mode),

### System info: ###
OS: 
**arch linux**

Desktop environment:
**gnome-shell** 1:44.2-1
**mutter** 44.2-1

Display server:
**wayland** 1.22.0-1

Kernel:
**linux-lts** 6.1.32-1

Other packages:
**libinput** 1.23.0-1
**libwacom** 2.7.0-1

### Logs and other info: ###
*>* libwacom-list-local-devices
```
devices:
- name: 'Wacom Intuos BT M'
  bus: 'bluetooth'
  vid: '0x056a'
  pid: '0x0379'
  nodes: 
  - /dev/input/event17: 'Wacom Intuos BT M Pad'
  - /dev/input/event16: 'Wacom Intuos BT M Pen'
```
*>* bluetoothctl info
```
Device DC:2C:26:E2:B5:B4 (public)
        Name: Intuos BT M
        Alias: Intuos BT M
        Class: 0x00002594
        Icon: input-tablet
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v056Ap0379d0000
```
journalctl log:
[wacom_journalctl.txt](https://github.com/linuxwacom/input-wacom/files/11738038/wacom_journalctl.txt)

dmesg log:
[wacom_dmesg.txt](https://github.com/linuxwacom/input-wacom/files/11738040/wacom_dmesg.txt)

JustCryen, 2023-06-13:
It actually seems to work without any issues while wired, so that's good.
But getting the bluetooth mode to a working state would be really useful.

Btw, honestly I'm not even sure where I should write about this issue.
So forgive me if you're not responsible for my problems and I should write somewhere else about it.  If that's the case, would you be so kind and redirect me to a proper place?

Pinglinux, 2023-06-13:
Hi @JustCryen,

You are at the right place to report/"complain about" Wacom device issues :).

Your issue is most likely caused by a patch I submitted recently. The patch applied to kernel 6.3 and backported to 5.4 and later stable branches. Jason fixed the issue [1]. The fix hasn't been accepted to upstream yet. Are you comfortable at applying patch and compiling wacom.ko from the source, i.e. from input-wacom?

[1] https://lkml.org/lkml/2023/6/8/1191

JustCryen, 2023-06-14:
Thanks for the clarification!

As far as I can see this is not only a program parch but it actually requires kernel parching.
As I'm not really familiar with kernel parching, I'd rather use it wired for now and wait for the patch to be implemented upstream.

Offpics, 2023-06-16:
I have the same issue, luckily the wired mode works.

JustCryen, 2023-06-30:
Hi, sorry for what will be basically an off-topic question but I don't want to open a new issue just because I want to ask something.
Is there a way to map custom area / region instead of relying on the reported screen size?
Is this available for other tablets and just not this one?

Pinglinux, 2023-06-30:
Are you looking for something like this: https://github.com/linuxwacom/xf86-input-wacom/wiki/Area-mapping? If your system is based on X server, you can use xsetwacom. If your system is based on Wayland, please checkout libinput project (https://wayland.freedesktop.org/libinput/doc/latest/what-is-libinput.html)

JustCryen, 2023-07-01:
Yeah, I guess I'm looking for something similar.  I've seen the xinput / xsetwacom way of achieving this but I'm on Wayland.
FYI, I can remap buttons and change tablet mapping from absolute to relative already in the settings.  I just want to change the size / area of the tablet in relative mode. also dconf-editor has nothing more than what already is in the gnome-settings.
So yeah… for now I couldn't find anything about area / region mapping for Wayland so I'll look further in the sources you provided tomorrow.

JustCryen, 2023-07-02:
Okay, so from the arch wiki [libinput](https://wiki.archlinux.org/title/libinput#Configuration):
> For [Wayland](https://wiki.archlinux.org/title/Wayland), there is no libinput configuration file. The configurable options depend on the progress of your desktop environment's support for them […] [libinput-config-git](https://aur.archlinux.org/packages/libinput-config-git/)AUR may be used as a work-around. Available options for that tool are documented in the [libinput-config README](https://gitlab.com/warningnonpotablewater/libinput-config).

So it seems like it's an issue with gnome not supporting area scaling?
The mentioned workaround doesn't support anything about graphics tablets.

Vilianexe, 2023-07-05:
> [1] https://lkml.org/lkml/2023/6/8/1191

thank you!, just question, if i understood correctly how the kernel mail works, the patch gonna land in linux 6.5? and can it gonna be backported?

Pinglinux, 2023-07-12:
The patch has been merged. It will be in the latest upstream kernel. It is also backported to kernel stable branches and this project under 4.5 path.

Offpics, 2023-08-10:
I'm on 6.4.9 kernel and it works perfectly. Thank you for the quick fix @Pinglinux!

Pinglinux, 2023-09-28:
The BT fix was merged upstream and backported to input-wacom. The tablet area mapping request goes with libinput/Wayland. We'll work with libinput project (@whot) to figure out a solution.

closed 2023-09-28 by Pinglinux