linuxwacom/input-wacom #373: input-wacom line 38/2275 possible syntax related error help

abyssmal131, 2023-06-24:
Hello,

So, I havea Cintiq 16 on Nobara Linux (a fedora fork) that doesn't work when I plug it in. The computer sees it, none of my inputs go through/in settings I can neigher calibrate or use the "Test Your Settings" function or move my pointer at all. So I'm trying to go through these Linux Wacom Project.

I got the xf86-input-wacom section going, I can see the cintiq 26 has a .tablet file in /share/libwacom, the only thing I can't do is install the kernel driver as explained. I've downloaded the file, and run the provided command and this is what I get:

----------------------------------------------------------------------------------------------------------
BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/6.3.7-200.fsync.fc37.x86_64/source

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0'
Making all in 4.5
make[2]: Entering directory '/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/6.3.7-200.fsync.fc37.x86_64/source M=/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5
make[3]: Entering directory '/usr/src/kernels/6.3.7-200.fsync.fc37.x86_64'
  CC [M]  /home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.o
/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c: In function ‘wacom_update_name’:
/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:38:41: error: implicit declaration of function ‘hid_is_using_ll_driver’; did you mean ‘wacom_is_using_usb_driver’? [-Werror=implicit-function-declaration]
   38 | #define wacom_is_using_usb_driver(hdev) hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                                         ^~~~~~~~~~~~~~~~~~~~~~
/home/USERr/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:2275:21: note: in expansion of macro ‘wacom_is_using_usb_driver’
 2275 |                 if (wacom_is_using_usb_driver(wacom->hdev)) {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:38:71: error: ‘usb_hid_driver’ undeclared (first use in this function); did you mean ‘to_hid_driver’?
   38 | com_is_using_usb_driver(hdev) hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                                                             ^~~~~~~~~~~~~~

/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:2275:21: note: in expansion of macro ‘wacom_is_using_usb_driver’
 2275 |                 if (wacom_is_using_usb_driver(wacom->hdev)) {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:38:71: note: each undeclared identifier is reported only once for each function it appears in
   38 | com_is_using_usb_driver(hdev) hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                                                             ^~~~~~~~~~~~~~

/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:2275:21: note: in expansion of macro ‘wacom_is_using_usb_driver’
 2275 |                 if (wacom_is_using_usb_driver(wacom->hdev)) {
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c: In function ‘wacom_probe’:
/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:38:71: error: ‘usb_hid_driver’ undeclared (first use in this function); did you mean ‘to_hid_driver’?
   38 | com_is_using_usb_driver(hdev) hid_is_using_ll_driver(hdev, &usb_hid_driver)
      |                                                             ^~~~~~~~~~~~~~

/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.c:2829:13: note: in expansion of macro ‘wacom_is_using_usb_driver’
 2829 |         if (wacom_is_using_usb_driver(hdev)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:252: /home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5/wacom_sys.o] Error 1
make[3]: *** [Makefile:2037: /home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5] Error 2
make[3]: Leaving directory '/usr/src/kernels/6.3.7-200.fsync.fc37.x86_64'
make[2]: *** [Makefile:43: all] Error 2
make[2]: Leaving directory '/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0/4.5'
make[1]: *** [Makefile:372: all-recursive] Error 1
make[1]: Leaving directory '/home/USER/Downloads/Figuring_out_Wacom/input-wacom-1.0.0'
make: *** [Makefile:307: all] Error 2
Build Failed
[USER@USERinput-wacom-1.0.0]$ 

----------------------------------------------------------------------------------------------------

So its saying, in /input-wacom-1.0.0/4.5/wacom_sys.c:2275, on line 38 and 2275 there's maybe a syntax problem. I'm basically code illiterate, so I don't know how to fix it. I'm pretty sure it is syntax because with the xf86 download, there were several errors that prevented a proper install.

I'm still working on this myself, but if anyone can see anything on those lines that is just wrong, I'd appreciate the assistance.

Thank you.

Pinglinux, 2023-06-26:
From the error message "wacom_sys.c:38:71: error: ‘usb_hid_driver’ undeclared", I think something is missing in your development environment setup. Everything works well when I built the driver on Fedora 37.

abyssmal131, 2023-06-27:
I did what you said, and I got further, but I run into problems when "loading the module". In /sys/module/wacom*/version, there is no "wacom". I ran the troubleshooting commands from the input-wacom install from source page:

sudo rmmod wacom
sudo rmmod wacom_w8001
sudo modprobe wacom
sudo modprobe wacom_w8001

Which gives me the errors: module wacom is not currently loaded, followed by the wacom w8001 not being loaded.

Also, running:

grep "" /sys/module/wacom*/version

comes up with "No such file or directory", even though its plugged in.

The pen still doesn't do anything. the command:
modinfo wacom | grep version

comes up with:

version:        v2.00-0.49.0.43.g7b934c4
srcversion:     165B0877257EFC51F803239

So, does that mean the kernel driver is loaded?

The command:
libwacom-list-local-devices

Responds with:
- name: 'Wacom Cintiq 16'
  bus: 'usb'
  vid: '0x056a'
  pid: '0x03ae'
  nodes: 
  - /dev/input/event5
  - /dev/input/event4

It seems like everything works, but the pen commands are just not working. It could be the pen, but what else might it be besides a broken pen because I just don't think the pen doesn't work.

Pinglinux, 2023-06-28:
Did you run [ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed" ]

> version: v2.00-0.49.0.43.g7b934c4

tells us that you are running the last version of the driver. Your newly built driver should have v2.00-1.0. in it. Is it possible that you didn't install the newly built driver? You would need to enter your password to install it. Anyway, double check your built and install steps.

In fact, even without the new kernel driver, I think wacom.ko with v2.00-0.49.0.43.g7b934c4 should work for your device.

If your device didn't even move the cursor, it could be a kernel driver issue. In that case, post the output from the dmesg command here. I'll take a look. 

If you just can not calibrate the tablet or can not see the tablet in the "Wacom Tablet" panel under Gnome Settings, that's a Gnome issue.

abyssmal131, 2023-06-28:
I typed in dmesg and this very long text popped up (I don't know what dmesg is):

[    0.000000] microcode: updated early: 0xf0 -> 0xf6, date = 2022-12-26
[    0.000000] Linux version 6.3.7-200.fsync.fc37.x86_64 (mockbuild@494c3510b2ae4dd183bf4558d3faaa24) (gcc (GCC) 12.3.1 20230508 (Red Hat 12.3.1-1), GNU ld version 2.38-27.fc37) #1 SMP PREEMPT_DYNAMIC Tue Jun 13 07:45:18 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd4,gpt2)/vmlinuz-6.3.7-200.fsync.fc37.x86_64 root=UUID=5aaa77f3-e274-441b-a861-df4de3393f49 ro rootflags=subvol=@ quiet splash resume=UUID=881abae3-bf0e-4dca-8530-eeee9b1f9807
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: xstate_offset[9]:  960, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x21f, context size is 968 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 3632
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x0000000079e47fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e48000-0x0000000079e48fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079e49000-0x0000000079e69fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e6a000-0x0000000079e6afff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079e6b000-0x000000007d613fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007d614000-0x000000007de13fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007de14000-0x000000007df43fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007df44000-0x000000007e143fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007e144000-0x000000007effdfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007effe000-0x000000007effefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007efff000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000207dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x774eb018-0x77506257] usable ==> usable
[    0.000000] e820: update [mem 0x774eb018-0x77506257] usable ==> usable
[    0.000000] e820: update [mem 0x774dd018-0x774ea857] usable ==> usable
[    0.000000] e820: update [mem 0x774dd018-0x774ea857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000040400000-0x00000000774dd017] usable
[    0.000000] reserve setup_data: [mem 0x00000000774dd018-0x00000000774ea857] usable
[    0.000000] reserve setup_data: [mem 0x00000000774ea858-0x00000000774eb017] usable
[    0.000000] reserve setup_data: [mem 0x00000000774eb018-0x0000000077506257] usable
[    0.000000] reserve setup_data: [mem 0x0000000077506258-0x0000000079e47fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000079e48000-0x0000000079e48fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000079e49000-0x0000000079e69fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000079e6a000-0x0000000079e6afff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000079e6b000-0x000000007d613fff] usable
[    0.000000] reserve setup_data: [mem 0x000000007d614000-0x000000007de13fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007de14000-0x000000007df43fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007df44000-0x000000007e143fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007e144000-0x000000007effdfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007effe000-0x000000007effefff] usable
[... cut ...]
[23566.668566] input: USB OPTICAL MOUSE  as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2.2/1-3.1.2.2:1.0/0003:18F8:0F97.0062/input/input182
[23566.668656] hid-generic 0003:18F8:0F97.0062: input,hidraw0: USB HID v1.10 Mouse [USB OPTICAL MOUSE ] on usb-0000:00:14.0-3.1.2.2/input0
[23566.671384] input: USB OPTICAL MOUSE  Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2.2/1-3.1.2.2:1.1/0003:18F8:0F97.0063/input/input183
[23566.723362] hid-generic 0003:18F8:0F97.0063: input,hiddev96,hidraw1: USB HID v1.10 Keyboard [USB OPTICAL MOUSE ] on usb-0000:00:14.0-3.1.2.2/input1

abyssmal131, 2023-06-28:
I think I did get mixed up at some point. I'm going through the steps again now

abyssmal131, 2023-06-28:
This is what I did:

Previously, I tried inputing this command:

if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"

Which results in "error: implicit declaration of function " problem from the initial post.

I followed the steps you mentioned:

$ git clone https://github.com/linuxwacom/input-wacom.git
$ cd input-wacom
$ ./autogen.sh

then...

if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"

I did have to input my password.

I tested the module, with my tablet plugged in, with:

grep "" /sys/module/wacom*/version

The response:

grep: /sys/module/wacom*/version: No such file or directory

Again, libwacom seems to be working because the command:

libwacom-list-local-devices

Lists the response:
- name: 'Wacom Cintiq 16'
  bus: 'usb'
  vid: '0x056a'
  pid: '0x03ae'
  nodes: 
  - /dev/input/event5
  - /dev/input/event4

I haven't touched really any xsetwacom settings. I did put a copy of 70-wacom.conf, from the "Configuring X Tutorial", into /etc/X11/xorg.conf.d. There is also a copy of 60.wacom.rules in /etc/udev/rules.d which I added the following line to the bottom of the long list of entries (as I have a cintiq-16):

ATTRS{idVendor}=="056a", ATTRS{idProduct}=="03ae", SYMLINK+="input/tablet-cintiq16"

I can't remember if I added these or they're just there, but these are in /lib/udev/rules.d:

70-wacom.rules
65-libwacom.rules
69-xserver-xorg-input-wacom.rules

I'm pretty sure that's everything I've done.

I hear what you're saying that may be within gnome. There isn't like an "enable/disable pen" button, is there? I watched someone turn that button on in Linux Mint in the Mouse and Touchpad settings, I don't have that, but it really just seems like everything but the pen is working.

Pinglinux, 2023-06-28:
Ok, from your dmesg output I see your tablet is driven by hid-generic driver, not by wacom driver. Please issue the following commands to see if you get anything:

sudo rmmod wacom
sudo modprobe wacom

Move the pen on the tablet to see if cursor moves.

abyssmal131, 2023-06-28:
For the following:

sudo rmmod wacom

I get:
rmmod: ERROR: Module wacom is not currently loaded

For the following:
sudo modprobe wacom

I get nothing. Nothing happens.

Pinglinux, 2023-06-28:
After "sudo modprobe wacom", can you move the cursor with your pen?

abyssmal131, 2023-06-28:
!!!

abyssmal131, 2023-06-28:
YES. I. CAN!

abyssmal131, 2023-06-28:
i could have sworn I input that command.

Its in the troubleshooting steps:

sudo rmmod wacom
sudo rmmod wacom_w8001
sudo modprobe wacom
sudo modprobe wacom_w8001

What the heck

abyssmal131, 2023-06-28:
dude, do you have a "buy me a coffee"? Make one and I will buy you a coffee
https://www.buymeacoffee.com/

What just happened and if it happens again, what do I do?

Pinglinux, 2023-06-28:
Ok, now your pen is driven by Wacom kernel driver. You issue is either in libwacom or the Gnome Control Center. Please post the issue to libwacom to see if Peter has a suggestion for you.

The coffee is unnecessary. It's my pleasure to support users like you ;).

abyssmal131, 2023-06-28:
WOOOOOOO!

Boom baby. BOOM! I'm so happy, thank you.

Um. So, I'll post a link to this issue in libwacom? I'll do that. Thank you again! May your std tests always come up negative.

closed 2023-06-28 by abyssmal131

abyssmal131, 2023-06-28:
Alright, so I restarted my computer and it doesn't work again. I'm assuming those commands will work again, but I'll post what dmesg spits out first:

[    0.000000] microcode: updated early: 0xf0 -> 0xf6, date = 2022-12-26
[    0.000000] Linux version 6.3.7-200.fsync.fc37.x86_64 (mockbuild@494c3510b2ae4dd183bf4558d3faaa24) (gcc (GCC) 12.3.1 20230508 (Red Hat 12.3.1-1), GNU ld version 2.38-27.fc37) #1 SMP PREEMPT_DYNAMIC Tue Jun 13 07:45:18 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd5,gpt2)/vmlinuz-6.3.7-200.fsync.fc37.x86_64 root=UUID=5aaa77f3-e274-441b-a861-df4de3393f49 ro rootflags=subvol=@ quiet splash resume=UUID=881abae3-bf0e-4dca-8530-eeee9b1f9807
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: xstate_offset[9]:  960, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x21f, context size is 968 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 3632
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x0000000079e47fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e48000-0x0000000079e48fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079e49000-0x0000000079e69fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e6a000-0x0000000079e6afff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079e6b000-0x000000007d613fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007d614000-0x000000007de13fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007de14000-0x000000007df43fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007df44000-0x000000007e143fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007e144000-0x000000007effdfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007effe000-0x000000007effefff] usable
[    0.000000] BIOS-e820: [mem 0x000000007efff000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000207dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x774ca018-0x774e5257] usable ==> usable
[    0.000000] e820: update [mem 0x774ca018-0x774e5257] usable ==> usable
[    0.000000] e820: update [mem 0x774bc018-0x774c9857] usable ==> usable
[    0.000000] e820: update [mem 0x774bc018-0x774c9857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000040400000-0x00000000774bc017] usable
[    0.000000] reserve setup_data: [mem 0x00000000774bc018-0x00000000774c9857] usable
[    0.000000] reserve setup_data: [mem 0x00000000774c9858-0x00000000774ca017] usable
[    0.000000] reserve setup_data: [mem 0x00000000774ca018-0x00000000774e5257] usable
[    0.000000] reserve setup_data: [mem 0x00000000774e5258-0x0000000079e47fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000079e48000-0x0000000079e48fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000079e49000-0x0000000079e69fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000079e6a000-0x0000000079e6afff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000079e6b000-0x000000007d613fff] usable
[    0.000000] reserve setup_data: [mem 0x000000007d614000-0x000000007de13fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007de14000-0x000000007df43fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007df44000-0x000000007e143fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007e144000-0x000000007effdfff] reserved
[... cut ...]
[   22.454459] ---[ end trace 0000000000000000 ]---
[   24.598663] rfkill: input handler disabled
[   24.606280] ntfs3: Max link count 4000
[   24.606282] ntfs3: Enabled Linux POSIX ACLs support
[   24.606283] ntfs3: Read-only LZX/Xpress compression included
[   24.992006] ntfs3: sdd1: volume is dirty and "force" flag is not set!
[   64.730959] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   64.730961] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[melkor@Melkor ~]$

abyssmal131, 2023-06-28:
Just to verify, I input the following commands again:

sudo rmmod wacom
sudo modprobe wacom

And again, my pen started working. Perhaps its a permissions thing? When I turn on my computer, I have to input my password to run some programs. Maybe wherever the drivers are, its got root privileges and I just have to do this step everytime. I personally don't care. It works, I'm happy.

Also, hotplugging seems to work too - so I can unplug my tablet then plug it back in and it still works. It seems like just restarting requires the commands you provided.

Pinglinux, 2023-06-28:
I see. The root cause is in the system start/reboot process. You are running a Fedora-like system. Please try the following:

make sure wacom.ko is under "/lib/modules/$(uname -r)/kernel/drivers/hid/", then
update the initramfs by the dracut [1] tool

[1]: https://linuxconfig.org/how-to-build-an-initramfs-using-dracut-on-linux

abyssmal131, 2023-06-28:
I'll do that and tell you what happens

abyssmal131, 2023-06-28:
In:

/lib/modules/(Nobara Kernel)/kernel/drivers/hid

There is a:

wacom.ko.xz

There is, however, nothing with the exact name: wacom.ko

abyssmal131, 2023-06-28:
There is a wacom.ko in:

/usr/lib/modules/(Nobara Kernel)/updates/

abyssmal131, 2023-06-28:
From the link you sent, the command:

sudo dracut --kver $(uname -r)

or is it

sudo dracut --kver uname -r

??

That should build the initramfs and place it into the /boot directory.

Is build the same as install and/or update?

The other command listed: --install
Perhaps that is appropriate, so:

sudo dracut --install uname -r

But, like I said in the previous message, I'm not sure if the file you mentioned is already where its supposed to be or not.

abyssmal131, 2023-06-28:
i have to go to sleep, but I'll keep responding to this. I'm going to remove the dmesg comment as its contents may be sensitive. I don't know, but better safe than sorry.

abyssmal131, 2023-06-28:
Cool, thanks. I sent you an email.