linuxwacom/input-wacom #374: Right click with two finger tap happens only one in ten times

stibinator, 2023-06-26:
I have a Wacom Intuos Pro L with Ubuntu 22.04.2 LTS / Gnome 42.5 / X11 and I have an issue where two-finger tapping to produce a right-click only registers about once in ten times. There does not seem to be any pattern to how it registers: tapping slowly or quickly, fingers close together, fingers spread apart, long or short taps all seem to have about the same effect.

I can see the taps being registered, but they're not being translated into a click most of the time. This is me repeatedly two-finger-tapping on the tablet until it finally registers a right click:
``` ~ xinput --test "Wacom Intuos Pro L Finger touch"
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[2]=0 a[3]=0 a[4]=0 
button press   3 
button release 3 
motion a[2]=0 a[3]=0 a[4]=0 
motion a[0]=10245 a[1]=6720 a[2]=0 a[3]=0 a[4]=0 
motion a[1]=6756 a[2]=0 a[3]=0 a[4]=0 
```

und3f, 2023-09-15:
I have the same issue after upgrading to Debian bookworm -- it is hard to trigger right click with two fingers.

stibinator, 2023-09-18:
I've installed syngestures which fixes the problem, I had to disable quite all the other gestures that it comes with by default because they were causing all sorts of annoyances. But it works for two finger tap to click. https://github.com/mqudsi/syngesture

Pinglinux, 2024-05-17:
There are many multitouch solutions on Linux. Depends on the distribution in use, the feature can be different. With that said, it is not a kernel driver feature. So, it doesn't belong here.

closed 2024-05-17 by Pinglinux

stibinator, 2024-05-19:
Switching to Arch also worked.