linuxwacom/input-wacom #376: DPI scaling and cursor offset

jrotolo1, 2023-07-05:
Hello!

It seems that having the "Global scale" parameter in the "Display Configuration" section of "System Settings" in Kubuntu 20.04 set to anything but 100% breaks the mapping in some applications including native KDE tools. It’s worth noting that some applications seem unaffected, such as web browsers like Firefox or Brave. In my research, I have seen a few people raise a similar issue and there may be a connection to a Qt bug, but no one seems to mention a direct correlation to DPI scaling. 

I have two 4k screens and prefer a value of 150% in Global scale so that everything scales nicely but, but the cursor is clearly and functionally offset, or "detached", from the pen position. This easily demonstrated by dragging a selection on several files in Dolphin where the cursor becomes visibly offset from the selection box making the tablet unusable.

Putting the Global Scale to 100% fixes the issue, but text is too small at this resolution. The temporary fix is going System Settings -> Fonts -> Force font DPI to something like 140, but this isn't ideal for some applications. 

Is there some way to get the tablet (Intuos Pro M [PTH-660]) working with Global Scale at a value other than 100?

I appreciate your time and work!

jigpu, 2023-10-13:
This sounds very similar to https://github.com/linuxwacom/xf86-input-wacom/issues/93 and https://bugreports.qt.io/browse/QTBUG-77826. At least one person has reported that they only experienced when running with HIDPI scaling.

Please try running an affected <program> from the command line as follows and let us know if it has an effect: `QT_XCB_TABLET_LEGACY_COORDINATES=1 <program>`

0x450x6c, 2023-10-18:
I have similar issue in the PC with 4k monitor.

However in the laptop with FHD - I don't have this issue.

System configuration almost identical for both devices.

Workaround `QT_XCB_TABLET_LEGACY_COORDINATES=1 <program>` works.

jigpu, 2023-11-30:
Closing due to inactivity.