linuxwacom/input-wacom #378: No such file or directory

afnecors, 2023-07-15:
I'm on Ubuntu 22.04.2 LTS. I installed the latest release [v1.0.0](https://github.com/linuxwacom/input-wacom/releases/tag/v1.0.0)

```
❯ modinfo wacom | grep version
version:        v2.00-input-wacom
srcversion:     EE6A1F6B957262843DA12F9
vermagic:       5.15.0-76-generic SMP mod_unload modversions
```

After rebooting the system I attached the tablet (Wacom One Small) to the computer but it gives me this message:
```
❯ grep "" /sys/module/wacom*/version
grep: /sys/module/wacom*/version: No such file or directory
```

Pinglinux, 2023-07-27:
Not sure what's going on with your 22.04. I have 20.04. I see the actual output with the grep command. Can you issue "sudo modprobe wacom" to see if it loads the driver? Also, please check if there is anything related to Wacom/wacom in the output of "dmesg" command

jigpu, 2023-10-13:
Closing due to inactivity. If you are still experiencing this issue, please let us know.

closed 2023-10-13 by jigpu