linuxwacom/input-wacom #386: Autodesk Maya + Wacom = Linux hell

Toquita3D, 2023-08-11:
Hi Wacom!

I just bought my first Wacom product ever, and sadly it's not working with the only program I make a living with = Autodesk Maya.
Maya is made by Autodesk, and is the industry-standard for film and CGI, used by Disney and basically everyone in the CG deparment. I've even made a project for their VP in the past.

First, let me start by saying I bought 3 tablets from 3 different brands to test this, and the results are:
- XP-Pen (closed source driver) = Maya works fine
- Huion (closed source driver) = Maya works fine
- Huion (linuxwacom open-source driver + Digimend Kernel drivers) = Maya doesn't work
- Wacom (linuxwacom open-source driver) = Maya doesn't work

I have arthritis and I found that I can't work with Mouse+Keyboard, but using a graphics tablet makes me fully capable of working and I have no pain at all.
I've been using a Huion tablet for some years and it works fine under Linux with Maya (H320M). I then bought an XP-Pen tablet and it also works fine (Deco 01 v2).

Recently I've been curious about Wacom and decided to buy an Intuos Pro S. The size was perfect for my use, but as you'll see in the following video, some of the most important Maya functions don't work.
([VIDEO HERE](https://www.youtube.com/watch?v=s35HdWs3-Hs))

To try this, you can download a trial version of Maya here: https://www.autodesk.com/products/maya/free-trial

I'm currently running:
- Debian 12.1;
- KDE 5.27.5;
- libwacom-common 2.6.0-1;
- NVIDIA RTX 3060 12 GB;
- 128 GB Corsair Vengeance PRO DDR4;
- Ryzen 7 5800X;

I wasn't sure how long a fix of this nature can take, so unfortunately I started the return/refund thing on Amazon just in case I can't get it in time or it's impossible to fix.
If this isn't a fast fix please take all the time necessary. Even if I don't have my Wacom tablet anymore (I can hold it for a week then I have to mail it back) I'm hoping that this fix can help other Wacom + Maya users that are planning to come to Linux, and once this is done I can buy another Wacom tablet (or keep this one in the case of it being a quick fix).

If you have any questions, let me know.

Kind regards,
Toquita

Pinglinux, 2023-09-26:
Hi @Toquita3D, sorry for not getting back to you sooner.

Thank you for sharing your testing results with us. We are definitely willing to figure out the root cause of the issue. You mentioned that the features work for Blender. Does it use the same modifiers as Maya? If yes, my guess is something in Maya caused the issue. If not, what modifiers do you use for Blender?

Hope you have time to provide further details. Thank you,
Ping

Toquita3D, 2023-09-26:
Hey @Pinglinux , thank you for responding!

Sadly, Blender and Maya use different modifiers, so I wouldn't be able to test the same modifiers on both programs.
However, Substance Painter uses the same modifiers as Maya and it works absolutely fine.

I contacted Autodesk Support but they said there was nothing they could do and that it's not a Maya problem.

Sadly I returned my Wacom tablet short after opening this bug.

Let me know if there's anything I can do to help.

Kind regards,
Amarildo

jigpu, 2023-09-27:
> EDIT: forgot to mention that the pen modifier (button 2 "key +Alt button +2") works fine. It's the tablet button modifiers that don't work.
A couple questions:
1. What happens when you change the pen button modifier to be one of the problematic combos? I.e., set `key +Alt button +1` or `key +Alt button +3` on the pen button (instead of the ExpressKey) and then try to tumble / zoom?
2. What happens if you set `button +1` or `button +3` on the pen and then try to tumble or zoom while holding the Alt key?

The ExpressKeys send their data out from a different input device (the "pad" device) then the pen does. Its possible that Maya will only move the viewport in response to the device that has the mouse button held, but since that's the "pad" rather than the pen, nothing happens. I'm not sure how XP Pen and Huion represent their devices, that that might explain why they behave differently.

Pinglinux, 2023-09-27:
Thank you Amarildo for your continuous engagement! Open source community needs people like you!

As @jigpu pointed out, we report EK and pen events as separate tool. That's good for app developers since they know the events are from different type of tools, up front. When they want to combine those events into one action, they need to combine the events. We'll test Maya to prove our theory. We'll explain the details to Autodesk developers.

closed 2023-09-27 by Pinglinux

Toquita3D, 2023-10-07:
Thank you, @Pinglinux . If you feel there's any development over this issue, please let me know. I'd love to buy a new Wacom Intuos Pro S tablet and test this again.