linuxwacom/input-wacom #393: Unknown device_type for 'Wacom Co.,Ltd. Wacom One pen display 11.6"'

twrightsman, 2023-09-26:
I attempted to use a Wacom One 12 tablet with my computer but got the following error:

```
16:59:13 kernel: usb 1-8: new high-speed USB device number 3 using xhci_hcd
16:59:13 kernel: usb 1-8: New USB device found, idVendor=056a, idProduct=03cd, bcdDevice= 1.82
16:59:13 kernel: usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
16:59:13 kernel: usb 1-8: Product: WACOM One 12 HUB
16:59:13 kernel: usb 1-8: Manufacturer: WACOM  Co.,Ltd.
16:59:13 kernel: hub 1-8:1.0: USB hub found
16:59:13 kernel: hub 1-8:1.0: 3 ports detected
16:59:13 kernel: usb 1-8.3: new full-speed USB device number 4 using xhci_hcd
16:59:13 kernel: usb 1-8.3: New USB device found, idVendor=056a, idProduct=03cf, bcdDevice= 0.05
16:59:13 kernel: usb 1-8.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
16:59:13 kernel: usb 1-8.3: Product: Wacom Billboard
16:59:13 kernel: usb 1-8.3: Manufacturer: Wacom Co.,Ltd.
16:59:13 kernel: hid-generic 0003:056A:03CF.0002: hiddev0,hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom Billboard] on usb-0000:00:14.0-8.3/input0
16:59:13 kernel: usbcore: registered new interface driver usbhid
16:59:13 kernel: usbhid: USB HID core driver
16:59:13 kernel: wacom 0003:056A:03CF.0002: Unknown device_type for 'Wacom Co.,Ltd. Wacom Billboard'. Ignoring.
16:59:14 kernel: usb 1-8.2: new full-speed USB device number 5 using xhci_hcd
16:59:14 kernel: usb 1-8.2: New USB device found, idVendor=056a, idProduct=03ce, bcdDevice= 1.01
16:59:14 kernel: usb 1-8.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
16:59:14 kernel: usb 1-8.2: Product: Wacom One pen display 11.6"
16:59:14 kernel: usb 1-8.2: Manufacturer: Wacom Co.,Ltd.
16:59:14 kernel: usb 1-8.2: SerialNumber: <redacted>
16:59:14 kernel: input: Wacom One pen display 11.6" Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2:1.0/0003:056A:03CE.0003/input/input24
16:59:14 kernel: wacom 0003:056A:03CE.0003: hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom One pen display 11.6"] on usb-0000:00:14.0-8.2/input0
16:59:14 kernel: wacom 0003:056A:03CE.0004: Unknown device_type for 'Wacom Co.,Ltd. Wacom One pen display 11.6"'. Ignoring.
16:59:14 kernel: wacom 0003:056A:03CE.0005: Unknown device_type for 'Wacom Co.,Ltd. Wacom One pen display 11.6"'. Ignoring.
```

I built and installed `input-wacom` v1.1.0 and can confirm that it is loaded:

```
$ grep "" /sys/module/wacom*/version
v2.00-input-wacom
```

Pinglinux, 2023-09-27:
The device works out of box if your running on a kernel 4.12 or later. If you built the driver from input-wacom, it should work even if you run on a kernel older than 4.12. 

I am on Fedora 38 (Linux fedora 6.5.5-200.fc38.x86_64). I didn't install the driver from input-wacom. My pen worked when I connected to the system. I see a few "Unknown device_type" too. That's fine since they are non-input USB ports.

[ 4705.820752] usb 1-5.2: Product: DTC121
[ 4705.820756] usb 1-5.2: Manufacturer: Wacom Co.,Ltd.
[ 4705.820795] usb 1-5.2: SerialNumber: 3AHS1E2000020
**[ 4705.838558] input: Wacom DTC121 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.2/1-5.2:1.0/0003:056A:03ED.0008/input/input41**
[ 4705.839183] wacom 0003:056A:03ED.0008: hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. DTC121] on usb-0000:00:14.0-5.2/input0
**[ 4705.840562] wacom 0003:056A:03ED.0009: Unknown device_type for 'Wacom Co.,Ltd. DTC121'. Ignoring.
[ 4705.841921] wacom 0003:056A:03ED.000A: Unknown device_type for 'Wacom Co.,Ltd. DTC121'. Ignoring.**

What bothers me is that it seems hid-generic driver grabbed the device first. Can you do the following steps and post the output from your dmesg?

$ sudo rmmod wacom
$ sudo modprobe wacom

twrightsman, 2023-09-28:
After reading your comment I went ahead and tried the pen and to my surprise the mouse cursor moved on my laptop screen, though the Wacom's display remained black.

I am running Debian 12/Bookworm (`Linux 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux`).

Here is the `journalctl` output after running the commands you asked for:

```
Sep 27 19:58:00  sudo[44609]: twrightsman : TTY=pts/4 ; PWD=/home/twrightsman ; USER=root ; COMMAND=/usr/sbin/rmmod wacom
Sep 27 19:58:00  sudo[44609]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Sep 27 19:58:00  kernel: hid-generic 0003:056A:03CF.0007: hiddev0,hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom Billboard] on usb-0000:00:14.0-8.3/input0
Sep 27 19:58:00  kernel: input: Wacom Co.,Ltd. Wacom One pen display 11.6" System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2:1.0/0003:056A:03CE.0008/input/input59
Sep 27 19:58:00  kernel: input: Wacom Co.,Ltd. Wacom One pen display 11.6" Stylus as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2:1.0/0003:056A:03CE.0008/input/input60
Sep 27 19:58:00  kernel: input: Wacom Co.,Ltd. Wacom One pen display 11.6" as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2:1.0/0003:056A:03CE.0008/input/input61
Sep 27 19:58:00  kernel: hid-generic 0003:056A:03CE.0008: input,hiddev1,hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom One pen display 11.6"] on usb-0000:00:14.0-8.2/input0
Sep 27 19:58:00  kernel: input: Wacom Co.,Ltd. Wacom One pen display 11.6" as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2:1.2/0003:056A:03CE.0009/input/input62
Sep 27 19:58:00  kernel: hid-generic 0003:056A:03CE.0009: input,hidraw3: USB HID v1.10 Mouse [Wacom Co.,Ltd. Wacom One pen display 11.6"] on usb-0000:00:14.0-8.2/input2
Sep 27 19:58:00  kernel: hid-generic 0003:056A:03CE.000A: hiddev2,hidraw4: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom One pen display 11.6"] on usb-0000:00:14.0-8.2/input3
Sep 27 19:58:00  sudo[44609]: pam_unix(sudo:session): session closed for user root
Sep 27 19:58:01  systemd-logind[892]: Watching system buttons on /dev/input/event17 (Wacom Co.,Ltd. Wacom One pen display 11.6" System Control)
Sep 27 19:58:05  sudo[44676]: twrightsman : TTY=pts/4 ; PWD=/home/twrightsman ; USER=root ; COMMAND=/usr/sbin/modprobe wacom
Sep 27 19:58:05  sudo[44676]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Sep 27 19:58:05  kernel: wacom 0003:056A:03CF.0007: Unknown device_type for 'Wacom Co.,Ltd. Wacom Billboard'. Ignoring.
Sep 27 19:58:06  kernel: input: Wacom One pen display 11.6" Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2:1.0/0003:056A:03CE.0008/input/input66
Sep 27 19:58:06  kernel: wacom 0003:056A:03CE.0008: hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom One pen display 11.6"] on usb-0000:00:14.0-8.2/input0
Sep 27 19:58:06  kernel: wacom 0003:056A:03CE.0009: Unknown device_type for 'Wacom Co.,Ltd. Wacom One pen display 11.6"'. Ignoring.
Sep 27 19:58:06  kernel: wacom 0003:056A:03CE.000A: Unknown device_type for 'Wacom Co.,Ltd. Wacom One pen display 11.6"'. Ignoring.
Sep 27 19:58:06  sudo[44676]: pam_unix(sudo:session): session closed for user root
```

Pinglinux, 2023-09-28:
Good to know that the pen moved your cursor. It means Wacom driver worked. If the tablet's screen was dark, it is most likely a power cable or power supply issue. Which cable are you using to power the tablet? Please try both cables to see if it makes a difference.

twrightsman, 2023-09-28:
I purchased a Thunderbolt cable today and can confirm the cable was indeed the issue... I did not have the official Wacom cable. Thank you for your help!

closed 2023-09-28 by twrightsman